esp8266/Arduino #6025: LIGHT_SLEEP_T disables after few requests

tadabrowski, 2019-04-29:
#### Platform

- Hardware:				ESP-12e
- Core Version:			latest
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I'm having a strange problem with a high power consumption after multiple request to the server. I've managed to reproduce it on a modified HelloServer example.

When the program leaves the setup function, power consumption drops. It maintains at a low level. I enter the IP address in the web browser - I get the "hello from esp8266!" response. But when I press F5 multiple times, power consumption rises to the 68mA level and stays. Sometimes it goes back to a low level again after a minute, but rarely.

https://youtu.be/oGBFYEqD97A

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
// modified HelloServer example

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#ifndef STASSID
#define STASSID "tdsiec"
#define STAPSK  "wifi4782"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

ESP8266WebServer server(80);

void handleRoot() {
  server.send(200, "text/plain", "hello from esp8266!");
}

void handleNotFound() {
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void) {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  
  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
   
  wifi_set_sleep_type(LIGHT_SLEEP_T);
}

void loop(void) {
  server.handleClient();
  delay(1000);
}

```

TD-er, 2019-04-30:
I've seen this happen also, but not by calling the light sleep.
Just using lots of delays in the execution does also reduce power consumption, but as soon as you have a number of network requests, its power consumption rises and remains high for a while (sometimes even about 10 minutes), even though the number of delay calls remains the same.

Tech-TX, 2020-01-26:
With Automatic Light Sleep the power level is heavily dependent on both the time you spend in delay() and the WiFi traffic, as TD-er noted.  As soon as you start transmitting the power will go to full (68mA) at least briefly, sometimes for seconds.  Also, if your program goes off to LaLa Land and doesn't hit that **delay(1000)** then the power goes to full.  If you lose the connection the power goes to full.  If you're hitting F5 every 2-3 seconds it'll stay in full power until you get bored and stop hitting refresh.  You don't have any way to see where your code is at any point in time other than the WiFi response.  You might want to add a serial monitor with a bunch of "I'm here!" statements so you can see what's gone wrong.  I hit it around 100 refreshes and it never hung up on me, it always returned to Automatic Light Sleep within a few seconds.

devyte, 2020-02-02:
Closing per previous comment.

closed 2020-02-02 by devyte