esp8266/Arduino #6028: Exception thrown with example StreamHTTPClient.ino

mockersf, 2019-04-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8285 device
- Core Version:			2.5.0-101-gfeb3988b
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Wemos D1 mini Lite
- Flash Mode:			qio
- Flash Size:			1MB
- lwip Variant:		v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:		115200

### Problem Description

When using an http client with a WiFiClient to read the response as a stream, an exception is thrown when deleting the client

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Sketch is the one from here : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/StreamHttpClient/StreamHttpClient.ino

```cpp

/**
   StreamHTTPClient.ino

    Created on: 24.05.2015

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    HTTPClient http;

    WiFiClient client;

    Serial.print("[HTTP] begin...\n");

    // configure server and url
    http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html");
    //http.begin(client, "jigsaw.w3.org", 80, "/HTTP/connection.html");

    Serial.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = http.GET();
    if (httpCode > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTP] GET... code: %d\n", httpCode);

      // file found at server
      if (httpCode == HTTP_CODE_OK) {

        // get lenght of document (is -1 when Server sends no Content-Length header)
        int len = http.getSize();

        // create buffer for read
        uint8_t buff[128] = { 0 };

        // get tcp stream
        WiFiClient * stream = &client;

        // read all data from server
        while (http.connected() && (len > 0 || len == -1)) {
          // get available data size
          size_t size = stream->available();

          if (size) {
            // read up to 128 byte
            int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));

            // write it to Serial
            Serial.write(buff, c);

            if (len > 0) {
              len -= c;
            }
          }
          delay(1);
        }

        Serial.println();
        Serial.print("[HTTP] connection closed or file end.\n");

      }
    } else {
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }

    http.end();
  }

  delay(10000);
}
```

### Debug Messages

```
[SETUP] WAIT 4...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 13
cnt 

connected with XXXXXXXXX, channel 2
dhcp client start...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
ip:192.168.0.27,mask:255.255.255.0,gw:192.168.0.254
[SETUP] WAIT 1...
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
<HTML>
<HEAD>
  <!-- Created with AOLpress/2.0 -->
  <TITLE>Connection Header</TITLE>
</HEAD>
<BODY>
<P>
<IMG ALT="Jigsaw" BORDER="0" WIDTH="212" HEIGHT="49" SRC="/icons/jigsaw">
<H1>
  The <I>Connection</I> header
</H1>
<P>
This page will be served to you with the following headers:
<P>
<CODE>ExtensionHeader: ExtensionValue<BR>
Connection: ExtensionHeader</CODE>
<P>
If you're getting this page through a proxy, you should <I>not</I> see the
<I>ExtensionHeader</I> !
<P>
  <HR>
<BR>
<A HREF="mailto:jigsaw@w3.org">jigsaw@w3.org</A>
</BODY></HTML>

[HTTP] connection closed or file end.
Fatal exception 28(LoadProhibitedCause):
epc1=0x4021443a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000208, depc=0x00000000

Exception (28):
epc1=0x4021443a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000208 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc00 end: 3fffffc0 offset: 01a0
3ffffda0:  00000000 4bc6a7f0 0000111a 3fffff68  
3ffffdb0:  00000000 00000000 4bc6a7f0 00000000  
3ffffdc0:  00000000 00000000 40100142 9fbe76c8  
3ffffdd0:  00000000 4bc6a7f0 3ffe8508 3fffff68  
3ffffde0:  0000111a ffffffff 3ffef1e4 40202448  
3ffffdf0:  0000012c 00000000 00000064 40205352  
3ffffe00:  3ffee50c 000000f7 000000f7 3fffff68  
3ffffe10:  3ffee414 00000000 3fffff68 40202971  
3ffffe20:  3ffe8666 3ffee4d4 3ffffe60 402029ad  
3ffffe30:  3ffe861b 00000000 3ffffe60 402029d4  
3ffffe40:  3ffee414 00000000 00000001 402034cd  
3ffffe50:  3ffee414 00000000 00000001 4020120f  
3ffffe60:  00000000 00000000 3fffff68 3ffef2bc  
3ffffe70:  000d000f ff101f72 fe000050 fe001388  
3ffffe80:  3ffef2d4 0015001f ff000000 70747468  
3ffffe90:  00000000 00101dc8 00000000 00000000  
3ffffea0:  00fffeb0 3ffef1bc 0011001f ffefeffe  
3ffffeb0:  00000000 00000000 0000001e 00000000  
3ffffec0:  00000000 00000000 ffffffff 00000000  
3ffffed0:  0000000a 00000000 00000000 00000012  
3ffffee0:  00000000 00000000 2267726f 67696a3e  
3ffffef0:  40776173 6f2e3377 2f3c6772 3c0a3e41  
3fffff00:  444f422f 2f3c3e59 4c4d5448 6c750a3e  
3fffff10:  493c2064 746f6e3e 3e492f3c 65657320  
3fffff20:  65687420 3e493c0a 65747845 6f69736e  
3fffff30:  6165486e 3c726564 203e492f 503c0a21  
3fffff40:  20200a3e 3e52483c 52423c0a 413c0a3e  
3fffff50:  45524820 6d223d46 746c6961 696a3a6f  
3fffff60:  77617367 2e337740 40207410 00000000  
3fffff70:  00001388 00001115 00000000 3ffef1e4  
3fffff80:  00000026 00000000 3ffee414 40201904  
3fffff90:  3fffdad0 00000000 3ffee414 3ffee47c  
3fffffa0:  3fffdad0 00000000 3ffee44c 40205320  
3fffffb0:  feefeffe feefeffe 3ffe8508 40100459  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vfeb3988b
~ld

```

```
Decoding stack results
0x40100142: millis() at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 186
0x40202448: ClientContext::wait_until_sent(int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h line 328
0x40205352: __yield() at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/cores/esp8266/core_esp8266_main.cpp line 100
0x40202971: WiFiClient::flush(unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 318
0x402029ad: WiFiClient::stop(unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 326
0x402029d4: WiFiClient::stop() at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h line 76
0x402034cd: HTTPClient::~HTTPClient() at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 131
0x4020120f: loop() at /Users/francois/Dev/arduino/http_stream/http_stream.ino line 98
0x40201904: ESP8266WiFiMulti::addAP(char const*, char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 39
0x40205320: loop_wrapper() at /Applications/Arduino.app/Contents/Java/hardware/esp8266git/esp8266/cores/esp8266/core_esp8266_main.cpp line 125
```

Issue sound similar to #5216

earlephilhower, 2019-04-30:
Can you please try with the GIT head version?  I believe the fix for the issue you mentioned was put in after 2.5.0 came out.

mockersf, 2019-04-30:
I opened the issue with HEAD of master

With v2.5.0, the stack trace is a little bit different:
```
0x40100142: millis at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40205f84: esp_yield() at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 91
0x40203a48: ClientContext::wait_until_sent(int) at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/include/ClientContext.h line 328
0x40100544: _umm_free at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1300
0x40203a9d: WiFiClient::flush(unsigned int) at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 318
0x402036fd: WiFiClient::stop(unsigned int) at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 326
0x40201422: delay at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 54
0x40204a81: HTTPClient::~HTTPClient() at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 131
0x4020281b: loop() at /Users/francois/Dev/arduino/http_stream/http_stream.ino line 98
0x40202ef8: ESP8266WiFiMulti::addAP(char const*, char const*) at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 39
0x40206030: loop_wrapper() at /Users/francois/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

devyte, 2019-04-30:
@mockersf 
At the beginning of loop() is this:
```cpp
    HTTPClient http;

    WiFiClient client;
```

What happens if those two lines are inverted?

earlephilhower, 2019-04-30:
It works, thanks to the C++ guarantee of destructor ordering in reverse declaration order.  At least, I hope that's guaranteed.

I can repro it myself w/head but only w/no debugging at all.  Enabling debug of any kind seems to protect that double-freed memory.

Solution is to update the example showing proper declaration order, or move the client construction out of the if{} completely so it's guaranteed a longer life than the httpclient.

mockersf, 2019-04-30:
Inverting the two lines works in both the example and my own sketch !

I think it would be better to invert them in the example, as in my case having them out of the `if` wouldn't have helped me.

mockersf, 2019-04-30:
Thanks @devyte !

PR #6030 (2019-05-01): Update StreamHttpClient.ino
Fix WiFiClient vs. HttpClient declaration order
Fixes #6028
commit 33a4c6a5 (2019-05-01): Update StreamHttpClient.ino (#6030)

Fix WiFiClient vs. HttpClient declaration order
Fixes #6028
closed 2019-05-01 by earlephilhower