esp8266/Arduino #6033: ESP8266 hangs when using HttpClient.readString

saikiranmpr, 2019-05-01:
Dear team, I have an issue with respect to getString() method. We are using ESP8266 where internet connection is little poor. Hence, after sending the HTTP POST request and when trying to read the response from the server, the ESP sometimes hangs and never comes out of it until hard reset is performed manually. This happens randomly. Kindly help me in resolving this. I am attaching the code below:

```
int httpCode = http.POST(postData);  
if (httpCode == 200 && http.getSize() > 0) {
    String payload = http.getString();    //Get the response payload
}
```

arihantdaga, 2019-05-03:
I think i have had a similar issue. I do not know exactly when my esp got stuck, i have a big program, and few devices. In some of the devices with a little poor network, i faced that they just got stuck and no watchdog was triggered. Neither software nor hardware watchdog was triggered. I could not replicate the situation but it just occured 2-3 times in few of my devices. I trie with latest github repository as well as 2.5.0

saikiranmpr, 2019-05-05:
Were you able to find any solution for this?

arihantdaga, 2019-05-05:
Havent found it yet.. I am still working on it.

d-a-v, 2019-05-05:
@saikiranmpr Do you have poor connectivity like @arihantdaga has ?

`HTTPClient::writeToStreamDataBlock` does not handle timeouts.
Can you check if that's your issue by adding on this line:

https://github.com/esp8266/Arduino/blob/ac53c2998e58d373d46a73bbdb7b5292677087b1/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L1383

```
        DEBUG_HTTPCLIENT("%ul: %d/%d bytes read\r\n", millis(), bytesRead, readBytes);
```
Then re-run and enable `Toold>Debug Port>Serial` and `Tools>Debug Level>HTTP Client` ?

saikiranmpr, 2019-05-06:
@d-a-v Yes, I have poor internet connectivity. What making it worse is, we have deployed these devices in a remote area and this issue happens randomly and hence, I am unable to properly reproduce it with consistency at my location. However, to identify the issue this is the progress we made. Earlier, we used PubSubClient and HttpClient for communication. Now, we removed the HttpClient and are using only the PubSubClient for data transfer over MQTT. Even upon using the PubSubClient, the issue is happening. What I feel is the issue with WiFi and PubSubClient mentioned below is happening:

[https://github.com/knolleary/pubsubclient/issues/444](url)

For your reference, this is how we are handling MQTT in loop:
```
void mqtt_handle() {

  if (WiFi.status() == WL_CONNECTED) {

    mqtt_count = 0;
    if (!pubsubclient.connected()) {
      mqtt_reconnect();
    }
    else {
      Serial.println("MQTT already connected");
    }

    if (mqtt_count < mqtt_attempts) {
      pubsubclient.loop();
    }
    else {
      Serial.println("couldn't connect to mqtt");
    }
  }
}

void mqtt_reconnect() {
  mqtt_count = 0;
  Serial.print("Attempting MQTT connection...");
  while (!pubsubclient.connected() && (mqtt_count < mqtt_attempts)) {
    pubsubclient.setServer(mqtt_server, 8883);
    pubsubclient.setCallback(callback);
    mqtt_count++;
    if (pubsubclient.connect("ESP8266Client_1013", "user", "password")) {
      Serial.println("connected");
      pubsubclient.subscribe("node/control/1013");
      pubsubclient.subscribe("status/1013");

    } else {
      Serial.print("failed, rc=");
      Serial.print(pubsubclient.state());
      Serial.println(" Attempting again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(1000);
    }
  }
}
```

Please let me know if anything is wrong in what I am doing. Also, thank you very much for supporting.

d-a-v, 2019-05-06:
@saikiranmpr This may be is a different issue (no http client).
Please explain where you think freezing is happening preferably in a new issue.

arihantdaga, 2019-05-06:
@saikiranmpr  I think what you are doing is , may not be a good idea. This is almost never ending while loop in poor connectivity. From what i can see, if the network is poor or if there is no internet in the wifi, then all the time, your code is just busy retrying connection with mqtt broker. I think you can change that and attempting retries after a certain time but without using the delay. (I am not sure, but if you are new to this, you can look for basic example of ledBlink without delay in arduino ide, u'll get an idea of what i am talking about). 
I think in this case it isnt really the issue of httpClient.

closed 2019-05-06 by d-a-v

saikiranmpr, 2019-05-07:
@arihantdaga There is a limit using condition (mqtt_count < mqtt_attempts) .

PR #6056 (2019-05-06): Honor timeout in HTTPClient
fix #6033 
(bug is reproduced and fixed)