esp8266/Arduino #6034: analogwritefreq clamps legacy ?

tablatronix, 2019-05-01:
I am looking at the freq clamps on `analogWriteFreq`

```c++
extern void __analogWriteFreq(uint32_t freq) {
  if (freq < 100) {
    analogFreq = 100;
  } else if (freq > 40000) {
    analogFreq = 40000;
  } else {
    analogFreq = freq;
  }
}
```

Curious why these limits exist, startWaveform and tone do not have such min max clamping, 

This seems to be possibly legacy limits based on old timer pwm

from eps8266 datasheet

>  the function of PWM is realized by interruption of the
> timer, the minimum resolution reaches as high as 44 ns. PWM frequency range is
> adjustable from 1000 μs to 10000 μs, i.e., between 100 Hz and 1 kHz. When the PWM
> frequency is 1 kHz, the duty ratio will be 1/22727, and a resolution of over 14 bits will be
> achieved at 1 kHz refresh rate

But it seems we are using esp8266_waveform generator soo....

Thoughts?

earlephilhower, 2019-05-01:
The limits come from internal implementation details. The lower limit comes from internal counter overflow (CPU clock cycles) and the upper is because there is an interrupt taken on each edge so if you go too high there is no time for the actual application or even to finish the interrupt you started.

They could probably move a little in either direction, but it's better to use other methods to generate lower frequencies and special hardware to make higher ones.

closed 2019-05-01 by earlephilhower

tablatronix, 2019-05-01:
What was more interesting is that different methods have different clamps

analogwritefreq is clamped 100-40000

Tone also seems to cap out at 5kz ( so 1-5000 )

```
static void _startTone(uint8_t _pin, uint32_t high, uint32_t low, unsigned long duration) {
...
  high = std::max(high, (uint32_t)100);
  low = std::max(low, (uint32_t)100);
```

earlephilhower, 2019-05-01:
There it could be upped to higher, yes.  I think I just put in something that seemed sane at the time w/o actually thinking deeply about it.  If it's a problem, feel free to make a PR to move that comparison to 13us per or so.  The same underlying limits apply, though.  Tone is now just that simple wrapper around the tickless generator code.