esp8266/Arduino #6043: Unable to compile even the blink.ino sketch

mikekgr, 2019-05-03:
The bellow error appeared to my Arduino IDE ( Running in Windows 10 ) :
In file included from E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp:35:0:

C:\Users\kranidis\AppData\Local\Temp\arduino_build_25694/core/core_version.h:1:33: error: unable to find numeric literal operator 'operator"" xffffffff'

 #define ARDUINO_ESP8266_GIT_VER 0x0xffffffff

                                 ^

E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp:65:78: note: in expansion of macro 'ARDUINO_ESP8266_GIT_VER'

 extern const uint32_t __attribute__((section(".ver_number"))) core_version = ARDUINO_ESP8266_GIT_VER;

                                                                              ^
exit status 1
Error compiling for board Generic ESP8266 Module.

I have ESP8266 updated from GIt Master branch

Please keep an eye what is wrong and I am here to supply you any further information you may ask.

Thanks and Best Regards,
Mike Kranidis

devyte, 2019-05-03:
That core_version.h looks corrupted.

mikekgr, 2019-05-03:
Dear @devyte,
thanks a lot for your reply. What do you suggest? To erase the esp8266 folder and to the git install again?

devyte, 2019-05-03:
That's what i would do.

mikekgr, 2019-05-03:
Dear @devyte I have some more observations to you. After having delete the esp8266 directory and following the instructions I did the git installation. Then I just tried the Blink sketch. Unfortunately I faced the same error... Then I re delete the esp8266 directory and I copy one back up ( again, git based ) but this back up contains the git sync of 28/03/2019. So this backup is working fine, compiled the bink.ino sketch etc. Then, I did git update and sorry to say, the problem appeared again. There is something wrong in your latest commits I suppose. Please check and I am here if you need anything from me.
Thanks.

devyte, 2019-05-03:
I just did a clean git install and it works as expected. 
Closing as can't reproduce.

...on Ubuntu, though, not Windows, but still.

>I copy one back up
> contains the git sync of 28/03/2019

Are you installing via git clone or just copying the source tree? did you call get.py after your re-install?

mikekgr, 2019-05-03:
Dear @devyte 
of course I installed using git clone and following the documentation. It is very strange why this is happening... I don't know, I following your git master in my two installations ( one Windows 10, the other Windows 7 Both affected ) for two years now without problems/breakage except this one. Is there any body here with Windows to check? Please...
Thanks anyway. Not to forget, to say you a *VERY BIG* thanks for all the work you have done for this ESP8266 Arduino Core.

mikekgr, 2019-05-03:
17 April 2019 was my last esp8266 git sync and successful compilation if this Info matters to you.

devyte, 2019-05-03:
You're affected, so you should pursue. Try syncing to each commit after that to figure out which one breaks your build. That could provide a clue.

devyte, 2019-05-03:
At a glance, suspect commit c5efb92 Apr/23/2019

devyte, 2019-05-03:
My pythonese isn't very good, but at the very least that default arg to generate in the above commit  looks fishy. Reopening. 
CC @mcspr

mikekgr, 2019-05-03:
> You're affected, so you should pursue. Try syncing to each commit after that to figure out which one breaks your build. That could provide a clue.

Yes I can do that but please tell me the git commands that I have to use ( sorry, not git experience... ) OR even better using the git gui if possible...

mikekgr, 2019-05-03:
1st not good indication: ( this is the output when I did in the git gui Remote -> Fetch from origin )
From https://github.com/esp8266/Arduino
   f1395190..a67aa569  master            -> origin/master
   9ed57a05..1ee31cb4  devyte-configTime -> origin/devyte-configTime
Fetching submodule libraries/ESP8266SdFat
From https://github.com/earlephilhower/ESP8266SdFat
   92ab4b5..2499d4e  master     -> origin/master
Fetching submodule libraries/SoftwareSerial
From https://github.com/plerup/espsoftwareserial
   56654d2..22c53f8  master     -> origin/master
 * [new tag]         4.0.0            -> 4.0.0
 * [new tag]         5.0.0            -> 5.0.0
 * [new tag]         5.0.0-rc1        -> 5.0.0-rc1
 * [new tag]         5.0.0-rc2        -> 5.0.0-rc2
 * [new tag]         5.0.0-rc3        -> 5.0.0-rc3
 * [new tag]         5.0.1            -> 5.0.1
 * [new tag]         5.0.2            -> 5.0.2
 * [new tag]         5.0.3            -> 5.0.3
 * [new tag]         5.0.4            -> 5.0.4
 * [new tag]         rx_hw_interrupts -> rx_hw_interrupts
 * [new tag]         v.2018.04        -> v.2018.04
 * [new tag]         v.2018.12        -> v.2018.12
 * [new tag]         v.2019.03        -> v.2019.03
 * [new tag]         v.2019.03.1      -> v.2019.03.1
 * [new tag]         v.2019.03.2      -> v.2019.03.2
 * [new tag]         v2019.03.3       -> v2019.03.3
 * [new tag]         v2019.03.4       -> v2019.03.4
 * [new tag]         v3.4.1-lerup     -> v3.4.1-lerup
 * [new tag]         v4.0.0           -> v4.0.0
Fetching submodule tools/sdk/lwip2/builder
From https://github.com/d-a-v/esp82xx-nonos-linklayer
   a501b57..2314329  master     -> origin/master
Fetching submodule tools/sdk/ssl/bearssl
From https://github.com/earlephilhower/bearssl-esp8266
   2e1b248..a143020  master     -> origin/master
error: Server does not allow request for unadvertised object b178c0ff023ff87a4754636b624a6d18e6036a07

mcspr, 2019-05-03:
Yes, default git_ver is wrong. `0x` is unnecessary because format string below already has it.

But does this mean git is not installed / not in PATH? 
@mikekgr 
can you run git commands in the console (cmd.exe) at all?
does this work? `git -C "E:\ARDUINO_ESP_DEV\arduino-IDE\hardware\esp8266com\esp8266\" rev-parse --short=8 HEAD`

mikekgr, 2019-05-03:
Dear @mcspr 
yes after applying/editing the file tools/makecorever.py and manually change the def generate(path, platform_path, git_ver="ffffffff", git_desc="unspecified"):
everything is working fine again. Sorry I can;t work with your git -C ... command

d-a-v, 2019-05-04:
> But does this mean git is not installed / not in PATH?

I indeed think it is because git is not in PATH when arduino ide is started so the default value 0xffff is taken in the python script. @mcspr your #6044 seems valid to me, thanks!

@mikekgr you wouldn't have found this bug if `git` command were in your system path.
Can you run `git` in a `cmd.exe` shell ?

mikekgr, 2019-05-04:
Dear @d-a-v 
Yes your assumption is right, I have git ( for Windows ) installed in a so called portable installation, ( all development tools, Arduino, Git Notepad++ and many more, are all installed in an external hard drive drive and such are portable to have everything with me, provided that I will have Windows OS in the Target machine ), that means in order to execute it, I need to provide the full path. That way, I can execute it from cmd and using shortcuts ( Windows shortcuts is something like the symbolic links in the Unix/Linux world.

Thanks and Best Regards,
Mike Kranidis

d-a-v, 2019-05-04:
@mikekgr Thanks for this report and @mcspr for the fix !

PR #6044 (2019-05-04): Fix default GIT_VER in makecorever.py
fixes #6043
closed 2019-05-04 by d-a-v