esp8266/Arduino #6049: detachInterrupt crashes after using argument in __attachInterruptArg()

dok-net, 2019-05-04:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			WEMOS D1 mini
- Core Version:			95fd7b55e462f470b4b892936d09a04646dec601
- Development Env:		Visual Micro IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			LOLIN(WEMOS) D1 R2 & mini
- Flash Mode:			[qio|dio|other]
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              921600

### Problem Description
Using detachInterrupt after __attachInterruptArg with a non-null argument, that's also not incidentally of type ArgStructure (as defined in FunctionalInterrupt.h), leads to a crash.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```extern "C" void ICACHE_RAM_ATTR __attachInterruptArg(uint8_t pin, void(*userFunc)(void*), void* fp, int mode);

class Interrupt
{
public:
	void ICACHE_RAM_ATTR isr() {}
	static void ICACHE_RAM_ATTR isr_static(void* self)
	{
		if (self) static_cast<Interrupt*>(self)->isr();
	}
};

Interrupt intr;

void setup()
{
	Serial.begin(9600);
	delay(1000);
	__attachInterruptArg(D6, Interrupt::isr_static, &intr, CHANGE);
	detachInterrupt(D6);
}

// Add the main program code into the continuous loop() function
void loop()
{

}
```

### Debug Messages

```
Exception (9):
epc1=0x4020328b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000017 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 01a0
3fffff70:  4010637c 3ffee248 402010f8 3fffefa0  
3fffff80:  401003e4 3ffee2b4 3ffee348 4010043c  
3fffff90:  3fffdad0 00000000 3ffee219 40201061  
3fffffa0:  feefeffe feefeffe feefeffe 402011a0  
3fffffb0:  feefeffe feefeffe 3ffe84f4 40100739  
<<<stack<<<
```

devyte, 2019-05-06:
@dok-net the __attachInterruptArg() call is not really exposed and hence not meant to be used by a user in its current form (notice the __ prefix).
That means that strictly speaking this is not a bug.
However, we do want to expose an attachInterruptArg() function that works correctly, and I think you have a PR about that, so I'm keeping this open as an enhancement.

devyte, 2019-05-06:
BTW, I invited you to our serial gitter channel. Please join so we can better coordinate efforts. If you didn't receive the mention let me know and I'll send it again.

dok-net, 2019-05-09:
@devyte Haven't seen any invite anywhere.

dok-net, 2019-05-12:
This issue is fixed by #6048.
The other PRs with similar concerns make more sense if that PR is accepted first - it fixes a real bug plus adds an API that I am lobbying for because I need/desire for EspSoftwareSerial: the proper Arduino.h-exposed attachInterruptArg().

PR #6048 (2019-07-04): Bugfix: attach interrupt (#6049)
Fixes: #6049 

- BUG FIX: __attachInterruptArg is known in the field, but it's use fails due to incorrect delete of arg pointer via cleanupFunctional. This PR guards against this case.
- Original author commented that ScheduledFunctions.(h|cpp) were added by mistake, and should be removed.
- ICACHE_RAM_ATTR was previously applied to attachInterrupt() / detachInterrupt() functions, but reviewer of #6003 commented that this can be reverted to save some precious IRAM, as attaching / detachInterrupt from inside ISR are corner cases.
- Properly expose attachInterrup [... cut]
closed 2019-07-04 by devyte