esp8266/Arduino #6050: Obsolete warning comment; copyright header missing

dok-net, 2019-05-05:
https://github.com/esp8266/Arduino/blob/95fd7b55e462f470b4b892936d09a04646dec601/cores/esp8266/Schedule.h#L9
@hreintke What is your opinion on:
- removing the warning?
- the proper copyright notice
- is the limit on SCHEDULED_FN_MAX_COUNT 32 still reasonable in the light of increased use for scheduled interrupts? Of the top of my mind, using scheduled interrupts for SW Serial, there are a lot of interrupts in a short interval of time :-)

devyte, 2019-05-06:
@dok-net the reason for this limit is that alloced nodes don't ever get freed. Instead, they get saved for recycling. 
The original code had a limit of 10 (if memory serves). When this code was updated, we decided on a limit of 32. The thinking was that, if you have to queue more than 32 callbacks before unwinding in the loop, you probably have bigger problems in your sketch that should be solved first.
So, before the limit is increased, I'd like to see a real(istic) example of how the current limit of 32 can be hit. Can you please provide one?

dok-net, 2019-12-02:
CoopTask can use cores/esp8266/Schedule for round-robin scheduling, but the device won't handle as many tasks to let the 32 limit become a problem.
@devyte There's still copyright notice.

PR #7653 (2020-10-15): Add Copyright notice to Schedule.h
#Fixes #6050
closed 2020-10-15 by d-a-v