esp8266/Arduino #6061: Potential issue with stdio in version 2.4.2

Muller59, 2019-05-09:
L.S.

when running the program below 

`#include <stdio.h>

char  _line1[] = "0-0:1.0.0(190205140804W)";
char *_line2 = "0-0:1.0.0(190205140804W)";

void setup() {
  Serial.begin(115200);
  printf("\nBooting ...\n");
  printf("PF-_line1 = %s \n", _line1);
  Serial.print("SP-_line1 = ");
  Serial.println(_line1);
  printf("PF-_line1 = %s\n", _line1);
  printf("Booting ...\n");
}

void loop() {
  // put your main code here, to run repeatedly:
}`

the output is

11:29:43.653 -> 
11:29:43.653 -> Booting ...
11:29:43.653 -> PF-_line1 = 0-0:1.0.0(190205140804W) 
11:29:43.653 -> SP-_line1 = 0-0:1.0.0(190205140804W)
11:29:43.653 -> PF-_line1 = 0-0:1.0.0(190205140804W)
11:29:43.653 -> Booting ...

when changing the 2nd printf statement from   printf("PF-_line1 = %s \n", _line1);
to   printf("**\n**PF-_line1 = %s \n", _line1); this is the output. (added a newline in front of text)

11:47:35.643 -> 
11:47:35.643 -> Booting ...
11:47:35.643 -> 
11:47:35.643 -> SP-_line1 = 0-0:1.0.0(190205140804W)
11:47:35.643 -> PF-_line1 = 0-0:1.0.0(190205140804W)
11:47:35.643 -> Booting ...

Note the MISSING 2nd Line

what gives?

Regards,
Rob

earlephilhower, 2019-05-10:
@Muller59, please try with latest GIT or 2.5.0 release.  Your code after the modification gives the proper output even with the `\n`:

````
Booting ...

PF-_line1 = 0-0:1.0.0(190205140804W) 
SP-_line1 = 0-0:1.0.0(190205140804W)
PF-_line1 = 0-0:1.0.0(190205140804W)
Booting ...
````

Muller59, 2019-05-10:
@earlephilhower 

yes, that solved the problem. I have in the meantime upgraded to the last arduino release 1.8.9, but never rechecked the issue.
Thanks for alerting me.

closed 2019-05-10 by Muller59