esp8266/Arduino #6064: [ESP8266 - AdcRead()]: Read ADC delivers max. value of 1024 instead 1023 - BUG?

hasenradball, 2019-05-10:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware: ESP-12
- Core Version:	2.5.0
- Development Env: Arduino IDE 1.8.9
- Operating System: Windows/Ubuntu
### Settings in IDE

- Module: Wemos D1 mini / Nodemcu
- Flash Mode: dio
- Flash Size:  4MB
- lwip Variant: not known
- Reset Method: nodemcu
- Flash Frequency: 40Mhz
- CPU Frequency: 80Mhz
- Upload Using: SERIAL
- Upload Speed:  115200

### Problem Description
If i read the adc pin on the ESP8266 or the Wemos D1 mini I get a max. value of 1024.
But the documentation says it ist a 10-bit adc with a range of 0...1023!
Why do the AdcRead() function deliver 1024!!??

Is this problem known?

5chufti, 2019-05-10:
overflow?

hasenradball, 2019-05-10:
Means if creater than 1 V at the adc delivers 1024?
I dont believe.

TD-er, 2019-05-10:
Does the `AdcRead()` function return the output of the ADC, or does it apply some calibration or maybe uses a lookup table?

hasenradball, 2019-05-10:
It is the direct output of adcRead()

TD-er, 2019-05-10:
@hasenradball I get that, but what does that function return?
Is it the direct output of the ADC, or is it doing something in software on that data?

devyte, 2019-05-10:
Our adc read function is a thin wrapper around the SDK function. I suspect inside the SDK there is a filter of some kind, i. e. it measures several times and calculates the result to return. If that SDK function returns 1024, there is nothing we can do.
There is a fast read function in the SDK, take a look at document 2c NONOS API reference from Espressif.
Closing for now.

closed 2019-05-10 by devyte

hasenradball, 2019-05-11:
Hi,
thank you!
Ok understood the resolution of the function `system_adc_read()` is 1/1024.
There was some misunderstanding.
Is the function `system_adc_read_fast()` availabe in the core 2.5.0?

d-a-v, 2019-05-11:
It is available in the firmware. If it is usable/working it could be wrapped in the ESP:: class.
The FW documentation states WiFi must be disabled when using it.

hasenradball, 2019-05-11:
Hi,
I also would like to use the following function to measure the `vdd33`.

How can I do to make it available in the ESP class?
Can you give me some hint how to manage this and which steps I have to do?

[image]
[image]

d-a-v, 2019-05-11:
You just need to declare the function in your sketch, and try it. If it works we can encapsulate it later.

```
extern "C" void system_adc_read_fast (uint16_t* adc_addr, uint16_t adc_num, uint8+t adc_clk_div);
```

jandrassy, 2019-05-11:
@hasenradball here https://arduino-esp8266.readthedocs.io/en/latest/reference.html#analog-input

hasenradball, 2019-05-11:
@jandrassy Thanks I tested it.
I get `Read Vcc: 3061`.
Is the unit in mV?
Do you know?

hasenradball, 2019-05-11:
Hi Is this right in this way?
I would have expected Values between 0...1024?!?

[image]

5chufti, 2019-05-12:
as you posted yourself 5 comments above, the unit is 1/1024V, giving 3,44 V in your example

TD-er, 2019-05-12:
@hasenradball If you set the `ADC_MODE(AD_VCC)`, it will use an internal voltage reference to compare the supplied Vcc.
I don't know what the range will be in that mode, but I suspect it can be over 3.3V, since the ESP is allowed to run up-to 3.6V.
So I wouldn't be surprised if it can then measure upto 4.096V for example.
The only thing you do know is that it will use 10-bit A/D convertor.
So if you run it for a while and try to detect the smallest step size in these outputs, you may get an idea of its range.
Although, I'm not sure about how it will convert these values internally or even if it is taking multiple samples. If for example the samples are averaged, the smallest step size can appear to be 1, but the step size of the true resolution isn't.

As an example the output of my car battery, monitored by the ESP8266's 10-bit Analog input:
[image]

As you can see, there are clear jumps depicting the true A/D resolution, but there are also slight variations in between those jumps. That's what averaging multiple samples does.

So in short, in ADC_VCC mode, it will use an internal circuit to measure the VCC of the ESP. The call to read the value will then return a value in mVolt scale.
Without this mode, it will just output the not scaled value of the A/D converter, but like stated before it is already an averaged value and perhaps also does some calculations on the samples.
Only the fast read function call should return the direct output of the A/D converter.

hasenradball, 2019-05-12:
Thanks for your comment and explanation.
I also worked on this topic yesterday.

I checked yesterday, when using ADC_VCC I had to divide by 1024 to get the value in V.

Nice image, which values does your voltage divider has?

Do you have am example using the fast adc function?

TD-er, 2019-05-12:
I use something like a 2M2 or 2M7 resistor on a Wemos D1 mini.
So it can handle up-to 24V. (Wemos D1 mini has a 100k resistor on the ADC input)

I never used the fast ADC function myself.