esp8266/Arduino #6068: package_esp8266com_index.json contains garbage

obor00, 2019-05-11:
https://arduino.esp8266.com/stable/package_esp8266com_index.json contains garbage at the start of the file, and Arduino IDE can't decode it anymore:

Dropping version 1.6.5-947-g39819f0
Dropping version 2.5.0-beta1
Dropping version 2.5.0-beta2
Dropping version 2.5.0-beta3
Dropping version 2.4.0-rc1
Dropping version 2.4.0-rc2
{
  "packages": [
    {
...

probonopd, 2019-05-11:
http://arduino.esp8266.com/stable/package_esp8266com_index.json gives in Firefox:

`SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data`

Reference: https://github.com/probonopd/WirelessPrinting/pull/118#issuecomment-491498268

d-a-v, 2019-05-11:
Will be fixed by #6070

BasEmmen, 2019-05-11:
A couple of minutes ago i lost the ESP8266 boards from the list.
Is this the result of the incorrect Json ?

Skipping contributed index file /Users/Me/Library/Arduino15/package_esp8266com_index.json, parsing error occured:
com.fasterxml.jackson.core.JsonParseException: Unrecognized token 'Dropping': was expecting ('true', 'false' or 'null')
 at [Source: (FileInputStream); line: 1, column: 10]

d-a-v, 2019-05-11:
We are fixing this as fast as we can. We are waiting for CI to approve the fix. 
https://travis-ci.org/esp8266/Arduino/builds/531090576?utm_source=github_status&utm_medium=notification

gnbl, 2019-05-11:
Workarounds?
- Using a fixed file locally does not work (protocol error).
- Using the template file https://github.com/esp8266/Arduino/blob/master/package/package_esp8266com_index.template.json does not seem to make esp8266 available in boards manager either.

obor00, 2019-05-11:
I also tried to copy the file without Garbage on my own web server, and got an error "Erreur durant le téléchargement de http://obordes.com/package_esp8266.com"

gnbl, 2019-05-11:
This seems to work

https://gist.githubusercontent.com/gnbl/74d47c8dcacdee21d6e1440b79993b85/raw/156c17655b644b29f480fbe3a36e0642a5dc9ca2/package_esp8266com_index.json

but the upgrading to / installing the latest version gives an error.

So I'm currently installing 2.5.0 (I think).

gnbl, 2019-05-11:
Garbage is still in file - how long until the fix is propagated?

probonopd, 2019-05-11:
Still getting `SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data`

d-a-v, 2019-05-11:
It should be OK now

gnbl, 2019-05-11:
For me it isn't - have you tried?
https://arduino.esp8266.com/stable/package_esp8266com_index.json

probonopd, 2019-05-11:
For me it is neither. Still getting `SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data` in Firefox. Even after refreshing.

d-a-v, 2019-05-11:
Right, it is still there. I pushed 2.5.1 tag that will regenerate the file.
We have to wait another CI (https://travis-ci.org/esp8266/Arduino/builds/531102618)

zeeshanyshaikh, 2019-05-11:
Thank god i reached here, i have uninstalled my Arduino. I thought i messed with my libraries.

morphal, 2019-05-11:
Travis build is done:
https://travis-ci.org/esp8266/Arduino/builds/531102618
But garbage is still in the file:
https://arduino.esp8266.com/stable/package_esp8266com_index.json

probonopd, 2019-05-11:
It is working again for me now :+1: thanks

obor00, 2019-05-11:
Works ok. Thanks :)

FrancescoTalotta, 2019-05-11:
Does not work for me :(

CombiesGit, 2019-05-11:
Same!
 Fehler beim Herunterladen von https://github.com/esp8266/Arduino/releases/download/2.5.1/esp8266-2.5.1.zip

Version 2.5.0 is now working

Tommrodrigues, 2019-05-11:
@FrancescoTalotta It works now if you select version `2.5.0` - seems there is a problem with the [release](https://github.com/esp8266/Arduino/releases) of `2.5.1` which is why you are getting that error. Hopefully `2.5.1` will be released again properly soon.

tobozo, 2019-07-04:
https://arduino.esp8266.com/stable/package_esp8266com_index.json still has some 404 errors, faulty URLs can be found by filtering with 'gb404fb9' term

[image]

d-a-v, 2019-07-04:
True, these are faulty:
```
http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9.tar.gz
http://arduino.esp8266.com/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz (appears twice)
http://arduino.esp8266.com/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
```

earlephilhower, 2019-07-04:
Having the same file 2x is fine in this file.  Just means the 32 and 64 bit versions of the OS will use the 32bit binaries linked.

These are all 2.0.0 release files from Espressif.  I'll add 2.0.0 to the drop list for the next release.

earlephilhower, 2019-07-04:
Oops, sorry, @d-a-v, deleted your comment instead of mine.

There is no actual problem here.  Those files are not referenced by the platforms at all.  We could delete them, but someone using any of the Arduino toolchains (get.py, boards-manager) will never access that "tools" key.

I'll throw out a PR to clean them on the next release, but there is no rush...

earlephilhower, 2019-07-16:
Bug is closed, but the actual file won't be regenerated until the next release so you won't see any updates to the .json just yet.

PR #6070 (2019-05-11): hide dropping message in json package file
fix #6068
PR #6258 (2019-07-16): Clean tools key of obsolete version on next release
Fixes #6068

Drop from the tools key all version:"1.20.0-26-gb404fb9" entries (which
were pre-2.0.0 and whose entry in platform versions was deleted on last
release).
commit 29bedfa8 (2019-07-16): Clean tools key of obsolete version on next release (#6258)

Fixes #6068

Drop from the tools key all version:"1.20.0-26-gb404fb9" entries (which
were pre-2.0.0 and whose entry in platform versions was deleted on last
release).
closed 2019-07-16 by earlephilhower