esp8266/Arduino #6073: Rel 2.5.1 CRC error

Testato, 2019-05-11:
Via BoardManager on last arduino Ide 1.8.9 on Win10, it is not possible install the version 2.5.1 of the core.
If you try to install the ver 2.5.0 it works regularly

`CRC doesn't match, file is corrupted.`

elgerg, 2019-05-11:
Same on 1.8.8

artafakharide, 2019-05-11:
Me 2

makermelissa, 2019-05-11:
I downloaded the file that the .json file points to, ran a manual CRC and am getting `SHA-256:77874cf43f9bb22024ed9f23296517854c8d4cd8352a5ded27b60c90d2af6873` instead of the listed `SHA-256:a433b764ca75d90c3df632949ff91777afe06e4e07a691cd68e7cf8c51eafc8c` which is causing the failure.

sanchosk, 2019-05-11:
Same here, tried to download from multiple workstations, I always end up with SHA256 
77874cf43f9bb22024ed9f23296517854c8d4cd8352a5ded27b60c90d2af6873

fape, 2019-05-11:
Same here. 1.8.9.
https://arduino.esp8266.com/stable/package_esp8266com_index.json
But hash is different what is in https://github.com/esp8266/Arduino/releases/download/2.5.1/package_esp8266com_index.json

[image]

Jason2866, 2019-05-11:
zip file is missing under the link 
https://github.com/esp8266/Arduino/releases/download/2.5.1/esp8266-2.5.1.zip

d-a-v, 2019-05-11:
Well this one has to be good, please retry

makermelissa, 2019-05-11:
Looks like it was fixed a few minutes ago and is working.

closed 2019-05-11 by devyte