esp8266/Arduino #6078: nodemcu gets the wakeup (the blu led flash lightly) but dont wake up

liorhal1, 2019-05-12:
my sketch run perfect on nodemcu v2 old , i connect the D0 to rst with wire and all work fine

but now i buy another nodemcu V2 and V3 , and they dont wake up 
i see in the serial that the wakeup start, and blu led flash abit but dont wake up

try with resistore and dont wake upm try D3 [GPIO 0 ] and nothing
if i connect the reset to GND it start to work
anybody can help me ??

TD-er, 2019-05-15:
Maybe related to the issue mentioned [here](https://www.esp8266.com/viewtopic.php?f=6&t=17975) ?

devyte, 2019-05-15:
If it works with one and not with another, it strongly hints at hw.
So far, the suspected culprit is the pulse delay circuit. I suggest looking at that, or getting a non-clone module. 
Closing due to not a core issue.

closed 2019-05-15 by devyte