esp8266/Arduino #6079: Using program memory causes a reset

TMRh20, 2019-05-12:
### Basic Infos

- [+ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [+ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [+ ] I have tested that the issue is present in current master branch (aka latest git).
- [ +] I have searched the issue tracker for a similar issue.
- [ Could not get decoder working] If there is a stack dump, I have decoded it.
- [ +] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE
- Operating System:		[Windows|

### Settings in IDE

- Module:  			|Nodemcu|
- Flash Mode:			??
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[?nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Using program memory causes the device to reset.

The problem exists with my RF24 library [lines 579 to 582](https://github.com/nRF24/RF24/blob/master/RF24.cpp#L579)

It can be recreated easily using the example code at https://www.arduino.cc/reference/en/language/variables/utilities/progmem/

which I have included below
``` 
#include <avr/pgmspace.h>
const char string_0[] PROGMEM = "String 0"; // "String 0" etc are strings to store - change to suit.
const char string_1[] PROGMEM = "String 1";
const char string_2[] PROGMEM = "String 2";
const char string_3[] PROGMEM = "String 3";
const char string_4[] PROGMEM = "String 4";
const char string_5[] PROGMEM = "String 5";

// Then set up a table to refer to your strings.

const char *const string_table[] PROGMEM = {string_0, string_1, string_2, string_3, string_4, string_5};

char buffer[30];  // make sure this is large enough for the largest string it must hold

void setup() {
  Serial.begin(9600);
  while (!Serial);  // wait for serial port to connect. Needed for native USB
  Serial.println("OK");
}

void loop() {
  /* Using the string table in program memory requires the use of special functions to retrieve the data.
     The strcpy_P function copies a string from program space to a string in RAM ("buffer").
     Make sure your receiving string in RAM  is large enough to hold whatever
     you are retrieving from program space. */

  for (int i = 0; i < 6; i++) {
    strcpy_P(buffer, (char *)pgm_read_word(&(string_table[i])));  // Necessary casts and dereferencing, just copy.
    Serial.println(buffer);
    delay(500);
  }
}
```

### Debug Messages

```
Debug messages go here
```

earlephilhower, 2019-05-12:
Try compiling with warnings enabled.  GCC is complaining about your casts which are not legal:

````
In file included from /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include/string.h:163:0,
                 from /home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:33,
                 from sketch/sketch_may12a.ino.cpp:1:
/home/earle/Arduino/sketch_may12a/sketch_may12a.ino: In function 'void loop()':
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:113:57: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
     #define pgm_read_word(a)   pgm_read_word_unaligned(a)
                                                         ^
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/string.h:32:57: note: in definition of macro 'strcpy_P'
 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)
                                                         ^
/home/earle/Arduino/sketch_may12a/sketch_may12a.ino:31:30: note: in expansion of macro 'pgm_read_word'
     strcpy_P(buffer, (char *)pgm_read_word(&(string_table[i])));  // Necessary casts and dereferencing, just copy.
                              ^
Sketch uses 262468 bytes (25%) of program storage space. Maximum is 1044464 bytes.
Global variables use 26692 bytes (32%) of dynamic memory, leaving 55228 bytes for local variables. Maximum is 81920 bytes.
````

Basically, you're reading a 16-bit value (an Arduino "word") and trying to use it as a pointer.  Pointers are 32-bits on the 8266.

Use `pgm_read_ptr` instead and it works fine and is platform agnostic.

closed 2019-05-12 by earlephilhower

TMRh20, 2019-05-12:
Sorry if this is a non-issue, but using pgm_read_ptr works the same on AVR devices, but on ESP8266 I get an error compiling:

```
C:\Users\tmrh20\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type

     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))

                                                                                 ^

C:\Users\tmrh20\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/string.h:32:57: note: in definition of macro 'strcpy_P'

 #define strcpy_P(dest, src)          strncpy_P((dest), (src), SIZE_IRRELEVANT)

                                                         ^

C:\Users\tmrh20\AppData\Local\Temp\arduino_modified_sketch_5459\sketch_may12a.ino:31:30: note: in expansion of macro 'pgm_read_ptr'

     strcpy_P(buffer, (char *)pgm_read_ptr(&(string_table[i])));  // Necessary casts and dereferencing, just copy.
```

earlephilhower, 2019-05-12:
You're using an older release.  Try 2.5.1 which just came out and includes a fix (about 4 months old IIRC) for this.