esp8266/Arduino #6081: In new SD library several functions are now disabled

luc-github, 2019-05-13:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			latest git / 2.5.1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:  v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

As requested in gitter here are the disabled functions:

```
uint8_t type() {
        return 0;//card.type();
    }
    uint8_t fatType() {
        return 0;//volume.fatType();
    }
    size_t blocksPerCluster() {
        return 0;//volume.blocksPerCluster();
    }
    size_t totalClusters() {
        return 0;//volume.clusterCount();
    }
    size_t totalBlocks() {
        return 0;//(totalClusters() / blocksPerCluster());
    }
    size_t clusterSize() {
        return 0;//blocksPerCluster() * blockSize();
    }
    size_t size() {
        return 0;//(clusterSize() * totalClusters());
    }
```

earlephilhower, 2019-05-14:
@luc-github I did some basis sanity, and it looks good to me.  Give the PR a go and chime in if you have any issues.

luc-github, 2019-05-14:
Ok thank you
I will test and feedback

PR #6092 (2019-05-19): Re-add original SD FAT info access methods
Fixes #6081

The SD rewrite blanked out some of the internal FAT info.. Restore the
function calls and return proper values.
commit b5560759 (2019-05-19): Re-add original SD FAT info access methods (#6092)

Fixes #6081

The SD rewrite blanked out some of the internal FAT info.. Restore the
function calls and return proper values.

Because size() is used in many printf()s, we can't just change its
return type to uint64.  Instead, when size is > size-max warn.

Add SD.size64 which can be used by new apps who care about >4GB cards.
Prints a warning if debugging enabled
closed 2019-05-19 by earlephilhower