esp8266/Arduino #6085: Increased memory and code volume

laercionit, 2019-05-13:
Hello good day.

I used in my project the core 2.4.2
55% code usage
51% of memory usage.

When upgrading to core 2.5.1 I was amazed at the impact on code and memory volume.
83% code usage
76% of memory usage

Another unfortunately could not upgrade because my project will run out of resources on this core 2.5.1 hardware.

Jason2866, 2019-05-14:
Exceptions in code enabled? If yes turn it off via Arduino IDE (Boards Options)
This is new in 2.5.x   Core 2.4.2 didnt have this option to include.

janzieg, 2019-05-17:
I´m facing the same issue with free heap size!
My sketch used to have about 25k-30k free heap size with 2.5.0. 
With 2.5.1 I´m having between 5K-10K free heap size so many functions (e.g. webserver request) stopped to work or hang - i guess because of not having enough free heap ...?

laercionit, 2019-05-17:
I disabled everything I could, but it's still impossible to run the project ... the volume of global variables has increased a lot ....

devyte, 2019-05-17:
are you using wifi secure client/server?

laercionit, 2019-05-17:
No, I use client and server without SSL.

devyte, 2019-05-17:
Do you have debug disabled for your build?
Where are your vtables?
Are exceptions disabled?
Have you dumped the symbols to check the difference between 2.4.x and 2.5.x?

You ignored the issue template, so it's hard to help you. If this were widespread it would have been noticed and reported by now. I certainly don't see it in my own project. It could be something specific to your code, but without even a MCVE I'm inclined to just close this.

laercionit, 2019-05-18:
Hello, first of all I'm sorry for not giving more details, I just tried to share a data and know if other users had the same problem, because in my opinion this is not a bug.

**Before:**
CORE 2.4.2
Sketch uses 583396 bytes (55%) of program storage space. Maximum is 1044464 bytes.
Global variables use 42424 bytes (51%) of dynamic memory, leaving 39496 bytes for local variables. Maximum is 81920 bytes.
[image]

**After:**
CORE 2.5.1
Sketch uses 615176 bytes (58%) of program storage space. Maximum is 1044464 bytes.
Global variables use 62952 bytes (76%) of dynamic memory, leaving 18968 bytes for local variables. Maximum is 81920 bytes.

[image]

earlephilhower, 2019-05-18:
@laercionit, without code or a MAP there's nothing we can do to help you.  I suggest popping to the 8266 web forum and see if someone there can help you.  

https://www.esp8266.com/

Closing.

closed 2019-05-18 by earlephilhower

janzieg, 2019-05-21:
@laercionit Are you using SoftwareSerial ? I startet to comment out stuff in my sketch to test what causes strong heap usage - and it seems to be SoftwareSerial.

@earlephilhower 
The release notes say it was updated in 2.5.1 - i guess there the problem might be related to that.

Having SoftwareSerial instrances is enough to show the problem. I try to create  a simple sketch an reopen this..

earlephilhower, 2019-05-21:
@janzieg thanks, a MCVE would give us something to actually look at.

That said, though, SW Serial is an external library so you might want to move to there to get the proper eyes on it.

https://github.com/plerup/espsoftwareserial

laercionit, 2019-05-21:
Hello @janzieg
I use it indirectly through the ESPNexUpload.h Library
It has reference to SoftwareSerial.h, but it did not make any difference in the project with or without this library.

** CORE 2.4.2 **
- 55% of program storage space
- 51% dynamic memory

** CORE 2.5.1 **
- 58% of program storage space (Acceptable increase)
- 76% dynamic memory (significant increase, making project unfeasible)

Librarys:
#include <WiFi.h>
#include <Wire.h>
#include <Ticker.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>
#include "Arduino.h"
#include <EDB.h>
#include <SPI.h>
#include <SD.h>
#include <EEPROM.h>
#include <ESP8266HTTPClient.h>
#include <FS.h>
#include <pcf8574_esp.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266SSDP.h>
#include <ESP8266mDNS.h>
#include <ESP8266NetBIOS.h>
#include <ESP8266LLMNR.h>
#include <DNSServer.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>
#include <Update.h>
#include <WiFiMulti.h>
#include "ESP8266TrueRandom.h"
#include "NexUpload.h"

laercionit, 2019-05-21:
I did a test with libs, I do not think that's the problem.
I am separating the project code so that I can evaluate and guide it better. It may be some specific reference.

janzieg, 2019-05-21:
@laercionit I also think I´m having a different issue. My concern is about different behavior regarding free heap memory size during runtime and not about sketch storage space or global variable usage after compiling (I don´t see any big difference here for my sketch with 2.5.1/2.5.2 compared to 2.5.0). 

@earlephilhower I´m going to open an issue at https://github.com/plerup/espsoftwareserial when my test sketch is up.

Anyway, I´m going to link it...

laercionit, 2019-05-21:
My problem is summarized in the use of dynamic memory, this impacts directly on my project.
I can even understand the increase in space of the program, after all the core has undergone a major change, but I can not understand the increase in so much dynamic memory.

The problem appears in version 2.5.1, is there an explanation for this?

2.4.2
Sketch uses 583464 bytes (55%) of program storage space. Maximum is 1044464 bytes.
Global variables use 42180 bytes (51%) of dynamic memory, leaving 39740 bytes for local variables. Maximum is 81920 bytes.

2.5.0 (A significant reduction, without any code change, just the core)
Sketch uses 615588 bytes (58%) of program storage space. Maximum is 1044464 bytes.
Global variables use 40732 bytes (49%) of dynamic memory, leaving 41188 bytes for local variables. Maximum is 81920 bytes.

2.5.1 (PROBLEM)
Sketch uses 615348 bytes (58%) of program storage space. Maximum is 1044464 bytes.
Global variables use 62796 bytes (76%) of dynamic memory, leaving 19124 bytes for local variables. Maximum is 81920 bytes.

2.5.2 (PROBLEM)
Sketch uses 615380 bytes (58%) of program storage space. Maximum is 1044464 bytes.
Global variables use 62796 bytes (76%) of dynamic memory, leaving 19124 bytes for local variables. Maximum is 81920 bytes.

earlephilhower, 2019-05-23:
@laercionit, as I mentioned before, unless you can supply something concrete then there's nothing anyone but you can do.

Either make and post a smaller sketch which shows a large change in heap between versions, post two ELF files from different versions (not the *.bin but the *.elf with symbol tables we can compare), or dig into it yourself.  We can't read minds, and we can't debug without code showing a problem.

laercionit, 2019-05-23:
@earlephilhower 
I understood, thanks for the guidance:

Core 2.4.2 ELF: 
[2.4.2.zip](https://github.com/esp8266/Arduino/files/3213354/2.4.2.zip)

Core 2.5.2 ELF: [2.5.2.zip](https://github.com/esp8266/Arduino/files/3213381/2.5.2.zip)

earlephilhower, 2019-05-23:
RODATA (static variables with non-0 definitions) has exploded:
````
earle@server:~/Desktop/p1$ readelf -e 2.4.2.elf  | head -40
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Tensilica Xtensa Processor
  Version:                           0x1
  Entry point address:               0x401065d0
  Start of program headers:          52 (bytes into file)
  Start of section headers:          7753324 (bytes into file)
  Flags:                             0x300
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         5
  Size of section headers:           40 (bytes)
  Number of section headers:         590
  Section header string table index: 587

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .data             PROGBITS        3ffe8000 0000e0 001860 00  WA  0   0 16
  [ 2] .noinit           PROGBITS        3ffe9860 001940 000004 00  WA  0   0  4
  [ 3] .irom0.text       PROGBITS        40201010 00a160 0847c0 00  AX  0   0 16
  [ 4] .text             PROGBITS        40100000 0031ec 006f6c 00  AX  0   0  4
  [ 5] .rodata           PROGBITS        3ffe9870 001950 00189c 00  WA  0   0 16
  [ 6] .bss              NOBITS          3ffeb110 0031f0 0073c8 00  WA  0   0 16
  [ 7] .debug_frame      PROGBITS        00000000 08e920 02035c 00      0   0  4
  [ 8] .debug_info       PROGBITS        00000000 0aec7c 390f99 00      0   0  1
  [ 9] .debug_abbrev     PROGBITS        00000000 43fc15 054437 00      0   0  1
  [10] .debug_loc        PROGBITS        00000000 49404c 11981a 00      0   0  1
  [11] .debug_aranges    PROGBITS        00000000 5ad868 00a088 00      0   0  8
  [12] .debug_ranges     PROGBITS        00000000 5b78f0 021aa0 00      0   0  1
  [13] .debug_line       PROGBITS        00000000 5d9390 0ec6a1 00      0   0  1
  [14] .debug_str        PROGBITS        00000000 6c5a31 08e372 01  MS  0   0  1
  [15] .comment          PROGBITS        00000000 753da3 00156a 00      0   0  1
  [16] .xtensa.info      NOTE            00000000 75530d 000038 00      0   0  1
earle@server:~/Desktop/p1$ readelf -e 2.5.2.elf  | head -40
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Tensilica Xtensa Processor
  Version:                           0x1
  Entry point address:               0x401001d0
  Start of program headers:          52 (bytes into file)
  Start of section headers:          6783176 (bytes into file)
  Flags:                             0x300
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         5
  Size of section headers:           40 (bytes)
  Number of section headers:         822
  Section header string table index: 821

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .data             PROGBITS        3ffe8000 001000 00178c 00  WA  0   0 16
  [ 2] .noinit           PROGBITS        3ffe978c 00278c 000004 00  WA  0   0  4
  [ 3] .text             PROGBITS        40100000 00a000 000144 00  AX  0   0  4
  [ 4] .irom0.text       PROGBITS        40201010 012010 086ad0 00  AX  0   0 16
  [ 5] .text1            PROGBITS        40100148 00a148 0077d8 00  AX  0   0  8
  [ 6] .rodata           PROGBITS        3ffe9790 002790 0068e8 00  WA  0   0 16
  [ 7] .bss              NOBITS          3fff0078 009078 0074d8 00  WA  0   0 16
  [ 8] .debug_frame      PROGBITS        00000000 098ae0 01e220 00      0   0  4
  [ 9] .debug_info       PROGBITS        00000000 0b6d00 3061f7 00      0   0  1
  [10] .debug_abbrev     PROGBITS        00000000 3bcef7 03e499 00      0   0  1
  [11] .debug_loc        PROGBITS        00000000 3fb390 0c49ba 00      0   0  1
  [12] .debug_aranges    PROGBITS        00000000 4bfd50 007f60 00      0   0  8
  [13] .debug_ranges     PROGBITS        00000000 4c7cb0 019840 00      0   0  1
  [14] .debug_line       PROGBITS        00000000 4e14f0 0adcdf 00      0   0  1
  [15] .debug_str        PROGBITS        00000000 58f1cf 0972c9 01  MS  0   0  1
  [16] .comment          PROGBITS        00000000 626498 0013c9 00      0   0  1
````

It's grown from 0x189c to 0x68e8 bytes.

earlephilhower, 2019-05-23:
But names symbols in RODATA actually **shrunk** in size.  This means something else is in RODATA now that wasn't there before, and it's auto-generated somewhere in the compile process:

````
earle@server:~/Desktop/p1$ grep "^\.rodata" 242 | sort -k2 
.rodata 00000000 .rodata
.rodata 00000002 memp_ARP_QUEUE
.rodata 00000002 memp_IGMP_GROUP
.rodata 00000002 memp_PBUF
.rodata 00000002 memp_PBUF_POOL
.rodata 00000002 memp_RAW_PCB
.rodata 00000002 memp_SYS_TIMEOUT
.rodata 00000002 memp_TCP_PCB
.rodata 00000002 memp_TCP_PCB_LISTEN
.rodata 00000002 memp_TCP_SEG
.rodata 00000002 memp_UDP_PCB
.rodata 00000004 dns_mquery_v4group
.rodata 00000004 _global_impure_ptr
.rodata 00000004 ip_addr_any
.rodata 00000004 ip_addr_broadcast
.rodata 00000004 _ZL11NetworkAPIP
.rodata 00000004 _ZL13NetworkAPMask
.rodata 00000005 CSWTCH$58
.rodata 00000005 dhcp_discover_request_options
.rodata 00000005 _ZZN14SPIFFSFileImpl4readEPhjE8__func__
.rodata 00000005 _ZZN14SPIFFSFileImpl4seekEjN2fs8SeekModeEE8__func__
.rodata 00000005 _ZZNK14SPIFFSFileImpl4nameEvE8__func__
.rodata 00000005 _ZZNK14SPIFFSFileImpl4sizeEvE8__func__
.rodata 00000006 CSWTCH$75
.rodata 00000006 CSWTCH$87
.rodata 00000006 ethbroadcast
.rodata 00000006 ethbroadcast_LWIP2
.rodata 00000006 ethzero
.rodata 00000006 _ZZN14SPIFFSFileImpl5closeEvE8__func__
.rodata 00000006 _ZZN14SPIFFSFileImpl5flushEvE8__func__
.rodata 00000006 _ZZN14SPIFFSFileImpl5writeEPKhjE8__func__
.rodata 00000007 tcp_persist_backoff
.rodata 00000008 _ZZ7__yieldE8__func__
.rodata 00000009 _ZZNK14SPIFFSFileImpl8_getStatEvE8__func__
.rodata 00000009 _ZZNK14SPIFFSFileImpl8positionEvE8__func__
.rodata 0000000a _ZZL9loop_taskP11ETSEventTagE8__func__
.rodata 0000000c p05$2603
.rodata 0000000d tcp_backoff
.rodata 0000000f SDK_VERSION
.rodata 00000010 bp
.rodata 00000010 CSWTCH$17
.rodata 00000010 dp_h
.rodata 00000010 dp_l
.rodata 00000010 halF
.rodata 00000010 ln2HI
.rodata 00000010 ln2LO
.rodata 00000010 tcp_pcb_lists
.rodata 00000010 TWO52
.rodata 00000011 CSWTCH$11
.rodata 00000011 CSWTCH$15
.rodata 00000011 CSWTCH$9
.rodata 00000013 _ZZ18__cxa_pure_virtualE8__func__
.rodata 00000014 fpi$2568
.rodata 00000014 fpinan$2604
.rodata 0000001a _ZZSt25__throw_bad_function_callvE8__func__
.rodata 0000001e _ZZN10UdpContext4seekEjE19__PRETTY_FUNCTION__
.rodata 00000020 __sf_fake_stderr
.rodata 00000020 __sf_fake_stdin
.rodata 00000020 __sf_fake_stdout
.rodata 00000028 CSWTCH$86
.rodata 00000028 memp_pools
.rodata 00000028 __mprec_bigtens
.rodata 00000028 __mprec_tinytens
.rodata 00000028 tinytens
.rodata 00000030 lwip_cyclic_timers
.rodata 00000031 _ZZN13ClientContext10_connectedEP7tcp_pcblE19__PRETTY_FUNCTION__
.rodata 00000036 _ZZN13ClientContext18_write_from_sourceEP10DataSourceE19__PRETTY_FUNCTION__
.rodata 0000003c _ZZN16BufferDataSource10get_bufferEjE19__PRETTY_FUNCTION__
.rodata 00000046 _ZZN16BufferDataSource14release_bufferEPKhjE19__PRETTY_FUNCTION__
.rodata 00000054 _ZL8Weekdays
.rodata 0000008d _ZZN24BufferedStreamDataSourceI6StreamE10get_bufferEjE19__PRETTY_FUNCTION__
.rodata 00000094 _ZZN24BufferedStreamDataSourceI13ProgmemStreamE10get_bufferEjE19__PRETTY_FUNCTION__
.rodata 00000097 _ZZN24BufferedStreamDataSourceI6StreamE14release_bufferEPKhjE19__PRETTY_FUNCTION__
.rodata 0000009e _ZZN24BufferedStreamDataSourceI13ProgmemStreamE14release_bufferEPKhjE19__PRETTY_FUNCTION__
.rodata 000000a1 _ZL5Units
.rodata 000000c8 __mprec_tens
.rodata 00000101 _ctype_
earle@server:~/Desktop/p1$ grep "^\.rodata" 252 | sort -k2 
.rodata 00000000 .rodata
.rodata 00000001 _ZSt7nothrow
.rodata 00000002 memp_ARP_QUEUE
.rodata 00000002 memp_IGMP_GROUP
.rodata 00000002 memp_PBUF
.rodata 00000002 memp_PBUF_POOL
.rodata 00000002 memp_RAW_PCB
.rodata 00000002 memp_REASSDATA
.rodata 00000002 memp_SYS_TIMEOUT
.rodata 00000002 memp_TCP_PCB
.rodata 00000002 memp_TCP_PCB_LISTEN
.rodata 00000002 memp_TCP_SEG
.rodata 00000002 memp_UDP_PCB
.rodata 00000004 dns_mquery_v4group
.rodata 00000004 _global_impure_ptr
.rodata 00000004 ip_addr_any
.rodata 00000004 ip_addr_broadcast
.rodata 00000004 _ZL11NetworkAPIP
.rodata 00000004 _ZL13NetworkAPMask
.rodata 00000005 CSWTCH$58
.rodata 00000005 dhcp_discover_request_options
.rodata 00000006 CSWTCH$76
.rodata 00000006 CSWTCH$98
.rodata 00000006 ethbroadcast
.rodata 00000006 ethbroadcast_LWIP2
.rodata 00000006 ethzero
.rodata 00000007 tcp_persist_backoff
.rodata 0000000d tcp_backoff
.rodata 0000000f SDK_VERSION
.rodata 00000010 bp
.rodata 00000010 CSWTCH$17
.rodata 00000010 dp_h
.rodata 00000010 dp_l
.rodata 00000010 halF
.rodata 00000010 ln2HI
.rodata 00000010 ln2LO
.rodata 00000010 tcp_pcb_lists
.rodata 00000010 TWO52
.rodata 00000011 CSWTCH$12
.rodata 00000011 CSWTCH$15
.rodata 00000011 CSWTCH$9
.rodata 00000028 CSWTCH$103
.rodata 0000002c memp_pools
.rodata 00000040 lwip_cyclic_timers
.rodata 00000054 _ZL8Weekdays
.rodata 000000a1 _ZL5Units
````

earlephilhower, 2019-05-23:
Your 2.5.2 RODATA has tons more strings in it than 2.4.2.  For further stuff, you're going to need to look at the source code.  Maybe a PSTR or something is getting lost or undefined.

Check using `objdump -s -j .rodata 2.4.2.elf`

In 2.5.2 here are some pointers you can search your code for.  Things like "Reset system settings to factory default" are not in RODATA in 2.4.2 but are in RODATA (aka RAM) in 2.5.2 in your build.

````
earle@server:~/Desktop/p1$ objdump -s -j .rodata 2.5.2.elf  

2.5.2.elf:     file format elf32-little

Contents of section .rodata:
 3ffe9790 322e322e 31286366 64343866 33290000  2.2.1(cfd48f3)..
 3ffe97a0 bdb7fe3f 3cb8fe3f 312e302e 30000000  ...?<..?1.0.0...
 3ffe97b0 00000061 646d696e 00000000 00006171  ...admin......aq
 3ffe97c0 75617269 6e6f0000 00000000 00000000  uarino..........
 3ffe97d0 00000000 00000000 00000000 00000000  ................
 3ffe97e0 00000000 00000000 00000000 00000000  ................
 3ffe97f0 00000000 00000000 00000000 00000000  ................
 3ffe9800 00000000 00000000 02000000 01000000  ................
 3ffe9810 00000100 0100044d 51545404 00c20100  .......MQTT.....
 3ffe9820 004b0000 80250000 00e10000 00960000  .K...%..........
 3ffe9830 c0120000 60090000 0408100c 0408100c  ....`...........
 3ffe9840 64000000 14084000 dd070050 f2020001  d.....@....P....
 3ffe9850 00000000 00000000 00000000 00000000  ................
 3ffe9860 00000000 00000000 00000000 00000000  ................
 3ffe9870 00000000 00000000 00000000 00000000  ................
 3ffe9880 763c7935 ef39ea3d 763c7935 ef39eabd  v<y5.9.=v<y5.9..
 3ffe9890 0000e0fe 422ee63f 0000e0fe 422ee6bf  ....B..?....B...
 3ffe98a0 00000000 0000e03f 00000000 0000e0bf  .......?........
 3ffe98b0 00000000 00000000 06d0cf43 ebfd4c3e  ...........C..L>
 3ffe98c0 00000000 00000000 00000040 03b8e23f  ...........@...?
 3ffe98d0 00000000 0000f03f 00000000 0000f83f  .......?.......?
 3ffe98e0 00000000 00003043 00000000 000030c3  ......0C......0.
 3ffe98f0 0483fe3f 00000000 00002041 0a456e64  ...?...... A.End
 3ffe9900 0050726f 67726573 733a2025 7525250d  .Progress: %u%%.
 3ffe9910 00457272 6f725b25 755d3a20 00417574  .Error[%u]: .Aut
 3ffe9920 68204661 696c6564 00426567 696e2046  h Failed.Begin F
 3ffe9930 61696c65 6400436f 6e6e6563 74204661  ailed.Connect Fa
 3ffe9940 696c6564 00526563 65697665 20466169  iled.Receive Fai
 3ffe9950 6c656400 456e6420 4661696c 65640073  led.End Failed.s
 3ffe9960 6b657463 68006669 6c657379 7374656d  ketch.filesystem
 3ffe9970 00537461 72742075 70646174 696e6720  .Start updating 
 3ffe9980 00772b00 612b0068 7474703a 2f2f7365  .w+.a+.http://se
 3ffe9990 72766963 652e6171 75617269 6e6f2e63  rvice.aquarino.c
 3ffe99a0 6f6d2e62 722f6d61 696c2f00 46696c65  om.br/mail/.File
 3ffe99b0 73534443 6c6f7365 3a20436c 6f736520  sSDClose: Close 
 3ffe99c0 46696c65 205b005d 0046696c 65735344  File [.].FilesSD
 3ffe99d0 436c6f73 653a2054 52592043 6c6f7365  Close: TRY Close
 3ffe99e0 2046696c 65205b00 46696c65 73534445   File [.FilesSDE
 3ffe99f0 78697374 733a2046 696c6520 6f722050  xists: File or P
 3ffe9a00 61746820 65786973 7473205b 0046696c  ath exists [.Fil
 3ffe9a10 65735344 45786973 74733a20 46696c65  esSDExists: File
 3ffe9a20 206f7220 50617468 20646f65 73206e6f   or Path does no
 3ffe9a30 74206578 69737420 5b004669 6c657353  t exist [.FilesS
 3ffe9a40 4452656d 6f76653a 2044656c 6574696e  DRemove: Deletin
 3ffe9a50 67204669 6c65205b 0046696c 65735344  g File [.FilesSD
 3ffe9a60 52656d6f 76653a20 43616e6e 6f742064  Remove: Cannot d
 3ffe9a70 656c6574 65204669 6c65205b 0046696c  elete File [.Fil
 3ffe9a80 65735344 4d6b6469 723a2043 72656174  esSDMkdir: Creat
 3ffe9a90 6520466f 6c646572 205b0046 696c6573  e Folder [.Files
 3ffe9aa0 5344496e 69743a20 53442066 61696c00  SDInit: SD fail.
 3ffe9ab0 46696c65 73534449 6e69743a 20534420  FilesSDInit: SD 
 3ffe9ac0 52656164 7921002f 44420046 696c6573  Ready!./DB.Files
 3ffe9ad0 4653436c 6f73653a 20436c6f 73652046  FSClose: Close F
 3ffe9ae0 696c6520 5b004669 6c657346 53436c6f  ile [.FilesFSClo
 3ffe9af0 73653a20 54525920 436c6f73 65204669  se: TRY Close Fi
 3ffe9b00 6c65205b 0046696c 65734653 45786973  le [.FilesFSExis
 3ffe9b10 74733a20 46696c65 206f7220 50617468  ts: File or Path
 3ffe9b20 20657869 73747320 5b004669 6c657346   exists [.FilesF
[... 81 lines cut ...]
 3ffea040 64656c3d 41717561 72696e6f 203e3e20  del=Aquarino >> 
 3ffea050 00203e3e 20547970 653d5341 54203e3e  . >> Type=SAT >>
 3ffea060 204d6f64 656c3d41 71756172 696e6f20   Model=Aquarino 
 3ffea070 3e3e2056 616c7565 3d466169 6c656400  >> Value=Failed.
 3ffea080 203e3e20 54797065 3d504820 3e3e204d   >> Type=PH >> M
 3ffea090 6f64656c 3d417175 6172696e 6f203e3e  odel=Aquarino >>
 3ffea0a0 2000203e 3e205479 70653d50 48203e3e   . >> Type=PH >>
 3ffea0b0 204d6f64 656c3d41 71756172 696e6f20   Model=Aquarino 
 3ffea0c0 3e3e2056 616c7565 3d466169 6c656400  >> Value=Failed.
 3ffea0d0 50483153 6c6f7065 3a200050 4832536c  PH1Slope: .PH2Sl
 3ffea0e0 6f70653a 2000203e 3e205479 70653d4f  ope: . >> Type=O
 3ffea0f0 5250203e 3e204d6f 64656c3d 41717561  RP >> Model=Aqua
 3ffea100 72696e6f 203e3e20 00203e3e 20547970  rino >> . >> Typ
 3ffea110 653d4f52 50203e3e 204d6f64 656c3d41  e=ORP >> Model=A
 3ffea120 71756172 696e6f20 3e3e2056 616c7565  quarino >> Value
 3ffea130 3d466169 6c656400 203e3e20 54797065  =Failed. >> Type
 3ffea140 3d454320 3e3e204d 6f64656c 3d417175  =EC >> Model=Aqu
 3ffea150 6172696e 6f203e3e 2000203e 3e205479  arino >> . >> Ty
 3ffea160 70653d45 43203e3e 204d6f64 656c3d41  pe=EC >> Model=A
 3ffea170 71756172 696e6f20 3e3e2056 616c7565  quarino >> Value
 3ffea180 3d466169 6c656400 436c6f73 65640053  =Failed.Closed.S
 3ffea190 61666500 416c6572 74005570 00446f77  afe.Alert.Up.Dow
 3ffea1a0 6e005374 61626c65 004e6574 776f726b  n.Stable.Network
....
````

laercionit, 2019-05-23:
@earlephilhower , 

Hello, thanks.
But even though it's the same code without any change, I believe the new core 2.5.2 will do something different so that it has had this change in dynamic memory. Any guidance on what I should look for in the code to improve?

earlephilhower, 2019-05-23:
It's not really the same code, that's the problem, since the core has been updated significantly since the old days of 2.4.2. 

You need to find where those strings are defined and see what's going on there.  There could be a `#define` that you're using that's not present any more, or the strings could be in messages that were suppressed before, or something else.

devyte, 2019-05-23:
A lot of those strings look like debug info, I suggest checking that first, i.e. is debug really disabled, is are you sure yky don't have some define somewhere that could enable it, etc.

laercionit, 2019-05-23:
From what I've analyzed, all debug texts were coming in through Flash in 2.4.2

Now with 2.5.2 for some reason it is counting as dynamic memory.
See the code below, commenting on the dynamic memory is reducing, but this behavior did not exist in core 2.4.2.

if (DebugCheck(Verbose)) DebugSerial.println (String(F("FilesFSExists: File or Path exists [")) + String(Filename) + String(F("]")));

earlephilhower, 2019-05-23:
I think you're redefining `F()` or something somewhere in your code.  Your example works in the following sketch just fine, placing the text in flash under 2.5.2 (git head, which is the same):
````
void setup() {
  const char *Filename =  "abc";
  Serial.println (String(F("FilesFSExists: File or Path exists [")) + String(Filename) + String(F("]")));
}

void loop() {
}
````

````
earle@server:~/Arduino/hardware/esp8266com/esp8266/tools/sdk/lib$ objdump -s -j .rodata /tmp/arduino_build_263740/sketch_may23a.ino.elf  

/tmp/arduino_build_263740/sketch_may23a.ino.elf:     file format elf32-little

Contents of section .rodata:
 3ffe8510 322e322e 31286366 64343866 33290000  2.2.1(cfd48f3)..
 3ffe8520 0408100c 0408100c 64000000 14084000  ........d.....@.
 3ffe8530 dd070050 f2020001 00000000 00000000  ...P............
 3ffe8540 00000000 00000000 00000000 00000000  ................
 3ffe8550 00000000 00000000 00000000 00000000  ................
 3ffe8560 0483fe3f 00000000 00002041 61626300  ...?...... Aabc.
 3ffe8570 0d0a006e 616e0069 6e66006f 7666002e  ...nan.inf.ovf..
 3ffe8580 00256400 6265666f 72650061 66746572  .%d.before.after
 3ffe8590 00202530 32780000 25732025 750a0000  . %02x..%s %u...
 3ffe85a0 70700000 18fe3400 64657600 6d616300  pp....4.dev.mac.
 3ffe85b0 706d0000 66706d00 2c0100f6 f4f8f0f5  pm..fpm.,.......
 3ffe85c0 f1f2f3f9 fafbfcfd feff00f4 f6f7f8f2  ................
 3ffe85d0 f5f1f8f3 f9fafbfc fdfeff00 ffffffff  ................
 3ffe85e0 fffff0f1 f2f3f4f5 f6f7f8f9 fafbfcfd  ................
 3ffe85f0 feff00f0 f1f2f3f4 f5f6f7f8 f9fafbfc  ................
 3ffe8600 fdfeff00 73656e64 5f6f6666 65720064  ....send_offer.d
 3ffe8610 68637073 2073656e 64206163 6b006468  hcps send ack.dh
 3ffe8620 63707320 73656e64 206e616b 00000000  cps send nak....
 3ffe8630 000000ff ffffffff ff000000 7c86fe3f  ............|..?
 3ffe8640 7a86fe3f 7886fe3f 7686fe3f 7486fe3f  z..?x..?v..?t..?
 3ffe8650 7286fe3f 7086fe3f 6e86fe3f 6c86fe3f  r..?p..?n..?l..?
 3ffe8660 6a86fe3f 6886fe3f 84021000 10001000  j..?h..?........
 3ffe8670 08002000 10002000 b8002800 1c000000  .. ... ...(.....
 3ffe8680 8ce8fe3f 90e8fe3f 84e8fe3f 94e8fe3f  ...?...?...?...?
 3ffe8690 03060c18 30607801 02030405 06070707  ....0`x.........
 3ffe86a0 07070707 2e6c6f63 616c0000 e00000fb  .....local......
 3ffe86b0 fa000000 bcbf2040 e8030000 60052140  ...... @....`.!@
 3ffe86c0 e8030000 b4222140 60ea0000 141c2140  ....."!@`.....!@
 3ffe86d0 f4010000 d81d2140 64000000 44322140  ......!@d...D2!@
 3ffe86e0 64000000 e42a2140 e8030000 dcc52040  d....*!@...... @
 3ffe86f0 01031c06 2a000000 ffffffff 00000000  ....*...........
 3ffe8700 6d616900 74696d00 494e4600 4e414e00  mai.tim.INF.NAN.
 3ffe8710 30006525 6c640049 6e66696e 69747900  0.e%ld.Infinity.
 3ffe8720 4e614e00 4300504f 53495800 232d302b  NaN.C.POSIX.#-0+
 3ffe8730 2000686c 4c006566 67454647 00000000   .hlL.efgEFG....
 3ffe8740 18122040 64172040 30312040 c4312040  .. @d. @01 @.1 @
 3ffe8750 38122040 7c172040 3c312040 e4312040  8. @|. @<1 @.1 @
 3ffe8760 6887fe3f 00e9fe3f                    h..?...?        

earle@server:~/Arduino/hardware/esp8266com/esp8266/tools/sdk/lib$ objdump -s -j .irom0.text /tmp/arduino_build_263740/sketch_may23a.ino.elf   | grep File
 40238400 0112c110 0df00000 5d000000 46696c65  ........]...File
 40238410 73465345 78697374 733a2046 696c6520  sFSExists: File 
````

laercionit, 2019-05-23:
Unfortunately I am not doing any redefinition of F ().
As I mentioned the same project using core 2.4.2 has different dynamic memory usage than 2.5.2.
Obviously it's a different core, and what I've identified with your guidance is that it's related to all the debug messages in my project.

If you let me know your email, I can even share the entire project with you, so you can confirm what I'm mentioning. The same code without macro reset F (). Totally different behavior in core 2.5.2.

laercionit, 2019-05-23:
void setup() {
  const char *Filename =  "abc";

  /*
   * Comment debug: result below
   * Sketch uses 257144 bytes (24 % ) of program storage space. Maximum is 1044464 bytes.
  *  Global variables use 26540 bytes (32 % ) of dynamic memory, leaving 55380 bytes for local variables. Maximum is 81920 bytes.
  */

  /*
   * Uncommented debug: result below
   * Sketch uses 261100 bytes (24%) of program storage space. Maximum is 1044464 bytes.
   * Global variables use 26660 bytes (32%) of dynamic memory, leaving 55260 bytes for local variables. Maximum is 81920 bytes.
  */
  
  Serial.println (String(F("FilesFSExists: File or Path exists [")) + String(Filename) + String(F("]")));
}

void loop() {
  // put your main code here, to run repeatedly:

}

earlephilhower, 2019-05-23:
Your comment let gcc completely remove the "abc" string and the entire Serial port object (and anything it references) from heap.  That's not really what's going on with your stuff.

RODATA strings are stored in flash and then copied to the top of heap at system startup.

laercionit, 2019-05-23:
@earlephilhower

The problem is not in the string ABC, when commenting and uncommenting the Serial.println you will see the difference of dynamic memory.

What I want to understand is what has changed to try to tailor the debug of my system to the characteristics of the new core 2.5.2 or if it is something that should be fixed in core 2.5.2.

Please run the above code, comment Serial.println and then uncomment.

bhushrockzz, 2021-05-09:
I am a beginner in IoT and I am trying to connect my nodemcu 1.0 with my router 
Code:

#include<ESP8266WiFi.h>

char *ssid ="POCO F1_XRPY";
char *pwd="1234_8765";

void setup() {
  
  Serial.begin(115200);
  WiFi.begin(ssid, pwd);
  while (WiFi.status() != WL_CONNECTED) 
  {
     delay(500);
     Serial.print(".");
  }
  
  Serial.println("");
  Serial.println("WiFi connection Successful");
  Serial.print("The IP Address of ESP8266 Module is: ");
  Serial.print(WiFi.localIP());// Print the IP address
}
  // put your setup code here, to run once:

void loop() {
  // put your main code here, to run repeatedly:
}
I got output in prompt as
Executable segment sizes:
IROM: 236736          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 26892   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1260  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 816   ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 24976 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 265704 bytes (25%) of program storage space. The maximum is 1044464 bytes.
Global variables use 27052 bytes (33%) of dynamic memory, leaving 54868 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM10
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:f5:20:37:fe:97
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 269856 bytes to 199062...
Wrote 269856 bytes (199062 compressed) at 0x00000000 in 17.8 seconds (effective 121.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

and in the serial monitor, I got all ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
Could you please suggest me what to do