esp8266/Arduino #6094: Exception(29) using a lot of libraries (webserver, pubsubclient, U8g2lib, etc..)

odilonafonso, 2019-05-14:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [921600]

### Problem Description
After a few hours of running, the application receives an exception 29.
I tried to use the serial to debug, without success.
I have not identified where the problem might be.
I use a series of libraries, among which:
#include <TimeLib.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266httpUpdate.h>
#include "DHTesp.h"
#include <PubSubClient.h>
#include <ArduinoJson.h>
#include "FS.h"
#include <U8g2lib.h>
#include <Wire.h>

This application was working fine for several consecutive days until some time ago when after a few updates of the Arduino IDE version and some libraries, it started to give this problem.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
The application consists of more than 20 source files and more than 3000 lines of code. I do not know where the problem might be, to put it here.
I'll do this as soon as I have any idea where the error might be.

### Debug Messages
From Exception decoder:

Exception (29):
epc1=0x40226434 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe10:  3fff38a8 3ffeab6c ffffffff 3fff5904  
3ffffe20:  3fff2ff0 3fff2ff0 00000000 40227470  
3ffffe30:  3fff4e74 00000001 00000000 3fff5904  
3ffffe40:  3fff2ff0 00000000 3fff2fec 402264e5  
3ffffe50:  4022984e 3ffea8cc 3fff2fec 40229a50  I'll do this as soon as I have any idea where the error might be.
3ffffe60:  00000000 3ffffe90 00000200 4000050c  
3ffffe70:  3ffea8cc 3ffea8cc 3fff2fec 40229a7e  
3ffffe80:  3fff4abc 3fff2fec 3fff30b0 4022bcab  
3ffffe90:  c0a80f01 3fff31e4 00000000 40236027  
3ffffea0:  00000000 000000f9 3fff593e 3fff309e  
3ffffeb0:  000000f5 3fff30b0 000000f0 00000004  
3ffffec0:  00000004 00000004 00000000 3fff5904  
3ffffed0:  000000fa 10110101 00640104 3fff5904  
3ffffee0:  3fff30d4 3fff5936 3fff4b04 3fff5904  
3ffffef0:  3fff30d4 3fff5936 3fff4b04 40229d10  
3fffff00:  00000001 00000043 3fff2fec 00000000  
3fffff10:  00000000 00000124 3fffff80 3fff5922  
3fffff20:  3fff2fec 3fff5904 3fff30d4 4022d0dc  
3fffff30:  00000014 00000001 0000007f 3fff2fec  
3fffff40:  3fffdc80 3fff3594 3fff38ac 3fff361c  
3fffff50:  00000008 3fff2fec 3fff5904 40226849  
3fffff60:  3fffdc80 3fff3594 3fff38ac 4010453c  
3fffff70:  40243cf2 3fff3594 3fff38ac 40243d04  
3fffff80:  3fff5914 3fff5904 00000002 3fff1bb0  
3fffff90:  4023ebd7 00000000 3fff38ac 40245c63  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x40226434: _wctomb_r at ../../../.././newlib/libc/stdlib/wctomb_r.c line 24
EXCVADDR: 0x00000000

Decoding stack results
0x40227470: tcp_slowtmr at core/tcp.c line 1137
0x4022984e: tcp_input at core/tcp_in.c line 435
0x40229a50: pbuf_free_LWIP2 at core/pbuf.c line 771
0x40229a7e: pbuf_free_LWIP2 at core/pbuf.c line 791
0x4022bcab: dhcp_reboot at core/ipv4/dhcp.c line 1245
0x40229d10: pbuf_copy at core/pbuf.c line 971
0x4022d0dc: igmp_send at core/ipv4/igmp.c line 780
0x40226849: __fpclassifyd at ../../../.././newlib/libm/common/s_fpclassify.c line 19

devyte, 2019-05-15:
As it stands, this issue serves no purpose, especially without a MCVE. I am therefore closing.
I suggest continuing your investigation and if you reach a conclusion report back here, or open a new issue with the required info.
Tips:
-from the log it looks like mem corruption, so look for that. 
-remove parts of your app one by one and test the remaining code. 
-test individual pieces of your code that ate standalone as you can, e. g. some string manipulation function or some html formatter, etc. Think paranoid while testing. 
-update the core version. A whole lot has been fixed since 2.4.1, including bugs that can cause crashes.

closed 2019-05-15 by devyte

odilonafonso, 2019-05-15:
Ok @devyte . Core updated to 2.5.1
I will make new tests.Thanks.
There are more than 3000 lines of code. I wish I had some idea where the error might be.
Do you have any idea where the problem might be? As I said before, this application has been running for over a year, and it usually stays on for days and even months without stopping.
I am suspicious of a new routine that I implemented to look for time on an ntp server, but I have also been updating some libraries, the executable code has greatly increased the size.

odilonafonso, 2019-05-15:
Before, the code occupied 33% of program storage.
Now:
Sketch uses 416884 bytes (39%) of program storage space. Maximum is 1044464 bytes.
Global variables use 38996 bytes (47%) of dynamic memory, leaving 42924 bytes for local variables. Maximum is 81920 bytes.
I need httpUpdate to work, as I have users dependent on updating.

odilonafonso, 2019-05-16:
Hello @devyte 

I recompiled my application after updating the core to 2.5.1 and is already active, being monitored by the serial, for more than 20 hours.
No crashes.

It was two weeks of intense research and attempts to discover the error.
I found dozens of reports of people complaining about this same problem - Exception (29).

I think it's interesting to comment somewhere, so that others do not go through what I went through.

**> "Exception (29) for no apparent reason in core 2.4.1"**