esp8266/Arduino #6099: BearSSL vs Node.js 11: Incoming protocol or record version is unsupported.

suculent, 2019-05-15:
Which one is supported? Just trying to configure Node.js but it should work out of the box.

```
BSSL:_connectSSL: start connection
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Incoming protocol or record version is unsupported.'
```

Those are supported ciphers on my server:

```
$ nmap --script ssl-enum-ciphers -p 7443 thinx.cloud
Starting Nmap 7.70 ( https://nmap.org ) at 2019-05-15 16:21 CEST
Nmap scan report for thinx.cloud (207.154.230.212)
Host is up (0.011s latency).

PORT     STATE SERVICE
7443/tcp open  oracleas-https
| ssl-enum-ciphers: 
|   TLSv1.0: 
|     ciphers: 
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|     compressors: 
|       NULL
|     cipher preference: server
|   TLSv1.1: 
|     ciphers: 
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|     compressors: 
|       NULL
|     cipher preference: server
|   TLSv1.2: 
|     ciphers: 
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_ARIA_256_GCM_SHA384 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_ARIA_128_GCM_SHA256 (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (ecdh_x25519) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (ecdh_x25519) - A
|       TLS_RSA_WITH_AES_256_GCM_SHA384 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CCM_8 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CCM (rsa 2048) - A
|       TLS_RSA_WITH_ARIA_256_GCM_SHA384 (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_GCM_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CCM_8 (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CCM (rsa 2048) - A
|       TLS_RSA_WITH_ARIA_128_GCM_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA256 (rsa 2048) - A
|       TLS_RSA_WITH_AES_256_CBC_SHA (rsa 2048) - A
|       TLS_RSA_WITH_AES_128_CBC_SHA (rsa 2048) - A
|     compressors: 
|       NULL
|     cipher preference: server
|_  least strength: A

Also:

Obtaining cipher list from OpenSSL 1.0.2g 1 Mar 2016.
Testing ECDHE-RSA-AES256-GCM-SHA384...YES
Testing ECDHE-RSA-AES256-SHA384...YES
Testing ECDHE-RSA-AES256-SHA...YES
Testing AES256-GCM-SHA384...YES
Testing AES256-SHA256...YES
Testing AES256-SHA...YES
Testing ECDHE-RSA-AES128-GCM-SHA256...YES
Testing ECDHE-RSA-AES128-SHA256...YES
Testing ECDHE-RSA-AES128-SHA...YES)
Testing AES128-GCM-SHA256...YES
Testing AES128-SHA256...YES
Testing AES128-SHA...YES
```

I'm using Node.js v11.9 (should support MPLN). I'd like to know, how to debug ciper negotiation, @earlephilhower pls?

earlephilhower, 2019-05-15:
Other folks are running Node.js and connecting fine, so there's something up with the config or source on one side or the other.  Without seeing any source code, there's nothing we ca do here.

The easiest way to check neg'n is to use Wireshark.  It can decode the SSL handshake packets easily and see what's being offered and what's being negotiated.

There is no BearSSL deeper debug option for handshake.  You can patch the pseudo-forth in the library and work that way, but you'll need to obviously build the library yourself from the repo.

suculent, 2019-05-15:
Thanks a lot. 

The node has default config, valid certs with and client as well with setInsecure() enabled. 

I’m wondering which cipher should I force in node. 

Odesláno z iPhonu

suculent, 2019-05-15:
I've got this working to the point, where HTTPS works and ciphers are properly exchanged. My issue was on the server-side, where I have actually two servers (Node.js) and Mosquitto. Once I've enabled ciphers that I've scanned from the Node.js in Mosquitto, everything works cool.

Just the MPLN is not working yet on my Node.js 11.19.0.

earlephilhower, 2019-05-15:
Make sure you use the ::setBufferSizes() call before starting a connection, or it will be ignored (since we need that space allocated to do the handshake).

suculent, 2019-05-15:
Works great, closing. I was expecting to have the MFLN status true after probing, but it got true after connect.

closed 2019-05-15 by suculent