esp8266/Arduino #6102: Compiler errors in FS.h using new V2.5.1

mars000, 2019-05-16:
Hi
I've just upgraded to core V2.5.1 using PlatformIO Espressif 8266 dev/platform v2.1.0

I now get the following compiler errors (I've made no changes to my code). Code compiles fine with previous version of Arduino ESP8266 code V2.5.0

I get the following compiler errors - any help would be appreciated:
[image]

earlephilhower, 2019-05-16:
Please try PR #6101.

mars000, 2019-05-19:
thanks - got it working. Is there a fix coming in next release ?

btw: casted info[x] from char to uint8_t and it removed compiler errors.
i.e. f.write( ((uint8_t) info[x]) );

earlephilhower, 2019-05-19:
The bugfix will be out in a week or so, with this and some other fixes included.

earlephilhower, 2019-05-22:
2.5.2 is out with the fix. Closing.

closed 2019-05-22 by earlephilhower