esp8266/Arduino #6103: [core 2.5.1] ESP12F interference with NRF24L01 after upgrade to new core 2.5.1

egzola, 2019-05-16:
After upgrading to the new version 2.5.1 library, ESP is causing interference on NRF24L01 receiving

jandrassy, 2019-05-16:
or SPI communication?

egzola, 2019-05-16:
> or SPI communication?

Nope. NRF is connected to an ATMEGA328P (via SPI), so the ATMEGA communicate via serial with the esp12F.

earlephilhower, 2019-05-16:
Can you try a HW serial connection, to debug?  SWSerial was updated and that probably has a higher chance of affecting the AVR->8266 serial comms than the 8266 itself causing add'l RF interference.

devyte, 2019-05-17:
What do you mean by interference on reception? I take that to mean wireless interference.
Or do you mean serial communication errors?

egzola, 2019-05-17:
I mean wireless interference... But alternatively may be software serial issues too. The fact is with 2.5.0 core i have no problem with my devices. I have about 30 devices like that, in PCBs developed specific for this project. With the core 2.5.1 all the 30s have issues. I downgraded again to 2.5.0 and all of them is ok back.

earlephilhower, 2019-05-17:
So it sounds like you've got a RF spectrometer and have determined that you're getting a higher noise floor w/the core and hence a lower SNR on the NRFs.  Since the 8266 Arduino core doesn't have direct access to the RF section (it's all binary blobs from the mfgr) then your best bet is to pop to the Espressif site and see if they have any field notes on the NONOS221 wifi performance.  They might have changed the PA or MAC control.

Alternatively, try pulling the older SW serial from 2.5.0 into the 2.5.1 version and see if that helps out.  If so, then you can check the https://github.com/plerup/espsoftwareserial/ repo and file something there.  A logic analyzer trace would be useful, if so, on the rx/tx pins.

jandrassy, 2019-05-17:
try to flash esp_init_data_default.bin matching the SDK used by the arduino core (and erase the flash before that or flash blank.bin to the RF_CAL parameter area). Addresses for different flash sizes are in the AT firmware reference in chapter "Downloading AT Firmware into the Flash"

egzola, 2019-07-07:
The problem persists, but I think the point of the problem rely on Soft Serial.

closed 2019-07-07 by egzola