esp8266/Arduino #6105: FILE_WRITE replacing existing content in File on SD-Card in 2.5.1

Hirschenberger-S, 2019-05-16:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			        [ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE 1.8.9]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module and Adafruit Feather Huzzah]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[80Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:            [921600] (serial upload only)

### Problem Description

After updating to 2.5.1, data that gets pushed to an existing file on the SD-Card via    
 file.println(data); gets no longer added below the last line in the file, but replaces the whole file content. So only the last data, that was pushed to the file, stays there.

I was not able to figure out whether the whole file is replaced or only the content.
Same issue with short and long file name.

Examples to reproduce the Error: SD/Datalogger or SD/ReadWrite

Or the one below.

Output on Serial Monitor at readout should be:

> test.txt:
> testing 1, 2, 3.
> testing 4, 5, 6.

But only is:
  

> test.txt:
> testing 1, 2, 3.

```
#include <SPI.h>
#include <SD.h>

File myFile;

void setup() {
  Serial.begin(9600);
  while (!Serial) {
  }

  Serial.print("Initializing SD card...");

  if (!SD.begin(4)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  myFile = SD.open("test.txt", FILE_WRITE);

  if (myFile) {
    myFile.println("testing 1, 2, 3.");
    myFile.close();

    myFile.println("testing 4, 5, 6.");
    myFile.close();
    Serial.println("done.");
  } 
else {
    Serial.println("error opening test.txt");
  }

  // re-open the file for reading:
  myFile = SD.open("test.txt");
  if (myFile) {
    Serial.println("test.txt:");

    // read from the file until there's nothing else in it:
    while (myFile.available()) {
      Serial.write(myFile.read());
    }
    // close the file:
    myFile.close();
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }
}

void loop() {
  // nothing happens after setup
}

```
### Debug Messages
none

earlephilhower, 2019-05-16:
The listed example has the sequence which is guaranteed to fail:

````
    myFile.println("testing 1, 2, 3.");
    myFile.close();

    myFile.println("testing 4, 5, 6.");
    myFile.close();
    Serial.println("done.");
````

After the first close, the writing of the 2nd line will fail since the file is no longer opened.

earlephilhower, 2019-05-16:
That said, upon looking at the code I do see the difference.  The attached PR reverts to the old behavior.  You can try it yourself in your code by simply adding  `| sdfat::O_APPEND` to the define in the header.

Hirschenberger-S, 2019-05-16:
Oh sorry... of course. i was to ambitious cutting out unnecessary parts to provide a short example.
correct Code-Snippet to show the behavior with and without the already requested modification:

```
  myFile = SD.open("test.txt", FILE_WRITE);
  if (myFile) {
    myFile.println("testing 1, 2, 3.");
    myFile.close();
    Serial.println("1st write done");
  }

myFile = SD.open("test.txt", FILE_WRITE);
  if (myFile) {
    myFile.println("testing 4, 5, 6.");
    myFile.close();
    Serial.println("2nd write done");
  } 
```

THANK YOU FOR that quick fix!
I have tested it and can confirm that it solves the "problem" and makes the new wrapper work just like the old library in this case.

earlephilhower, 2019-05-16:
Reopening to track the fix commit

PR #6106 (2019-05-17): Make FILE_WRITE append in SD.h wrapper
Fixes #6105
commit 968d6fcb (2019-05-17): Make FILE_WRITE append in SD.h wrapper (#6106)

* Make FILE_WRITE append in SD.h wrapper

Fixes #6105

* Update host tests
closed 2019-05-17 by earlephilhower