esp8266/Arduino #6110: Getting Error While Code Upload - NODEMCU ESP8266

Mugesh-Rajaram, 2019-05-18:
Hi, 
While am trying to upload the code from Arduino IDE to NODEMCU ESP8266 board getting some error can anyone help me to clear this error.

**Program**
```
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

char ssid[] = "MUGESH";
const char* password = "87654321";

void setup() {
  Serial.begin(9600);         //Serial connection
  WiFi.begin(ssid, password);   //WiFi connection

  while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion
    delay(500);
    Serial.println("Waiting for connection");
  }
  Serial.println("WIFI Connected with : " + WiFi.localIP());
}

void loop() {
  if (WiFi.status() == WL_CONNECTED) { //Check WiFi connection status
    HTTPClient http;    //Declare object of class HTTPClient
    http.begin("https://jsonplaceholder.typicode.com/posts");      //Specify request destination
    http.addHeader("Content-Type", "text/plain");  //Specify content-type header
    int httpCode = http.GET();   //Send the request
    String payload = http.getString();                  //Get the response payload
    StaticJsonDocument<500> jsonDoc;
//    JsonObject& jsonObj = jsonBuffer.parseObject(payload);
    deserializeJson(jsonDoc, payload);
    serializeJsonPretty(jsonDoc, Serial);
    Serial.println(httpCode);   //Print HTTP return code
    Serial.println(payload);    //Print request response payload
    http.end();  //Close connection
  } else {
    Serial.println("Error in WiFi connection");
  }
  delay(30000);  //Send a request every 30 seconds
}
```
**Error In Text**
```
Sketch uses 287632 bytes (27%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27156 bytes (33%) of dynamic memory, leaving 54764 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 84:0d:8e:86:3c:a9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 291792 bytes to 210262...

Writing at 0x00000000... (7 %)Traceback (most recent call last):
  File "C:\Users\IN\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 2698, in main
    operation_func(esp, args)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 2158, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio * 2)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 632, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 354, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 322, in command
    self.write(pkt)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 285, in write
    self._port.write(buf)
  File "C:/Users/IN/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/pyserial\serial\serialwin32.py", line 323, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
the selected serial port serial.serialutil.SerialTimeoutException: Write timeout
 does not exist or your board is not connected
```
**Error Image**

[image]

earlephilhower, 2019-05-18:
The error message from the uploader tells you the problem:
> the selected serial port does not exist or your board is not connected

Looks like a power or connectivity problem and the 8266 disconnects during upload.  Try a different usb port and cable, then please use the forum www.esp8266.com if you want add'l assistance, as this is not related to the core.

closed 2019-05-18 by earlephilhower

Mugesh-Rajaram, 2019-05-24:
@earlephilhower  After changing the USB port and the cable also the same error persist, even i have flashed my Firmware too.

shrakanis, 2019-05-30:
did you managed to fix it?
it happened to me too after Arduino libray update

snow-drop, 2019-06-01:
@shrakanis Same here. This is the first time I've use Node MCU I bought last week and seems that there is no fix available yet after days of searching on the Web.

jpradoar, 2019-06-01:
I have the same error.
but before write that I tested...
3 usb cables (all cables with data support, not only power)
3 usb ports with all cables
and then I bought a new nodemcu v3
I did same test but using arduino ide 1.8.8
I runned arduino ide and compile with my user and then with root.

but nothing has been running ok.

soft info
<pre>
arduino ide version: 1.8.9
os: ubuntu 18.04.2 LTS (Bionic Beaver)
kernel: 4.18.0-20-generic
</pre>

<pre>
Sketch uses 257672 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 26572 bytes (32%) of dynamic memory, leaving 55348 bytes for local variables. Maximum is 81920 bytes.
Sketch uses 257672 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 26572 bytes (32%) of dynamic memory, leaving 55348 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: ca:ac:11:43:aa:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 261824 bytes to 191216...
Writing at 0x00000000... (8 %)Traceback (most recent call last):
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2698, in main
    operation_func(esp, args)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2158, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio * 2)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 632, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 354, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 322, in command
    self.write(pkt)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 285, in write
    self._port.write(buf)
  File "/root/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/pyserial/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
the selected serial port serial.serialutil.SerialTimeoutException: Write timeout
 does not exist or your board is not connected
</pre>

Mugesh-Rajaram, 2019-06-02:
Hi @shrakanis  / @snow-drop 
After downgrading the ESP8266 version from 2.5.2 to 2.5.1 my issue is resolved, check once the same for you people also. 
https://github.com/esp8266/Arduino/issues/6110#issue-445707173

shrakanis, 2019-06-02:
Yesterday did the same ;) thanx

snow-drop, 2019-06-02:
@Mugesh-Rajaram Thank you very much! It worked for me too.

jpradoar, 2019-06-02:
Thanks, that worked for me.

jwrussert, 2019-06-09:
I think I just solved this same issue. There is text on my board saying to use a 9600 Baud. So I changed the upload speed to 9600. When I change the upload speed back to 115200 the error goes away.

sathyasathish, 2019-07-06:
still i'm getting the same error ...i changed the baud rate , esp8266 version to 5.2.1 in board manager....Please help me to solve this error..

denisfou, 2019-07-06:
Same errors here, version 2.5.2 of boards seems to be bugged, 2.5.1 is ok. 

I have pb with ESP01 and nodemcu with 2.5.2.

denisfou, 2019-07-06:
Same errors here, version 2.5.2 of boards seems to be bugged, 2.5.1 is ok.
I have pb with ESP01 and nodemcu with 2.5.2.

The cause is esptool.py v2.6 i guess :
Bad display (no scrolls when progessing upload)
Bad initialization : soft reset doesn't work on ESP just after download.

sunnybera, 2019-07-10:
> I think I just solved this same issue. There is text on my board saying to use a 9600 Baud. So I changed the upload speed to 9600. When I change the upload speed back to 115200 the error goes away.

This worked for me to fix serial communication issue with ESP8266 using Arduino IDE 1.8.9 on Ubuntu Release 18.04.1 LTS (Bionic Beaver) 32-bit.

sathyasathish, 2019-07-11:
Thank you guys!!It worked for me too...when i changed the baud rate......

vineethnair1991, 2019-08-29:
> Hi @shrakanis / @snow-drop
> After downgrading the ESP8266 version from 2.5.2 to 2.5.1 my issue is resolved, check once the same for you people also.
> [#6110 (comment)](https://github.com/esp8266/Arduino/issues/6110#issue-445707173)

this suggestion worked. Thanks

dhrumil1202, 2019-09-05:
After downgrading and changing the upload speed back to 115200 the issue resolved.try this one once!!!

mcxiaoke, 2019-09-24:
use baud 115200 worked for me when flashing micropython to nodemcu v3 board. thanks @jwrussert

mmaurier, 2020-02-09:
I had the same problem, all I had to do to fix it was to switch the port
Tools=>Port=>COM5

pitchaimuthufb, 2020-02-11:
still i'm getting the same error ...i changed the baud rate , esp8266 version to 5.2.1 in board manager....Please help me to solve this error..

vimalswaroopj, 2020-03-27:
Try using another USB port. In my case this was the problem i changed the port and the problem got solved. Hope this will help someone.

pitchaimuthufb, 2020-03-27:
Thanks a lot for your help,

Thank you so much.

unnati-ramteke, 2020-04-12:
my esp8266 version is 6.2.3 and the error says the selected serial port serial.serialutil.serialtimeoutexception:write timeout doest not exist or your board is not connected. What to do ?

mehdibolourian, 2020-07-20:
> my esp8266 version is 6.2.3 and the error says the selected serial port serial.serialutil.serialtimeoutexception:write timeout doest not exist or your board is not connected. What to do ?
> [image]

Try turning off your Bluetooth! The "Standard Serial over Bluetooth Link" may have the same COM port number as the ESP8266 and It may stop it from getting programmed right.

mahdipg, 2020-07-20:
who are you

Syed-Muddassir, 2020-09-30:
how to downgrade versions can anybody help?

Ychibo, 2020-10-04:
> how to downgrade versions can anybody help?
Click Boards Manager
Tools=>Board: <your board>=>Boards Manager...
In the search bar type ESP8266
Choose esp8266 by ESP8266 community. 
You will see the version you currently have. 
On the bottom left corner you will have a drop down menu where you can pick the version you want to use.

JhonatanPereira, 2020-11-18:
I'm having the same problem, but the upload works fine when I connect another ESP8266. The "faulty" one are beeing recognized by Windows, blinks when I press the RST button, apparently normal, but the upload got stuck at "Connecting...", and then I receive the "Write timeout" error. Does anyone have any suggestions?

waseber, 2020-11-19:
I'm having the same issue "the selected serial port _ does not exist or your board is not connected" regardless if I use an Arduino Nano, Uno, or FTDI. I've changed the baud rates as suggested above and even downgraded the the esp8266 firmware to 2.5.1 too , but nothing seems to work. Right now I have the ESP8266 connected to the FTDI ft232rl. I'm using IDE 1.8.13 (Windows).

I've been following schematics online for connections, but if someone could provide details on IDE config, etc. that would be super helpful at this point--I'll gladly switch back to the Nano or Uno if that's easier.

Thanks in advance.

hen-a-wijaya, 2020-12-11:
Hi, 
I'm having same issue. All suggestion already done. Downgrade to 2.5.1, change usb cable, change baud rate 9600 or 115200.
Could you share setting of working board manager?

Sketch uses 257696 bytes (51%) of program storage space. Maximum is 499696 bytes.
Global variables use 26572 bytes (32%) of dynamic memory, leaving 55348 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port COM4
Connecting...
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.1/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 460, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 440, in _connect_attempt
    self.sync()
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 379, in sync
    timeout=SYNC_TIMEOUT)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 322, in command
    self.write(pkt)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/esptool\esptool.py", line 285, in write
    self._port.write(buf)
  File "C:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/pyserial\serial\serialwin32.py", line 323, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
the selected serial port serial.serialutil.SerialTimeoutException: Write timeout
 does not exist or your board is not connected

[image]

Stradivario, 2022-08-08:
Trying to flash ESP8266 gives me the following error

```
esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash --flash_freq 80m --flash_mode dio --flash_size 4MB-c1 0x0000 espruino_2v08_esp8266_4mb_combined_4096.bin
esptool.py v3.1-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: f4:cf:a2:5b:ad:8a
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x026f
Compressed 4194304 bytes to 357516...
Writing at 0x0000a2d1... (13 %)Traceback (most recent call last):
  File "/home/rampage/.local/bin/esptool.py", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/home/rampage/esptool/esptool.py", line 4132, in <module>
    _main()
  File "/home/rampage/esptool/esptool.py", line 4125, in _main
    main()
  File "/home/rampage/esptool/esptool.py", line 3767, in main
    operation_func(esp, args)
  File "/home/rampage/esptool/esptool.py", line 3132, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "/home/rampage/esptool/esptool.py", line 142, in inner
    return func(*args, **kwargs)
  File "/home/rampage/esptool/esptool.py", line 796, in flash_defl_block
    self.check_command("write compressed data to flash after seq %d" % seq,
  File "/home/rampage/esptool/esptool.py", line 448, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/rampage/esptool/esptool.py", line 409, in command
    self.write(pkt)
  File "/home/rampage/esptool/esptool.py", line 372, in write
    self._port.write(buf)
  File "/home/rampage/.local/lib/python3.8/site-packages/serial/serialposix.py", line 636, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout
```

I didn't change anything since the last two months but it appears that now the flash is not working at all...

Curious what can cause it ? 

Is it possible some linux driver tty update since i didn't touch anything on this computer 

I am using Ubuntu 20.04.4 LTS

indrareddy333, 2022-09-22:
How to downgrade  ESP8266 2.5.2 to 2.5.1