esp8266/Arduino #6114: mDNS responder kills DHCP server on lwip v2

newHeiko, 2019-05-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it (no stack dump).
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.5.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Slackware Linux]

### Settings in IDE

- Module:  			[Nodemcu v0.9]
- Flash Mode:			[unknown]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Higher Bandwidth]
- Reset Method:		[unknown]
- Flash Frequency:	[unknown]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I'm trying to connect to a WiFi on startup, wait for a timeout, then open an AP to allow the user to connect.

This works fine on lwip v1.4.
This works fine on lwip v2 Higher Bandwith with MDNS disabled (lines starting with MDNS commented out of the MCVE below)
This fails on lwip v2 Higher Bandwith with MDNS enabled - the client connects to the WiFi but does not receive an IP address and the Serial port debugging shows:

ERROR: dhcps send ack (error -13)
ERROR: dhcps send ack (error -13)
ERROR: send_offer (error -13)
ERROR: send_offer (error -13)

Note: I have replaced all the "check for connection or timeout" handling with a simple delay(5000) in the MCVE below. Same problem.

### [MCVE] Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  WiFi.begin("undef", "undef");
  delay(5000);

  // open an AP for configuration if connection failed
  WiFi.disconnect();
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP("aptest");

  MDNS.begin("config");
  MDNS.addService("http", "tcp", 80);
}

void loop() {
  MDNS.update();
}
```

### Debug Messages from serial port

```
SDK:2.2.1(cfd48f3)/Core:2.5.1=20501000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
mode : sta(a0:20:a6:13:6c:ac) + softAP(a2:20:a6:13:6c:ac)
add if0
scandone
no undef found, reconnect after 1s
reconnect
scandone
bcn 0
del if1
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
add 1
aid 1
station: 10:0b:a9:7b:e0:44 join, AID = 1
ERROR: dhcps send ack (error -13)
ERROR: dhcps send ack (error -13)
ERROR: send_offer (error -13)
ERROR: send_offer (error -13)
```

JosephCottingham, 2019-06-06:
me too, thanks for the opening!

tarzan115, 2019-07-20:
me too, I'm facing this problem :/

9zigen, 2019-07-23:
Found a temporary solution, DO NOT USE WiFi.mode (WIFI_AP_STA), works fine in WIFI_AP or WIFI_STA

d-a-v, 2019-07-23:
When using both AP+STA, you need to start two different mDNS instances

```
MDNS.begin("name1", my-sta-IP-address)
```

and
```
MDNSResponder MDNS2;
MDNS2.begin("name2", my-ap-IP-addess)
```

...

```
// in loop()
MDNS.updtate();
MDNS2.update();
```

Still missing:
- documentation
- #6261 which will bring some fixes to this

newHeiko, 2019-07-23:
> Found a temporary solution, DO NOT USE WiFi.mode (WIFI_AP_STA), works fine in WIFI_AP or WIFI_STA

I can confirm this workaround works, but... I also want to scan for available WiFi networks, which automatically switches back to WIFI_AP_STA. Mostly, I have the "configuration client" connected already when I initialize a scan, so I don't need the DHCP server anymore, but that's not a nice solution - so I'll stay with LWIP v1.4 for the time being.

Updated MCVE:

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  Serial.setTimeout(10);
  Serial.setDebugOutput(true);

  WiFi.begin("undef", "undef");
  delay(5000);

  // open an AP for configuration if connection failed
  WiFi.disconnect();
  WiFi.mode(WIFI_AP);
  uint8_t mac[6];
  WiFi.macAddress(mac);
  String ssid = "wiClock-config" + String(mac[0], 16) + String(mac[5], 16);
  WiFi.softAP(ssid.c_str());

  MDNS.begin("config");
  MDNS.addService("http", "tcp", 80);
}

void loop() {
  MDNS.update();

  static uint32_t timeout = 15000;
  if(millis() > timeout)
  {
    Serial.println("Scanning for networks...");
    int n = WiFi.scanNetworks();
    for (int i = 0; i < n; i++)
    {
      Serial.println(WiFi.SSID(i));
    }
    timeout = millis() + 5000;
  }
}
```

> When using both AP+STA, you need to start two different mDNS instances

But... that only works if STA is connected, right? Also why should mDNS have any influence on the DHCP server at all?

Heiko

d-a-v, 2019-07-24:
@newHeiko I tried your MCVE in OP and the one just above (using `MDNS.begin("config", WiFi.softAPIP());`)  with latest master and I got no dhcps message.
Have you tried with latest up-to-date master ?

newHeiko, 2019-07-24:
> Have you tried with latest up-to-date master ?

No, sorry, I'm still on 2.5.2 (not 2.5.1 any more as per the IP)

Will check with master next week, I need a "stable" solution over the weekend. But it would be great if this is fixed :)

I'll also try explicitly specifying the WiFi.softAPIP() as you point out.

Thanks,
Heiko

Just to clarify: Yesterday's updated MCVE only fails after the first scan reactivates STA mode. So if the client connects before and does not disconnect, the failure is dormant.

psatya111, 2019-08-21:
I am using 2.5.2 and I can conform that esp AP mode(only AP) has also same issue.i have tested more than a week without restart and I found the issue.