esp8266/Arduino #6120: WiFiMulti.ino example confusion, and possible enhancement

RudyFiero, 2019-05-20:
### Basic Infos

- [*] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [*] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [*] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [*] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.5.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[none]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

In the setup() function of the WiFiMulti.ino sketch there is code to print out WiFi.localIP(). The print statements are conditional on  WL_CONNECTED being true. But the test is false as there is insufficient time given to establish a connection and obtain and I.P. address. There is no wait and no retries. It gives the impression that it would be printed but never is.

The code then exits the setup function and executes the loop() function. Here a test is done and if there has not yet been a connection established "WiFi not connected!" is printed. The test is repeated every second.

What is displayed in the serial monitor is the following. (in my case)

```
r001??r??#0?<0??????8O??p>??nE??< .... ------------------------------------- ...  Connecting Wifi...
WiFi not connected!
WiFi not connected!
WiFi not connected!
```

Now to be honest, I did not look closely at the code at first. I just assumed that the connection would be made, and the I.P. would be printed. Many of the other examples wait until the connection is established, printing .... (dots) until it does, and then prints out the IP. This is what I expected.

What I ended up seeing is **WiFi not connected!** and I assumed something did not work. I assumed I did not obtain a connection. I had no further feedback. I assumed something crashed. 

I made bad assumption on what I thought would be expected behaviour. But in actual fact the output was precisely in line with the code and the connection status. Unfortunately that was not helpful output. 

There are a number of options that could make the example more useful and less confusing. Blocking code can be used until a connection has been established, (as with other examples) then print out the obtained IP. That is a simple solution but I don't particularly like the blocking code,  and it perpetuates the simple example that is copied over and over, becoming the de facto standard for those learning to write code for the ESP8266. 

Another option is to add and else statement in the loop function. It currently prints  "WiFi not connected!" when there is no connection, then nothing. When connected it could print out "WiFi connected", "IP address: ", followed by WiFi.localIP().
  

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
/*             CURRENT EXAMPLE
    This sketch trys to Connect to the best AP based on a given list
*/

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti wifiMulti;

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  wifiMulti.addAP("MTS049", "---------");
  wifiMulti.addAP("MTS61752", "----------");

  Serial.println("Connecting Wifi...");
  if (wifiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
}

void loop() {
  if (wifiMulti.run() != WL_CONNECTED) {
    Serial.println("WiFi not connected!");
    delay(1000);
  }
}

```

```
/*    SIMPLE REVISED SKETCH
    This sketch tries to Connect to the best AP based on a given list
*/

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti wifiMulti;

void setup() {
  Serial.begin(115200);
  Serial.println("");

  WiFi.mode(WIFI_STA);
  wifiMulti.addAP("MTS049", "8658488726");
  wifiMulti.addAP("MTS61752", "6900979039");

  Serial.println("Connecting Wifi...");
  if (wifiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.println("WiFi connected, ");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
}

void loop() {
  if (wifiMulti.run() != WL_CONNECTED) {
    Serial.println("WiFi not connected!");

  }
  else {
    Serial.print("WiFi connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  delay(1000);
}

```

If I knew how to use Git I would make a pull request. Unfortunately I don't and am not prepared to figure all that out at this time. I am not a programmer.

brasofilo, 2020-04-06:
"...not a programmer" but extremely organized, thanks alot for this, quite useful

earlephilhower, 2021-01-13:
Closing as the example in question has been refactored and this doesn't seem to apply anymore.

closed 2021-01-13 by earlephilhower