esp8266/Arduino #6125: HTTPClient returning reponseCode "-1" when using PUT

Langtude-Larson, 2019-05-21:
#### Platform

- Hardware:				ESP-12
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE
- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other] (Can't tell)
- Flash Size: 		4MB (No Spiff)
- lwip Variant:   v2 Lower Memory
- Reset Method:		[ck|nodemcu] (Can't tell)
- Flash Frequency:	[40Mhz] (Can't tell)
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description
I've recently stumbled into a problem when doing a PUT, i can't tell for sure if the problem actually lies with the library or ESP board i'm using, but to my knowlagde the board itself should be able to PUT, again, not sure. 

I'm attempting to access the philps HUE API with the ESP

1. The PUT request doesn't go through, i've tried using other software from my PC to call PUT with these exact parameters and it works perfectly, just not with the ESP for some reason

2. The httpResponseCode i get is -1. I haven't been able to find other people having the same issue but i'm hoping someone here might be able to help me out

3. I've tested the connection with JSONPlaceholder to make sure it wasn't a fault with my HUE bridge and the same error is returned

The board i'm using is a WEMOS D1 - mini, it might be a clone since i don't see the WEMOS logo anywhere on it

### Arduino Sketch

```cpp
  http.addHeader("Content-Type", "text/plain");
  
  String put_string;
  put_string  = "{";
  put_string += "\"on\":true";
  put_string += "}";
  
  int httpResponseCode = http.PUT(put_string);
  Serial.println(put_string);
```

### Debug Messages

```
Error on sending PUT Request: -1
```

d-a-v, 2019-05-21:
Maybe Tasmota or ESPEasy? projects can help with this?
We don't have this hardware to test.

You can show us a complete working dialog using curl:
```
curl -v -X PUT -d '{on:true}' http://hue-bridge.local
```
and a run on esp8266 with httpclient debug mode enabled.
So we can see what is happening.

Langtude-Larson, 2019-05-21:
> curl -v -X PUT -d '{on:true}' http://hue-bridge.local

* TCP_NODELAY set
* Connected to ***.***.*.** (***.***.*.**) port 80 (#0)
> PUT /api/Iih97lD6oemOFTXUeCNR0UZk1gyUW5zJ9wfpYu63/lights/1/state HTTP/1.1
> Host: ***.***.*.**
> User-Agent: curl/7.55.1
> Accept: */*
> Content-Length: 11
> Content-Type: application/x-www-form-urlencoded
>
* upload completely sent off: 11 out of 11 bytes
< HTTP/1.1 200 OK
< Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0
< Pragma: no-cache
< Expires: Mon, 1 Aug 2011 09:00:00 GMT
< Connection: close
< Access-Control-Max-Age: 3600
< Access-Control-Allow-Origin: *
< Access-Control-Allow-Credentials: true
< Access-Control-Allow-Methods: POST, GET, OPTIONS, PUT, DELETE, HEAD
< Access-Control-Allow-Headers: Content-Type
< Content-type: application/json
<
[{"success":{"/lights/1/state/on":true}}]* Closing connection 0

Langtude-Larson, 2019-05-21:
> > curl -v -X PUT -d '{on:true}' http://hue-bridge.local
> 
> ```
> * TCP_NODELAY set
> * Connected to xxx.xxx.x.xx (xxx.xxx.x.xx) port 80 (#0)
> > PUT /api/Iih97lD6oemOFTXUeCNR0UZk1gyUW5zJ9wfpYu63/lights/1/state HTTP/1.1
> > Host: xxx.xxx.x.xx
> > User-Agent: curl/7.55.1
> > Accept: */*
> > Content-Length: 11
> > Content-Type: application/x-www-form-urlencoded
> >
> * upload completely sent off: 11 out of 11 bytes
> < HTTP/1.1 200 OK
> < Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0
> < Pragma: no-cache
> < Expires: Mon, 1 Aug 2011 09:00:00 GMT
> < Connection: close
> < Access-Control-Max-Age: 3600
> < Access-Control-Allow-Origin: *
> < Access-Control-Allow-Credentials: true
> < Access-Control-Allow-Methods: POST, GET, OPTIONS, PUT, DELETE, HEAD
> < Access-Control-Allow-Headers: Content-Type
> < Content-type: application/json
> <
> [{"success":{"/lights/1/state/on":true}}]* Closing connection 0
> ```
> 
> This is what i got from the ESP
> 
> ```
> [HTTP-Client][begin] url: xxx.xxx.x.xx/api/Iih97lD6oemOFTXUeCNR0UZk1gyUW5zJ9wfpYu63/lights/1/state
> [HTTP-Client][begin] failed to parse protocol
> [HTTP-Client][sendRequest] type: 'PUT' redirCount: 0
> [HTTP-Client] connect: HTTPClient::begin was not called or returned error
> [HTTP-Client][returnError] error(-1): connection refused
> ```

Langtude-Larson, 2019-05-21:
The error was missing the *"http://"* part and a trailing slash at the end. thanks d-a-v for asking me to put it into debug mode, helped a lot 💯

closed 2019-05-21 by Langtude-Larson