esp8266/Arduino #6130: Default hostname non-compliant with rfc952 (again)

philbowles, 2019-05-22:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:                             ALL			
- Core Version:			2.5.2
- Development Env:		Arduino IDE 1.8.9
- Operating System:		Windoze 10

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020

### Settings in IDE

N/A : applies to all

### Problem Description

The default WiFi chosen by the system is non-compliant with RFC952. This has been mentioned in the past  (e.g. https://github.com/espressif/ESP8266_NONOS_SDK/issues/208, https://github.com/esp8266/Arduino/issues/4415 )

Has it regressively crept back in or was it decided to ignore it back then?

philbowles, 2019-05-22:
It is even explicilty stated as such in: https://arduino-esp8266.readthedocs.io/en/2.5.2/esp8266wifi/station-class.html "Function returns String type. Default hostname is in format ESP_24xMACwhere 24xMAC are the last 24 bits of module’s MAC address."

d-a-v, 2019-05-22:
@philbowles Bug was fixed in (not default) sdk-2.2.x which you can enable with arduino ide menu when using generic esp8266 board.

About local copies of other firmwares (including default 2.1.x), we haven't binary-patched them.
@devyte @earlephilhower Why not doing this now in our local copies, considering upstream 2.2.x and probably also latest 3.x have it ?

devyte, 2019-05-22:
We can do that for a following release. It's just a matter of replacing 1 byte in the sdk lib.

philbowles, 2019-05-23:
@d-a-v I'm confused: Does this not mean I AM using sdk-2.2.x ?:

SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020

If so, it is absolutely NOT fixed, hence the post. What am I missing?

d-a-v, 2019-05-23:
Sorry my mistake.
Current/default one is 2.2.1. Test one is 2.2.2.

philbowles, 2019-05-26:
Thanks! Originally I thought that for some bizarre reason it might be causing an exception in my code, but the code has moved on and I now can't replicate the exception (thankfully!). Hence it's not a "showstopper", more a "FYI" so I will close it now.

Thanks for rapid response and keep up the great work.

closed 2019-05-26 by philbowles