esp8266/Arduino #6132: 800-900ms between loop() latency during wifi connection handling -2

dsptech, 2019-05-23:
#4891 
as requested in the above issue, I open a new one.

I'm try to execute some critical time code and discovered a 900ms loop latency in loop, a little time after that WiFi.begin is issued (persistent, connect and autoreconnect disabled).

I tested several configuration with no effects. Inthe last test,  WiFi.begin is issued after a WiFi.disconnect(true) and 200ms of delay.
The 900ms latency is the same at 80Mhz and 160Mhz clock, so I think about a kind of blocking polling loop. All tasks managed by the SDK (system_os_task) are not execute too.
The only way I found to manage a time critical code is to attach it to an hardware interrupt (i.e.: Timer1).
Commit: 968d6fcb0a9a2eab657178c95ee4302d6f470e46

devyte, 2019-05-23:
@dsptech the request was:
> open a new bug with some more info.

When you opened this issue you were presented with a template with specific instructions to follow and info to fill out, and that was completely ignored. Please either edit this issue and add the info, or close it and open a new one correctly.

dsptech, 2019-05-23:
Hi @devyte, 
sorry, the issue match exactly the referenced issue. When earlephilhower closed it, he asked to open a new bug if the problem is still on also using an updated Arduino framework.  

Attached to this message you could find the simplified sketch that I used to replicate the issue. I also traced the activity using a simple logic analyzer (see the attached pictures). 
The 2nd pulse on the "SETUP" track is the WiFi.begin call.

As you can see, the loop function is affected by a 887ms of latency due the WiFi activity . The cpu frequency does not affect the latency time.  
As additive test I also checked the Timer 1 interrupt (not in this sketch). That interrupt continue to work as expected (I used it as workaround for time critical functions).  

Regards.
[Esp8266Test-2.zip](https://github.com/esp8266/Arduino/files/3213534/Esp8266Test-2.zip)
[image]
[image]

devyte, 2019-05-23:
Which core version? is it really 2.4.1?

dsptech, 2019-05-23:
My git client report the commit: 968d6fc.  Is there another way to check it by inspecting the sources ?
I investigated a bit more. The latency happen just before the "CONNECTED" event

[image]

devyte, 2019-05-23:
> the issue match exactly the referenced issue
> commit: 968d6fc

So not the same. 
This issue is currently non compliant, and there's no way to tell at a glance what it's about. Please open a new issue, and follow the instructions, especially about including the MCVE instead of attaching a zip file.

closed 2019-05-23 by devyte

dsptech, 2019-05-23:
Ok,
as you prefer ....