esp8266/Arduino #6133: Upgrade to 2.5.2 = typedef conflict between spiffs_config.h and lwip-err-t.h

ericbeaudry, 2019-05-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			[Wemos Mini]
- Core Version:			[2.5.2]
- Development Env:		[Platformio|Visual Micro]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description
I'm attempting to upgrade to Core 2.5.2 (coming from 2.4.1) I cannot compile the project.
Same code builds without problems when using 2.4.1
I've reduce the sample to the minimum code possible to facilitate bug hunting.

On 2.4.1
Compiling 'test252' for 'WeMos D1 R2 & mini'
Program size: 248,120 bytes (used 24% of a 1,044,464 byte maximum) (2.12 secs)
Minimum Memory Usage: 32484 bytes (40% of a 81920 byte maximum)

On 2.5.2
spiffs.h:14: In file included from
test252.ino:9: from
 
spiffs_config.h: 21:17: error: conflicting declaration 'typedef int32_t s32_t
   typedef int32_t s32_t
 
Error compiling project sources
cc.h:37: In file included from
Build failed for project 'test252'
arch.h:48: from
debug.h:40: from
lwipopts.h:3581: from
opt.h:51: from
init.h:40: from
IPAddress.h:27: from
ESP8266WiFi.h:31: from
ESP8266WebServer.h:29: from
test252.ino:8: from
 
lwip-err-t.h: 8:28: error: 's32_t' has a previous declaration as 'typedef long int s32_t
   typedef signed     long    s32_t
 
spiffs.h:14: In file included from
test252.ino:9: from
 
spiffs_config.h: 22:18: error: conflicting declaration 'typedef uint32_t u32_t
   typedef uint32_t u32_t
 
cc.h:37: In file included from
arch.h:48: from
debug.h:40: from
lwipopts.h:3581: from
opt.h:51: from
init.h:40: from
IPAddress.h:27: from
ESP8266WiFi.h:31: from
ESP8266WebServer.h:29: from
test252.ino:8: from
 
lwip-err-t.h: 7:28: error: 'u32_t' has a previous declaration as 'typedef long unsigned int u32_t
   typedef unsigned   long    u32_t

I've tried with PlatformIO and VisualMicro both IDE are reporting the same error. And since all there is in the test program are 3 includes, I really don't know what I can be doing wrong.

Thanks for looking into this.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <Arduino.h>
#include <ESP8266WebServer.h>
#include <spiffs/spiffs.h>

void setup() {}

void loop() {}

### Debug Messages

```
Debug messages go here
```

d-a-v, 2019-05-23:
SPIFFS instance is global and defined in implicitly included `<FS.h>`.
Is there an issue when not using your `#include <spiffs/spiffs.h>` line ?

ericbeaudry, 2019-05-23:
Well that was simple! Removing that #define was all that was required. 
Thanks for the quick help. You can close the issue if you what.

closed 2019-05-23 by ericbeaudry

filimonic, 2020-07-03:
Thank you. 
Wiping out `#include <spiffs/spiffs.h>` from project helps.