esp8266/Arduino #6138: Net connection broken in ESP8266WiFi library Version 2.5.0 or later

SteveRMann, 2019-05-24:
----------------------------- Delete below -----------------------------
----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below. (As best as I can)

#### Platform

- Hardware:				NodeMCU, Wemos D1 mini
- Core Version:			See my description
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Lolin(Wemos D1) R2 & mini|NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:			not shown in tools
- Flash Size:                     4M(1M SPIFFS)
- lwip Variant:                  v2 Lower Memory
- Reset Method:		not shown in tools
- Flash Frequency:	        not shown in tools
- CPU Frequency:		80Mhz
- Upload Using:		not shown in tools
- Upload Speed:              921600

### Problem Description

I spent two days chasing this down, but you get the workaround in 2 minutes.  I suspect that this problem may be unique to my environment, but here is what I found.

I have a few Wemos D1 Mini boards and NodeMCU boards that were working on the local WiFi yesterday, yet today none of them will connect after compiling and uploading to the Wemos or NodeMCU. More puzzling is that I can flash Tasmota on them and they all connect to the WiFi, but none with this code that was working yesterday.  I recalled after a day of troubleshooting that I did a libraries upgrade.

I modified an example sketch (below) that just tries to connect to my local network, and instead of the marching dots, I show the WiFi.status().

```
 while (WiFi.status() != WL_CONNECTED) { 
    delay(1000);
    Serial.print(WiFi.status()); Serial.print(' ');
  }

```
I have determined that if I use ESP8266WiFi library Version 2.5.0 or later, this is what the sketch below returns:

```
Connecting to Kaywinnet ...
6 6 6 6 6 6 6 6 1 1 1 1 1 1 1 1 1 1 1 1 1 ... forever 

```

(6=WL_DISCONNECTED, and 1=WL_NO_SSID_AVAIL)

By reverting to ESP8266WiFi library Version 2.4.2, I can get a connection:

```
Connecting to Kaywinnet ...
6 6 6 6 6 6 3 

Connection established!
IP address:	192.168.1.36
Mode: STA+AP
PHY mode: N
Channel: 6
AP id: 0
Status: 5
Auto connect: 1
SSID (9): Kaywinnet
Passphrase (10): 806194xxxx
BSSID set: 0

```
(3=WL_CONNECTED)

### Debug Messages

None

Additional information:
My router is an Actiontec MI424WR on Verizon Wireless.
I don't know if it's related, but I have had a connection problem from Raspberry Pi's which was resolved by removing the quotes from the wpa_supplicant.conf file:

```
network={
        ssid="Kaywinnet"
        key_mgmt=NONE
        wep_key0=806194xxxx
}

```

philbowles, 2019-06-03:
watching this: I have a (possibly) similar problem

SteveRMann, 2019-06-03:
> watching this: I have a (possibly) similar problem

Which problem?  That ESP8266WiFi library Version 2.5.0 or later will not connect to WiFi?
If so, I am happy that I am not the only one experiencing this problem.

manumdk, 2019-06-03:
I've the same problem, did you solve the lost of connections ?

d-a-v, 2019-06-03:
Please try version 2.5.2, and use the tool menu option "Erase All Flash".
Once is sufficient, the option needs to be set back to "Only Sketch" by hand.
Please report back.

SteveRMann, 2019-06-03:
I'm not sure to whom you are directing the request, but here is my test sketch:

```
#include <ESP8266WiFi.h>

const char* ssid = "Kaywinnet";
const char* password = "806194edb8";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println('\n');

  WiFi.disconnect();
  delay(100);

  WiFi.begin(ssid, password);
  Serial.print("Connecting to ");
  Serial.print(ssid);
  Serial.println(" ...");

  while (WiFi.status() != WL_CONNECTED) { 
    delay(1000);
    Serial.print(WiFi.status()); Serial.print(' ');
  }

  Serial.println('\n');
  Serial.println("Connection established!");
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());
  WiFi.printDiag(Serial);
  Serial.println();
}

void loop() {}
```

I installed esp8266 Version 2.5.2, flashed (selecting  "all flash contents").  This is what I get:

> Connecting to Kaywinnet ...
> 6 6 6 6 6 6 6 6 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 

Reinstalled Version 2.4.2, flashed as above, and this is what I get:

> Connecting to Kaywinnet ...
> 6 6 6 6 6 6 3 
> 
> Connection established!
> IP address:	192.168.1.58
> Mode: STA+AP
> PHY mode: N
> Channel: 6
> AP id: 0
> Status: 5
> Auto connect: 1
> SSID (9): Kaywinnet
> Passphrase (10): 806194xxxx
> BSSID set: 0
>

cfarny, 2019-06-05:
I have the similar problem with my Feather Huzzah. It will successfully connect via the HTTPSRequest example when using the 2.4.2 library but not with any of the higher versions. I tried the Erase Flash option 'All Flash Contents' initially and switched to 'Only Sketch' afterwards. It will connect to my WiFi network but not to api.github.com ("connection failed").

jandrassy, 2019-06-05:
I believe this problem is caused by ignoring RF settings area on flash. The AT firmware reference has:

> RF parameter values, has to be downloaded into flash at least once. If the RF_CAL parameter area is initialized with blank.bin, this esp_init_data_default.bin has to be downloaded too.

The location of RF parameters area depends on the size of the flash. The application must be flashed with the right flash size for SDK to find the RF parameters area.

The esp8266 uses the values in esp_init_data_default.bin as start values for RF calibration. Sometimes It can calibrate starting with wrong values, sometimes not.

SteveRMann, 2019-06-05:
Jandrassy- perhaps I wasn't clear.  I have tried the experiment on several Wemos D1 Mini's and some NodeMCU boards.  I've tried Erase Flash: "Only Sketch", "Sketch + WiFi Settings" and "All Flash Contents".
In all cases no board will connect to WiFi if ESP8266 Version 2.5.0 or later is installed.  All boards will connect just fine with ESP8266 Version 2.4.2 installed.

Something has changed in the library.

jandrassy, 2019-06-05:
the esp8266 core uses Espressif SDK. 2.4.2 has a different SDK version then 2.5.0

SteveRMann, 2019-06-05:
And, that fixes the problem, how?
Look, I am a lowly Arduino IDE user, not a developer.  What surprises me is that there are so few with this issue.

jandrassy, 2019-06-05:
> And, that fixes the problem, how?
> Look, I am a lowly Arduino IDE user, not a developer. What surprises me is that there are so few with this issue.

did you try 2.5.2?

SteveRMann, 2019-06-05:
Yes, and I posted my test sketch and the results above. https://github.com/esp8266/Arduino/issues/6138#issuecomment-498453624

jandrassy, 2019-06-06:
did you try without `WiFi.disconnect();` or with `WiFi.setAutoConnect(false);`

SteveRMann, 2019-06-06:
Didn't fix it.  And I did use "Erase Flash: All Flash Content".  Tried it on two different boards (a NodeMCU and a Wemos D1 Mini).
2.4.2 works, 2.5.2 does not.

jandrassy, 2019-06-07:
can you try WPA authentication?

SteveRMann, 2019-06-08:
> can you try WPA authentication?

This is a home router- there is no username or password.  Or am I not understanding WPA?  I couldn't find any examples using Professor Google.

jandrassy, 2019-06-08:
WEP has a passphrase, WPA has password. both are provided as second parameter in WiFi.begin. you could try a smartphone as WiFi hotspot

SteveRMann, 2019-06-08:
Yes. Version 2.5.2 does connect to my phone as a hot spot.  But not to my home network.  Does this give a clue what is happening?

I went to my router to see if there was a setting for WPA, and it was hidden under four different pages on the router.  I turned on WPA, which turned off WEP automatically.  To maintain some compatibility, I gave it an ascii password that was the same as the WEP passphrase.  So, yes, not V2.5.2 does connect to my network.  But that created all new problems.  I have 65 devices on my IP scan, and after turning on WPA, only 20 of them would reconnect over WPA.  Fixing that would be a full-day job in itself because many of these devices are inside wall-boxes, inside of appliances and generally inconvenient to get to the flash pins.  Then I would have to find the sketches for each device that aren't running Tasmota.

So, in summary, you are saying that **V2.5.X of ESP8266WIFI.h does not support WEP?**

jandrassy, 2019-06-08:
it should support WEP. how long is your paraphrase? 64 characters? or shorter?

SteveRMann, 2019-06-09:
Thanks for working with me in this issue.
My WEP passphrase is 10 hex characters.

When I tried setting up WPA on the router I used the same hex characters as an ASCII password.  I really, really want to avoid reprogramming all of the devices on my network.

cfarny, 2019-06-13:
None of these suggestions fix the problem of 2.5.2 (or anything higher than 2.4.2) preventing my feather huzzah from connecting to a known website, through either my phone hotspot or my home network (using WPA). I've tried the Generic ESP8266 profile as well as the Adafruit Feather Huzzah profile that corresponds to my device.  

Any other suggestions out there? Could this be related to the manufacture date/version of my device?

jandrassy, 2019-06-13:
> None of these suggestions fix the problem of 2.5.2 (or anything higher than 2.4.2) preventing my feather huzzah from connecting to a known website, through either my phone hotspot or my home network [...]

and 2.4.2 works?

cfarny, 2019-06-13:
Yes, 2.4.2 works for http, on my phone hot spot & home router. I'd like to upgrade to 2.5.2 because I'm unable to connect to some secure sites via HTTPS, and I'm hoping the newer libraries resolve some of these related issues.

danbates2, 2019-06-13:
DHPC is where 2.5.2 gets stuck for me. The AP broadcast by the ESP shows on my laptop, but I can't connect to it. 2.4.2 works fine.
https://github.com/openenergymonitor/EmonESP/blob/master/src/wifi.cpp

d-a-v, 2019-06-17:
@SteveRMann 
Can you try to call
```
WiFi.enableInsecureWEP();
```
before `WiFi.begin(...)` ?

SteveRMann, 2019-06-18:
calling 
`WiFi.enableInsecureWEP();`
before WiFi.begin() appears to have fixed this issue. At least in my brief test.

Thanks for pointing me to this new member of the class.  I did a quick browse of the docs and don't see this anywhere.

closed 2019-06-18 by SteveRMann