esp8266/Arduino #6140: Board used:- ESP8266 - WeMos

jshankar494, 2019-05-24:
Board used:- ESP8266 - WeMos

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffbb0 end: 3fffffc0 offset: 01b0
3ffffd60:  3fffdad0 00000001 3ffef4dc 3ffef60c  
3ffffd70:  3fffdad0 00000001 3ffef4dc 3ffef60c  
3ffffd80:  3ffef200 00000000 3ffef4dc 401004be  
3ffffd90:  feefeffe 00000000 3ffef4dc 40203fc4  
3ffffda0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdc0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdd0:  feefeffe feefeffe feefeffe feefeffe  
3ffffde0:  feefeffe feefeffe feefeffe feefeffe  
3ffffdf0:  feefeffe feefeffe feefeffe feefeffe  
3ffffe00:  feefeffe feefeffe feefeffe feefeffe  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
3ffffe60:  feefeffe feefeffe feefeffe feefeffe  
3ffffe70:  feefeffe feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
3ffffeb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  feefeffe feefeffe feefeffe feefeffe  
3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
3fffff00:  feefeffe feefeffe feefeffe feefeffe  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef60c  
3fffffa0:  3fffdad0 00000000 3ffef5dc 4020c264  
3fffffb0:  feefeffe feefeffe 3ffe8570 401007a9  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vac02aff5
~ld
ISR not in IRAM!

Please help me to figure out the issue.

Thanks

_Originally posted by @jshankar494 in https://github.com/esp8266/Arduino/issues/5621#issuecomment-495477028_

5chufti, 2019-05-24:
what is the "WeMos development kit" ???
Did you not see the questionair when opening the issue?
see #6127 and close this non-issue

closed 2019-05-24 by jshankar494

yheynema, 2019-06-12:
As a hit, and may help, please refer to https://github.com/sandeepmistry/arduino-LoRa/issues/254
or even a longer discussion at:  #6127 
But I'm not convinced this the best fix, I mean having to fix all libraries in the world... :-s

franmagyar, 2019-07-26:
Try adding at the begining:

void ICACHE_RAM_ATTR handleInterrupt ();

It resolved the issue in my case

arfa123, 2019-08-15:
> Try adding at the begining:
> 
> void ICACHE_RAM_ATTR handleInterrupt ();
> 
> It resolved the issue in my case

how can we call that handleInterrupt ?
attachInterrupt(SWITCH_1, handleInterrupt , CHANGE);

void ICACHE_RAM_ATTR handleInterrupt ();{
}

arfa123, 2019-08-15:
the ' handleInterrupt ' function should defined before the ' attachInterrupt ()'