esp8266/Arduino #6142: Can not use attachinterrupt on ESP8266 and variants?

maikbasso, 2019-05-25:
Can not use interrupts because of the condition on this link below. With this condition, when using interrupts the hardware goes into an infinite loop of reboots. If we commenting this condition, the hardware works normally using interruptions.

https://github.com/esp8266/Arduino/blob/0a8f2a13a6625f5d4e9f83ac0f0107e2d62cdc6a/cores/esp8266/core_esp8266_wiring_digital.cpp#L177

d-a-v, 2019-05-25:
There is a very recent addition that requires ICACHE_RAM_ATTR in front of your interrupt function definition.

https://github.com/esp8266/Arduino/blob/master/doc/reference.rst#digital-io (end of paragraph)
https://arduino-esp8266.readthedocs.io/en/latest/reference.html#digital-io
#6141

devyte, 2019-05-25:
To be clear, the requirement has always been there. The referenced condition was added to enforce it, otherwise it was a matter of cache luck whether it would work or not, and hence we'd get a bunch of issues from users who weren't using it. 
The referenced PR just documents the requirement.
Closing as not an issue.

closed 2019-05-25 by devyte