esp8266/Arduino #6143: BearSSL::WiFiClientSecure::connect() throws Exception

braiden, 2019-05-26:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2d9253e46c41ef6949759ec028a356917d289831
- Development Env:		Platformio
- Operating System:		Debian Testing

### Settings in IDE

- Module:  			Wemos D1 mini
- Flash Mode:			dout
- Flash Size: 		4MB
- lwip Variant:        -DLWIP_OPEN_SRC -DLWIP_FEATURES=1 -DLWIP_IPV6=0 
- Reset Method:		nodemcu
- Flash Frequency:	80Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   1M

### Problem Description

BearSSL::WiFiClientSecure::connect() throws Exception while connecting. The issue happens about 3/10. So the sketch below fails in a few seconds. Both a hostname of "mqtt.2030.ltsapis.goog" and a call to configTime() are required or the bug doesn't happen (or happens much much less often.)

Also, an exception is not always thrown, sometimes connect() completes successfully but memory elsewhere not owned by wifi client is corrupted. I could provide another sketch demonstrating this, but thought the exception more useful.

The sketch omits X509List, same issue happens with valid certs.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <time.h>

WiFiClientSecure *client;

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.hostname("test");
  WiFi.begin("redacted", "redacted");
  while (WiFi.status() != WL_CONNECTED) delay(10);
  Serial.println("WiFi Connected");

  configTime(0, 0, "pool.ntp.org");
  while (time(nullptr) < 1514764800) delay(10);
  Serial.println("NTP Time Synced");

  client = new WiFiClientSecure();
}

void loop() {
  Serial.print(".");
  client->connect("mqtt.2030.ltsapis.goog", 443); // BAD
  // client->connect("google.com", 443); // OK
  client->stop();
  delay(100);
}
```

### Debug Messages

```
WiFi Connected
NTP Time Synced
..
Exception (28):
epc1=0x4021292e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000013 depc=0x00000000
```

```
Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x4021292e: pbuf_free_LWIP2 at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c:755
0x4020eda0: esp2glue_pbuf_freed at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:250
0x4023e157: pp_tx_idle_timeout at ??:?
0x40248292: pbuf_wrapper_release at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:247
 (inlined by) pbuf_free at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:708
0x4023def1: ppCheckTxIdle at ??:?
0x4023dfb0: ppCheckTxIdle at ??:?
0x4023d8ce: ppPeocessRxPktHdr at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000e19: ?? ??:0
0x40001878: ?? ??:0
0x40104214: call_user_start_local at ??:?
0x4010421a: call_user_start_local at ??:?
0x4010000d: call_user_start at ??:?
0x40100484: cont_ret at cont.S.o:?
0x40100425: cont_continue at cont.S.o:?
0x40253f80: sleep_reset_analog_rtcreg_8266 at ??:?
0x4021aaa0: sha2big_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c:165
 (inlined by) br_sha384_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c:207
0x4021b30b: sha2small_out at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c:249
0x40253cc4: sleep_reset_analog_rtcreg_8266 at ??:?
0x40227a99: br_multihash_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/multihash.c:141
0x4000050c: ?? ??:0
0x4023eb87: pp_attach at ??:?
0x4023ebd6: pp_attach at ??:?
0x4023ece2: pp_attach at ??:?
0x4023dc7f: ppTxPkt at ??:?
0x4023104b: ieee80211_output_pbuf at ??:?
0x40104793: wdt_feed at ??:?
0x40100d06: pp_post at ??:?
0x4020ea5d: glue2esp_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:299
0x4020ecdf: new_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:235
0x4020f0d0: ethernet_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:312
0x4020ea5d: glue2esp_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:299
0x40216600: etharp_output_to_arp_index at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:770
0x40100d06: pp_post at ??:?
0x40100d06: pp_post at ??:?
0x40103313: lmacRecycleMPDU at ??:?
0x40216854: etharp_output_LWIP2 at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:885
0x40101bfa: trc_NeedRTS at ??:?
0x40217268: ip4_output_if_opt_src at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1007
0x40217d44: mem_malloc at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:210
0x40101dd4: wDev_ProcessFiq at ??:?
0x402172b0: ip4_output_if_opt at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:820
0x40202019: WiFiClient::read(unsigned char*, unsigned int) at ??:?
0x402172d6: ip4_output_if at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:793
0x40217e4f: ip_chksum_pseudo at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c:395
0x40213733: tcp_output_control_segment at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:1956
0x40213629: tcp_output_alloc_header at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:1863
0x40213f7c: tcp_send_empty_ack at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:2057
0x4021406a: tcp_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:1319
0x4020614c: _umm_free at umm_malloc.cpp:?
0x40206574: free at ??:?
0x40217d68: mem_free at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:237
0x4021296e: pbuf_free_LWIP2 at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c:786 (discriminator 1)
0x40204c6a: yield at ??:?
0x40204cb5: optimistic_yield at ??:?
0x40203079: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at ??:?
0x4021d148: br_ssl_hs_client_run at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/ssl/ssl_hs_client.c:958
0x4021d0f0: pgm_read_word_inlined at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h:67
 (inlined by) br_ssl_hs_client_init_main at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/ssl/ssl_hs_client.c:904
0x4021cbfe: br_ssl_engine_hs_reset at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/ssl/ssl_engine.c:1305
0x40203328: BearSSL::WiFiClientSecure::_wait_for_handshake() at ??:?
0x40203520: BearSSL::WiFiClientSecure::_connectSSL(char const*) at ??:?
0x40204b7c: esp_yield at ??:?
0x4020511b: delay at ??:?
0x4020215d: WiFiClient::connect(IPAddress, unsigned short) at ??:?
0x4020363d: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at ??:?
0x40206e30: __run_user_rf_pre_init() at ??:?
0x40206e30: __run_user_rf_pre_init() at ??:?
0x4020110e: loop at ??:?
0x40204c30: loop_wrapper() at core_esp8266_main.cpp:?
```

braiden, 2019-05-26:
Update:

I replaced `tools/sdk/lib/libbearssl.a` with a version I built from https://www.bearssl.org/git/BearSSL@e4edfb84eda32c3b3aa548f975c4a13d406db989

Issue seems to be fixed. Maybe BearSSL lib needs to be updated? Or, maybe my library was compiled with different flags? I did:

```sh
make 
  CC=~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc \
  LD=~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-ld \
  AR=~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-ar \
  LDDLL=~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc 
```

and copied ~/build/libbearssl.a to tools/sdk/lib/libbearssl.a.

braiden, 2019-05-26:
Update 2:

Spoke too soon. With my version of `libbearssl.a` my real sketch doesn't working anymore. Aborts every time at BearSSL::WiFiClientSecure::_connectSSL(). So i'm not sure my previous comment is meaningful.

earlephilhower, 2019-05-27:
@d-a-v, I'm seeing this crashing inside LWIP wrappers.  Any ideas?

braiden, 2019-05-27:
I've been looking a lot at exception too, and think its misleading. My theory, and some evidence below, is the exception happens because bearssl corrupts LWP memory and causes the eventual crash. Instead I've been using a second sketch based on my comment: "sometimes connect() completes successfully but memory elsewhere not owned by wifi client is corrupted".

Seems like `br_ssl_engine_recvrec_ack` is writing out-of-bounds somehow. The following sketch demonstrates the issue:

The idea is `head` and `tail` are zero'd memory. Each loop, I check that that's still true, and if not, dump the contents. So eventually to program dumps non-zero content. When the corruption happens the corrupted address is mostly predicated, so I chased it further with gdb watch points to narrow down what was changing the values.

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <time.h>
//#include <GDBStub.h>

static const int pad = 768;
char *head;
WiFiClientSecure *client;
char *tail;

void hexDump(char *addr, int size) {
  for (char *row = (char *)(uint32_t(addr) & 0xfffffff0); row < addr + size; row += 16) {
    Serial.printf("%08x: ", (uint32_t) row);
    for (char *col = row; col < row + 16; col++) {
      if ((col < addr) || (col > addr + size)) {
        Serial.print("__");
      } else {
        Serial.printf("%02x", *col);
      }
    }
    Serial.printf("\r\n");
  }
  Serial.println();
}

void setup() {
  Serial.begin(115200);
  //gdbstub_init();

  WiFi.mode(WIFI_STA);
  WiFi.hostname("...");
  WiFi.begin("braiden", "...");
  while (WiFi.status() != WL_CONNECTED) delay(10);
  Serial.println("WiFi Connected");

  tail = (char *) calloc(pad, sizeof(char));
  client = new WiFiClientSecure();
  head = (char *) calloc(pad, sizeof(char));

  client->setX509Time(1558874730);
  Serial.printf("head=%08x\r\n", head);
  Serial.printf("client=%08x\r\n", client); 
  Serial.printf("tail=%08x\r\n", tail);
  Serial.println();
}

void loop() {
  Serial.print(".");
  if (memcmp(head, tail, pad)) {
    hexDump(head, pad);
    hexDump((char *)client, sizeof(WiFiClientSecure));
    hexDump(tail, pad);
    while (true) delay(1000);
  }
  client->connect("mqtt.2030.ltsapis.goog", 443);
  delay(1000);
}

```

Turns out `br_ssl_engine_recvrec_ack` corrupts the memory

WiFiClientSecureBearSSL.cpp:
```cpp
 523           Serial.printf("br_ssl_engine_recvrec_ack: %08x\r\n", *(uint16_t *)(0x3ffefe90));
 524           br_ssl_engine_recvrec_ack(_eng, rlen);
 525           Serial.printf("br_ssl_engine_recvrec_ack: %08x\r\n", *(uint16_t *)(0x3ffefe90));
```

Prints:
```
br_ssl_engine_recvrec_ack: 00000000
br_ssl_engine_recvrec_ack: 00009045
```

`0x3ffefe90` is part of the memory allocated by `tail = calloc()` so it should not be changed by `br_ssl_engine_recvrec_ack`.

d-a-v, 2019-05-27:
I tried OP sketch and it works no error if
- running @160Mhz
- and selecting basic SSL ciphers

Otherwise, I get WDT.

In that type of sketch, making a (SSL or not) connection to a server every 0.1s greatly increases the chances to be temporarily banned, making things worse to debug.

braiden, 2019-05-27:
"... and selecting basic SSL ciphers".

Does that imply -DBEARSSL_SSL_BASIC? I was also just trying that too. And it leads me to suspect the reason `mqtt.googleapis.com` works reliably but `mqtt.2030.ltsapis.goog` does not is a difference in ciphers?

`mqtt.googleapis.com` works regardless of `DBEARSSL_SSL_BASIC`.

`mqtt.2030.ltsapis.goog` never works with `-DBEARSSL_SSL_BASIC`. It throws no exception and has no negative side-effects, but the MqttClient wrapping the connection disconnects immediately. With `-DBEARSSL_SSL_BASIC` unset, it works completely or crashes.

Thanks both for your help!

d-a-v, 2019-05-27:
my tests are run on `mqtt.2030.ltsapis.goog` like in OP.

> mqtt.2030.ltsapis.goog never works with -DBEARSSL_SSL_BASIC

Have you tried at 160MHz (option in menu) ?
Are you using Arduino IDE ?

braiden, 2019-05-27:
I'm using platformio with `build_flags = -DBEARSSL_SSL_BASIC` and `board_build.f_cpu = 
160000000L`.

Running the basic connect loop sketch with `-DDEBUG_ESP_SSL` and without `-DBEARSSL_SSL_BASIC` will either print the expected `BSSL:Couldn't connect. Error = 'Chain could not be linked to a trust anchor.'` or crash.

Adding `-DBEARSSL_SSL_BASIC` the sketch only prints `BSSL:Couldn't connect. Error = 'Unknown error code.'`. It never crashes, but it also never connects. (The first version of code will complete connection if I provide trush chain).

braiden, 2019-05-27:
FYI, https://cloud.google.com/iot/docs/how-tos/mqtt-bridge#using_a_long-term_mqtt_domain cites `TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256` as the cipher suite for this server.

That's disabled if -DBEARSSL_SSL_BASIC is set.

earlephilhower, 2019-05-27:
Your sample code can never actually connect to anything. No trust anchor was set, and `setInsecure` wasn't used.  Enabling debugging will give you the warning.  See the BSSL_Validation example for more info.

That's a "no trust anchor" error and should not cause a crash, though.

You'll also want to be at 160mhz, too, or you'll take too long to connect and be punted off most servers.

earlephilhower, 2019-05-27:
> Seems like br_ssl_engine_recvrec_ack is writing out-of-bounds somehow. The following sketch demonstrates the issue ...

Thanks, let me look at that more closely now.  That part of the code has been pretty stable for a while, but your constant failing-to-connect may be exercising it in a different way than others' code...

earlephilhower, 2019-05-27:
Yes, it is stack overflow.  Very good debugging, @braiden!
Add "#include <StackThunk.h>" to the top and change your printf to "  Serial.printf("%d.", stack_thunk_get_max_usage());" and you see that it stabilizes at 5700 bytes.

I only allocate 5600 bytes, so there is your overwrite.

It's a 1-line fix in cores/esp8266/StackThunk.cpp.  Change 5600 to 5700 and your crashes should disappear.  I'll do a PR ASAP.

earlephilhower, 2019-05-27:
@braiden, can you confirm you're fixed with the latest GIT head?  See above about the connection requirements (setting a CA/insecure), but there shouldn't be a crash.

closed 2019-05-27 by earlephilhower

braiden, 2019-05-27:
Great, Thanks for the quick fix!

Verified at ...d83eabe
My test sketch didn't crash even after > 60 connects.
My real sketch connects and validates cert completely.

braiden, 2019-05-28:
I'm not sure why when I add a `NeoPixelBrightnessBus` to the code in this comment https://github.com/esp8266/Arduino/issues/6143#issuecomment-496061829 I start seeing corruption again.

`#define _stackSize (6000/4)` and it works now, but I'm not how to choose the right minimal value.

```cpp
NeoPixelBrightnessBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> pxl(1);

void setup() {
  Serial.begin(115200)
  pxl.Begin();
  ...
```

earlephilhower, 2019-05-28:
There is another PR just merged which puts a canary on the end of the stack and checks it every time BearSSL returns.  You could pull git head and give a whirl by dropping for your specific needs and seeing if it triggers.

As for the stack issue, NeoPixel eats a large amount of stack space in its interrupt handler.  And when you're doing SSL and an interrupt happens it uses the BearSSL stack.

braiden, 2019-05-28:
Makes sense. I don't need a NeoPixel so I'll just drop that for more predictable stack usage. Thanks again!

I'm already one latest commit. I saw "Abort Called", but didn't see "FATAL ERROR: BSSL stack overflow" print. So wasn't sure if that was the canary or not:

I'm happy to not use the skip NeoPixel, so i'm good, but let me know if there's any more testing i can do.

```
Abort called

>>>stack>>>

ctx: bearssl
sp: 3fff35a4 end: 3fff3764 offset: 01b0
3fff3754:  deadbeef 00000000 3fff1b94 40205178  

ctx: cont
sp: 3ffffdf0 end: 3fffffc0 offset: 0000
3ffffdf0:  00000008 00000000 00000001 4020835d  
3ffffe00:  00000609 00000005 93a5ccce 00000000  
3ffffe10:  40236248 402361f0 3fff4944 40235cfe  
3ffffe20:  3fff4e44 3fff1954 3fff4944 00000001  
3ffffe30:  00000001 00000001 3fff1b94 40208584  
3ffffe40:  3fff1954 00000000 3fff1b94 4020877c  
3ffffe50:  00000000 3ffffeb0 40211350 3fffefa0  
3ffffe60:  00000000 00000000 00000000 40211a77  
3ffffe70:  00000000 3fff3c2c 3fff1b94 402072bd  
3ffffe80:  000001bb 00000d50 00000000 3fff3aac  
3ffffe90:  000001bb 3fff1b94 3fff1954 3fff3aac  
3ffffea0:  000001bb 3fff1b94 3fff1954 40208899  
3ffffeb0:  4021f228 cec97d4a 4021f228 cec97d4a  
3ffffec0:  3fff465c 3ffe86f9 00000000 4020b47c  
3ffffed0:  3ffe88f5 00000000 3ffe88f4 4021309e  
3ffffee0:  3ffe86f8 00000000 3ffe86f7 00000001  
3ffffef0:  3ffefe8c 00000016 3ffefe8c 00000001  
3fffff00:  3fff4584 000000c2 3fffff50 40210252  
3fffff10:  3ffe88f3 ffffff5c 3fffff50 402102b2  
3fffff20:  3fff449c 3fff4584 3fffff50 40210478  
3fffff30:  4020eb04 00000000 3fff18e4 00000001  
3fffff40:  3ffefe8c 00000000 3fff18e4 40202c16  
3fffff50:  3fff465c 00c200cf fffef244 3fff449c  
3fffff60:  0047004f ff000000 00001388 4021ae42  
3fffff70:  3fff1904 3fff449c 3fff3aac 3fffff5c  
3fffff80:  3fffdad0 3ffef360 3ffefe8c 3ffeffbc  
3fffff90:  3fffdad0 3ffef360 3ffefe8c 40204f9b  
3fffffa0:  3fffdad0 00000000 3ffeff8c 40211404  
3fffffb0:  feefeffe feefeffe 3ffe8550 40100b3d  
<<<stack<<<
```

PR #6153 (2019-05-27): Expand BSSL stack to 5750 bytes
Fix #6143 which found a cipher combination which overran the old limit
of 5600 bytes (it required 5700 bytes).
commit d83eabe5 (2019-05-27): Expand BSSL stack to 5750 bytes (#6153)

Fix #6143 which found a cipher combination which overran the old limit
of 5600 bytes (it required 5700 bytes).