esp8266/Arduino #6146: BearSSL::WiFiClientSecure::_installClientX509Validator() crash (related to shared_ptr_base.h)

wvmarle, 2019-05-26:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-07]
- Core Version:			 [2d9253e]
- Development Env:		[Arduino IDE]
- Operating System:		[Mint]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout(compatible)]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [460800] (serial upload only)

### Problem Description
My project is sending out data to an external server (https connection, station mode - ESP connecting to my home network), and is queried by an Android app (http connection, using the access point mode - Android connecting to the ESP's WiFi network).

Sometimes (but it seems not always - the issue is hard to reliably reproduce but the stack trace is always the same) when the app contacts the ESP while the ESP is connected over https to the external server, a crash occurs:

0x402222c5: BearSSL::WiFiClientSecure::_installClientX509Validator() at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h line 986
:  (inlined by) ?? at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h line 316
:  (inlined by) ?? at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h line 598
:  (inlined by) ?? at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h line 614
:  (inlined by) BearSSL::WiFiClientSecure::_installClientX509Validator() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 877
0x40222cfc: BearSSL::WiFiClientSecure::_connectSSL(char const*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 962
0x40219c70: esp_yield at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 91
0x4021a8c3: delay at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring.cpp line 54
0x40206d6d: WiFiClient::connect(IPAddress, unsigned short) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/include/ClientContext.h line 136
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 170
0x40222eed: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
0x40225230: BearSSL::PrivateKey::getEC() const at ?? line ?
0x40225230: BearSSL::PrivateKey::getEC() const at ?? line ?
0x40216b5c: HTTPClient::connect() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1165
0x4021c16e: uart_write at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/uart.cpp line 498
0x40217820: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 655
0x40217d98: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
0x40273eba: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x402180a5: Print::write(char const*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.h line 60
0x40217d83: HardwareSerial::write(unsigned char) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 154
0x402179ba: HTTPClient::GET() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 575
0x402118e8: HydroMonitorLogging::sendPostData(char*) at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorLogging.cpp line 730
0x40225470: BearSSL::PrivateKey::getEC() const at ?? line ?
0x4022c4c0: _vsprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsprintf.c line 65
0x40217d98: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
0x4022000a: spiffs_object_truncate at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/spiffs/spiffs_nucleus.cpp line 1727
0x40211d41: HydroMonitorLogging::transmitMessages() at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorLogging.cpp line 337
0x40211caa: HydroMonitorLogging::transmitMessages() at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorLogging.cpp line 326
0x40204dab: handleAPI() at /home/wouter/Arduino/Williams_fridge/Fridge_control/webAPI.ino line 149
0x402251c8: BearSSL::PrivateKey::getEC() const at ?? line ?
0x40217d98: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
0x40218130: Print::println() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.cpp line 178
0x402121b5: HydroMonitorLogging::logData() at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorLogging.cpp line 250
0x40217d83: HardwareSerial::write(unsigned char) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 154
0x40224948: Print::write(char) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.h line 73
0x40218114: Print::print(char) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.cpp line 126
0x402106f4: HydroMonitorIsolatedSensorBoard::readSensor(bool) at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorIsolatedSensorBoard.cpp line 36
0x40207a07: TwoWire::requestFrom(int, int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Wire/Wire.cpp line 134
0x402042f9: loop at /home/wouter/Arduino/Williams_fridge/Fridge_control/loop.ino line 36
0x40219d20: loop_wrapper() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 125
0x401015d5: cont_wrapper at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 81

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

wvmarle, 2019-05-26:
Main sketch: https://github.com/wvmarle/Fridge_control
Library that goes with it: https://github.com/wvmarle/HydroMonitor

If only I knew better where/how this crash happens I'd be glad to make a separate sketch that can reliably reproduce it, but due to the nature of outgoing & incoming connections it's very hard to do so.

wvmarle, 2019-05-26:
Another stack trace triggered in the same way:

Decoding 120 results
0x40223024: WiFiServer::_accept(tcp_pcb*, long) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/include/ClientContext.h line 41
:  (inlined by) WiFiServer::_accept(tcp_pcb*, long) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiServer.cpp line 167
0x402230b0: WiFiServer::_s_accept(void*, tcp_pcb*, long) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiServer.cpp line 181
0x40230e16: tcp_process at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c line 942
:  (inlined by) tcp_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c line 438
0x4024f200: ieee80211_decap at ?? line ?
0x4021ca90: malloc at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 1677
0x40235f39: ip4_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 710
0x4021c84c: _umm_free at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 1304
0x4022dc75: ethernet_input_LWIP2 at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 188
0x4022daab: esp2glue_ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 441
0x40261a52: ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 363
0x40261a63: ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 371
0x4026383b: ets_snprintf at ?? line ?
0x40105440: call_user_start_local at ?? line ?
0x40105446: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x40101610: cont_ret at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 142
0x401015bd: cont_continue at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 51
0x40105c19: ets_timer_disarm at ?? line ?
0x40239c26: br_sha2small_round at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 87 (discriminator 2)
0x40239c1c: br_sha2small_round at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 85
0x40105c19: ets_timer_disarm at ?? line ?
0x40105c19: ets_timer_disarm at ?? line ?
0x40239e58: br_sha2small_round at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 102 (discriminator 2)
0x40239c1c: br_sha2small_round at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 85
0x40278bb0: chip_v6_unset_chanfreq at ?? line ?
0x40105282: lmacTxFrame at ?? line ?
0x40104e87: lmacMSDUAged at ?? line ?
0x401047fa: lmacRecycleMPDU at ?? line ?
0x40104d91: lmacMSDUAged at ?? line ?
0x4010504c: lmacProcessAckTimeout at ?? line ?
0x40103183: wDev_ProcessFiq at ?? line ?
0x40239984: sha2big_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c line 165
:  (inlined by) br_sha384_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c line 207
0x40103004: wDev_ProcessFiq at ?? line ?
0x4024697d: br_multihash_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/multihash.c line 141
0x4025cb14: ppTxPkt at ?? line ?
0x4025cb12: ppTxPkt at ?? line ?
0x4025da93: pp_attach at ?? line ?
0x4025dae2: pp_attach at ?? line ?
0x4025dbee: pp_attach at ?? line ?
0x4025cb97: ppTxPkt at ?? line ?
0x4024ff3f: ieee80211_output_pbuf at ?? line ?
0x401059bf: wdt_feed at ?? line ?
0x40101f3e: pp_post at ?? line ?
0x4022d689: glue2esp_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 299
0x4022d90b: new_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 235
0x4022dcfc: ethernet_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 312
0x40105282: lmacTxFrame at ?? line ?
0x40105282: lmacTxFrame at ?? line ?
0x40101f3e: pp_post at ?? line ?
0x40101f3e: pp_post at ?? line ?
0x40101f3e: pp_post at ?? line ?
0x4010531c: lmacTxFrame at ?? line ?
0x401031ae: wDev_ProcessFiq at ?? line ?
0x4010453f: lmacRecycleMPDU at ?? line ?
0x40102e2a: trc_NeedRTS at ?? line ?
0x401049a2: lmacRecycleMPDU at ?? line ?
0x40102e2a: trc_NeedRTS at ?? line ?
0x40104486: lmacProcessTxSuccess at ?? line ?
0x40104282: lmacProcessTXStartData at ?? line ?
0x40103307: wDev_ProcessFiq at ?? line ?
0x40103004: wDev_ProcessFiq at ?? line ?
0x40222947: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 452
0x4022293f: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 447
0x40232e5e: tcp_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c line 1319
[... 25 lines cut ...]
[... cut ...]
0x402106d8: HydroMonitorIsolatedSensorBoard::readSensor(bool) at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorIsolatedSensorBoard.cpp line 33
0x40207a07: TwoWire::requestFrom(int, int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Wire/Wire.cpp line 134
0x402042f9: loop at /home/wouter/Arduino/Williams_fridge/Fridge_control/loop.ino line 36
0x40219d20: loop_wrapper() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 125

earlephilhower, 2019-05-27:
Please give the full decoding, including the registers listing and the "last failed malloc" line.  I bet it's reporting an address somewhere, and that is the root of your problem.

I'm not able to go through your entire application and library, but this looks like an application memory leak.

wvmarle, 2019-05-27:
Understand you're not going through it all, that's too much if you don't know where to look.

Slightly different stack traces always end with:
last failed alloc call: 4022D552(1480)

An address that's not mentioned in the decoded stack trace, only lower values there. I'm not using any new() operator in this code, and am not aware of anything in my code that may cause leaks (doesn't mean it's not there, I may very well be doing something wrong).

The crashes are all linked to the actions as described above: while an outgoing connection is handled (during the https.GET() call - it's in the library, in HydroMonitorLogging::sendPostData(char*), file HydroMonitorLogging.cpp, is where it goes wrong), an incoming connection causes the program to crash. This I confirmed with lots of print statements that tell me exactly where in the code it is. Disabling that function (place a return 404 line at the top of it making, thus pretending the connection failed) makes the whole thing solid.

Abort called
03:04:59.487 -> 
03:04:59.487 -> >>>stack>>>
03:04:59.487 -> 
03:04:59.487 -> ctx: cont
03:04:59.487 -> sp: 3ffff900 end: 3fffffc0 offset: 01b0
03:04:59.487 -> 3ffffab0:  c012c008 c00dc003 3ffffbe0 402222c5  
03:04:59.487 -> 3ffffac0:  00000000 00000000 3ffffaf4 00000000  
03:04:59.487 -> 3ffffad0:  3fffb694 3fff60b4 3ffffbe0 40222cfc  
03:04:59.487 -> 3ffffae0:  00000000 3ffffb40 40219c70 3fffefa0  
03:04:59.487 -> 3ffffaf0:  00000000 00000000 00000000 4021a8c3  
03:04:59.487 -> 3ffffb00:  00000000 3fffbe44 3ffffbe0 40206d6d  
03:04:59.520 -> 3ffffb10:  000001bb 00000d50 3ffffccc 00000000  
03:04:59.520 -> 3ffffb20:  000001bb 3ffffbe0 3fffb694 00000000  
03:04:59.520 -> 3ffffb30:  000001bb 3ffffbe0 3fffb694 40222eed  
03:04:59.520 -> 3ffffb40:  40225230 5969807a 40225230 5969807a  
03:04:59.520 -> 3ffffb50:  3ffe8a18 3ffffbe0 3ffffcac 40216b5c  
03:04:59.520 -> 3ffffb60:  3ffe8976 00000000 3ffe8975 4021c16e  
03:04:59.520 -> 3ffffb70:  00000000 00000000 3ffffcac 40217820  
03:04:59.520 -> 3ffffb80:  0000edf0 0000000a 3fff0164 40217d98  
03:04:59.553 -> 3ffffb90:  40273eba 0000002e 3fff244c 402180a5  
03:04:59.553 -> 3ffffba0:  3ffe8974 3fff0164 00000000 40217d83  
03:04:59.553 -> 3ffffbb0:  00000000 fffffea0 3fff0164 00000001  
03:04:59.553 -> 3ffffbc0:  0000edf0 3ffffea0 3fff0164 402179ba  
03:04:59.553 -> 3ffffbd0:  0000edf0 3ffffea0 3fff0164 402118e8  
03:04:59.553 -> 3ffffbe0:  40225470 00000000 00003a98 0030482c  
03:04:59.553 -> 3ffffbf0:  3ffef4f8 3fffbe44 3fff535c 3fffafb4  
03:04:59.586 -> 3ffffc00:  00000000 00000000 3fff535c 00000000  
03:04:59.586 -> 3ffffc10:  00000000 00000000 00000000 00000000  
03:04:59.586 -> 3ffffc20:  00000000 3fff60b4 3fffb86c 3fffa204  
03:04:59.586 -> 3ffffc30:  3fffb66c 00000000 00000000 00000000  
03:04:59.586 -> 3ffffc40:  00004145 00000255 00000000 00000000  
03:04:59.586 -> 3ffffc50:  00000000 00000000 00000000 00000000  
03:04:59.586 -> 3ffffc60:  00000000 00000000 3ffe0001 4022c4c0  
03:04:59.586 -> 3ffffc70:  3ffffe51 00000000 7fffff1e ff000208  
03:04:59.619 -> 3ffffc80:  00000000 7fffffff 00000000 00000000  
03:04:59.619 -> 3ffffc90:  00000000 00000000 00000000 40217d98  
03:04:59.619 -> 3ffffca0:  00000000 00204800 3fff1940 00000000  
03:04:59.619 -> 3ffffcb0:  00000000 3ffffbe0 3fffb694 0016001f  
03:04:59.619 -> 3ffffcc0:  ffff2c9c 000001bb 3f001388 3fffaedc  
03:04:59.619 -> 3ffffcd0:  00c300cf ff301100 70747468 00000073  
03:04:59.619 -> 3ffffce0:  00fe8976 00000000 00000000 0021c16e  
03:04:59.653 -> 3ffffcf0:  3fffb70c 0011001f ff0016b6 00000000  
03:04:59.653 -> 3ffffd00:  00000000 00000003 00000000 00000000  
03:04:59.653 -> 3ffffd10:  00000000 ffffffff 00000000 4022000a  
03:04:59.653 -> 3ffffd20:  00000000 00000000 00000008 00000000  
03:04:59.653 -> 3ffffd30:  00000000 00000000 00000000 fffef8e5  
03:04:59.653 -> 3ffffd40:  3ffef97c 3ffef99d 3fff0164 3ffffe60  
03:04:59.653 -> 3ffffd50:  3ffefab8 3ffffea0 3ffef5b8 40211d41  
03:04:59.653 -> 3ffffd60:  3ffffeb0 3fffb814 3ffffe94 000015d0  
03:04:59.686 -> 3ffffd70:  70747468 2f2f3a73 2e777777 79746963  
03:04:59.686 -> 3ffffd80:  72647968 6e6f706f 2e736369 482f6b68  
03:04:59.686 -> 3ffffd90:  6f726479 696e6f4d 2f726f74 74736f70  
03:04:59.686 -> 3ffffda0:  64697246 61446567 702e6174 6f702f79  
03:04:59.686 -> 3ffffdb0:  72467473 65676469 61746144 6573753f  
03:04:59.686 -> 3ffffdc0:  6d616e72 72663d65 65676469 7026615f  
03:04:59.686 -> 3ffffdd0:  77737361 3d64726f 64697266 615f6567  
03:04:59.719 -> 3ffffde0:  7361705f 726f7773 6f6c2664 76656c67  
03:04:59.719 -> 3ffffdf0:  333d6c65 73656d26 65676173 5348703d  
03:04:59.719 -> 3ffffe00:  6f736e65 31302b72 2b413325 73756e75  
03:04:59.719 -> 3ffffe10:  2b6c6175 6c2b4870 6c657665 61656d2b  
[... 593 lines cut ...]
0x40217d98: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
0x40273eba: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x402180a5: Print::write(char const*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.h line 60
0x40217d83: HardwareSerial::write(unsigned char) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 154
0x402179ba: HTTPClient::GET() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 575
[... cut ...]
0x402106d8: HydroMonitorIsolatedSensorBoard::readSensor(bool) at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorIsolatedSensorBoard.cpp line 33
0x40207a07: TwoWire::requestFrom(int, int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Wire/Wire.cpp line 134
0x402042f9: loop at /home/wouter/Arduino/Williams_fridge/Fridge_control/loop.ino line 36
0x40219d20: loop_wrapper() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 125

wvmarle, 2019-05-27:
Another typical thing that I observer
- ESP8266 tries to make a connection to remote server (it's now in the server.GET() function, waiting for a response from my server - the whole call normally takes 1000-1100 ms to complete).
- My app connects to the ESP with a request for data while that GET() request is in progress.
- It hangs for some time at this point: anywhere from 4 to 124 seconds I have seen so far, varies wildly. usually this hang results in a stack dump as above, sometimes it also comes out of it, processes the incoming request, and everything continues as normal.

The GET request itself has never been logged at my server side, it didn't come through.

d-a-v, 2019-05-27:
> last failed alloc call: 4022D552(1480)

It seems like it is lwIP trying to allocate a packet (to send or receive, I can't tell)

> The crashes are all linked to the actions as described above: while an outgoing connection is handled (during the https.GET() call - it's in the library, in HydroMonitorLogging::sendPostData(char*), [...]

You can
- instrument your code with `Esp.getFreeHeap()`
- use the debug option OOM - a message will be displayed on NULL malloc/new
- use the lwip-v2 option (saves memory)

I think we need to make an MCVE with TCP and/or SSL and short memory to be able to check and fix if it leads to a crash (so far I always tried to stay far from short memory).

wvmarle, 2019-05-27:
> It seems like it is lwIP trying to allocate a packet (to send or receive, I can't tell)

Could be both, as it's supposed to both send and receive data, but neither seems to happen. It just gets stuck.

I've added the Esp.getFreeHeap() in loop() already to see what happens to the heap in every run of loop(), printing the new value if it has changed since the previous run. Values hover around 33,000, when a request comes in it drops down by about 900 bytes, which a few runs of loop() later are freed and it returns to the previous value. This appears to be quite constant, I see the same numbers time and again. Looks like a big enough number to me, isn't it?

I found the OOM option; enabled it; recompiling. I'll try to trigger the problem again. I can't find the lwip-v2 option in the Arduino IDE.

wvmarle, 2019-05-27:
I here have another decoded stack trace - appears to be related, but it's a bit different: the "last malloc call" address is in the stack trace, pointing to SPIFFS.

However my debug messages (from Serial.print() statements) show that my code continued beyond the point of that SPIFFS read (it reads the data from the log file to be sent out), implying that this call actually did work.

Then it tried to transmit the data, code got stuck for about 45 mins before continuing. My millis() counter indicated 840 seconds passed, that's just over 13 minutes. Apparently even the millis() counter got hung up here.

This was followed by a print() statement which comes from near the end of yet ANOTHER function, earlier prints of that function were missing (and it should never have executed that function in the first place): that indicates some kind of memory corruption, as execution continued at the wrong place, and the below stack dump appeared. I only posted the decoded part.

If only I had a sure-fire way of triggering this... I'd have the MVCE already.

    16:41:02.965 -> last failed alloc call: 4022D60A(1480)

    0x40222379: BearSSL::WiFiClientSecure::_installClientX509Validator() at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h line 986
    :  (inlined by) ?? at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h line 316
    :  (inlined by) ?? at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h line 598
    :  (inlined by) ?? at /home/wouter/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr.h line 614
    :  (inlined by) BearSSL::WiFiClientSecure::_installClientX509Validator() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 877
    0x40222db0: BearSSL::WiFiClientSecure::_connectSSL(char const*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 962
    0x40219d24: esp_yield at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 91
    0x4021a977: delay at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring.cpp line 54
    0x40206df9: WiFiClient::connect(IPAddress, unsigned short) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/include/ClientContext.h line 136
    :  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 170
    0x40222fa1: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
    0x402252e8: BearSSL::PrivateKey::getEC() const at ?? line ?
    0x402252e8: BearSSL::PrivateKey::getEC() const at ?? line ?
    0x40216c00: HTTPClient::connect() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1165
    0x4021c222: uart_write at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/uart.cpp line 498
    0x402178c4: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 655
    0x40217e4c: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
    0x40273fd6: sleep_reset_analog_rtcreg_8266 at ?? line ?
    0x40218159: Print::write(char const*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.h line 60
    0x40217e37: HardwareSerial::write(unsigned char) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 154
    0x40217a5e: HTTPClient::GET() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 575
    0x40211974: HydroMonitorLogging::sendPostData(char*) at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorLogging.cpp line 730
    0x40225528: BearSSL::PrivateKey::getEC() const at ?? line ?
    0x4022c578: _vsprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsprintf.c line 65
    0x40217e4c: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
    0x40103004: wDev_ProcessFiq at ?? line ?
    0x4022000a: spiffs_object_modify at ?? line ?
    0x40211b8e: HydroMonitorLogging::transmitData() at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorLogging.cpp line 288
    0x4022d60a: operator new(unsigned int) at /workdir/arena/gcc/xtensa-lx106-elf/libstdc++-v3-nox/libsupc++/../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_op.cc line 52
[... cut ...]
    0x40207a93: TwoWire::requestFrom(int, int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Wire/Wire.cpp line 134
    0x40204309: loop at /home/wouter/Arduino/Williams_fridge/Fridge_control/loop.ino line 36
    0x40219dd4: loop_wrapper() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 125
    0x401015d5: cont_wrapper at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 81

wvmarle, 2019-05-27:
There we go again: another stack trace, this time it's got a lot of things that I've never seen before, probably from the OOM option.

    Exception (29):
    epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000
    (raw stack trace skipped)
    last failed alloc call: 4021AEDC(192)

Decoded:

    Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
    Decoding 113 results
    0x4025143a: ieee80211_alloc_proberesp at ?? line ?
    0x40251358: ieee80211_alloc_proberesp at ?? line ?
    0x402557d1: cnx_update_bss_more at ?? line ?
    0x402514fb: ieee80211_send_proberesp at ?? line ?
    0x4021cd07: umm_calloc at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/umm_malloc/umm_malloc.cpp line 1704
    0x40277010: sleep_reset_analog_rtcreg_8266 at ?? line ?
    0x4021af48: calloc_loc at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/heap.cpp line 134
    0x401031b6: wDev_ProcessFiq at ?? line ?
    0x4024e0cc: hostap_input at ?? line ?
    0x4024df9f: hostap_input at ?? line ?
    0x40253fa7: sta_input at ?? line ?
    0x40254012: sta_input at ?? line ?
    0x4025dd6b: pp_attach at ?? line ?
    0x4026d9e5: get_sar_dout at ?? line ?
    0x4026e688: tx_pwctrl_cal at ?? line ?
    0x40255d81: cnx_node_search at ?? line ?
    0x4024d7db: hostap_input at ?? line ?
    0x4026ea38: tx_pwctrl_background at ?? line ?
    0x4025d32a: pp_tx_idle_timeout at ?? line ?
    0x4025cc03: ppPeocessRxPktHdr at ?? line ?
    0x40219cef: loop_task(ETSEventTag*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 133
    0x40105448: call_user_start_local at ?? line ?
    0x4010544e: call_user_start_local at ?? line ?
    0x4010000d: call_user_start at ?? line ?
    0x40101618: cont_ret at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 142
    0x401015c5: cont_continue at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/cont.S line 51
    0x40279930: chip_v6_unset_chanfreq at ?? line ?
    0x40239c58: sha2big_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c line 165
    :  (inlined by) br_sha384_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2big.c line 207
    0x4023a4c3: sha2small_out at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 249
    0x40279674: chip_v6_unset_chanfreq at ?? line ?
    0x40246c51: br_multihash_update at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/multihash.c line 141
    0x4010048f: void sws_isr0>() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 38
    0x4010048f: void sws_isr0>() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 38
    0x40101161: interrupt_handler at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring_digital.cpp line 169
    0x40101150: interrupt_handler at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring_digital.cpp line 165
    0x40105c21: ets_timer_disarm at ?? line ?
    0x40101090: interrupt_handler at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring_digital.cpp line 133
    0x40101090: interrupt_handler at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_wiring_digital.cpp line 133
    0x40105315: lmacTxFrame at ?? line ?
    0x40104149: lmacIsIdle at ?? line ?
    0x40104149: lmacIsIdle at ?? line ?
    0x4010528a: lmacTxFrame at ?? line ?
    0x40101f46: pp_post at ?? line ?
    0x4010533f: lmacRxDone at ?? line ?
    0x40101f46: pp_post at ?? line ?
    0x4010533f: lmacRxDone at ?? line ?
    0x40101f46: pp_post at ?? line ?
    0x40102c5d: trc_NeedRTS at ?? line ?
    0x40101f46: pp_post at ?? line ?
    0x40102e32: trc_NeedRTS at ?? line ?
    0x40101f46: pp_post at ?? line ?
    0x4010533f: lmacRxDone at ?? line ?
    0x40102c5d: trc_NeedRTS at ?? line ?
    0x40102e32: trc_NeedRTS at ?? line ?
    0x40103272: wDev_ProcessFiq at ?? line ?
    0x4010300c: wDev_ProcessFiq at ?? line ?
    0x40222b83: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 447
    0x40222b83: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 447
[... 25 lines cut ...]
    0x402178c0: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 655
    0x40217e48: HardwareSerial::write(unsigned char const*, unsigned int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h line 158
    0x402742d2: sleep_reset_analog_rtcreg_8266 at ?? line ?
    0x40218155: Print::write(char const*) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/Print.h line 60
[... cut ...]
    0x40210770: HydroMonitorIsolatedSensorBoard::readSensor(bool) at /home/wouter/Arduino/libraries/HydroMonitor/src/HydroMonitorIsolatedSensorBoard.cpp line 33
    0x40207a93: TwoWire::requestFrom(int, int) at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/Wire/Wire.cpp line 134
    0x40204309: loop at /home/wouter/Arduino/Williams_fridge/Fridge_control/loop.ino line 36
    0x40219de4: loop_wrapper() at /home/wouter/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/core_esp8266_main.cpp line 125

wvmarle, 2019-05-27:
I see references to SoftwareSerial in that stack trace: that's indeed yet another process that is running on the same chip. Every 3 seconds a sensor sends a message with sensor readings of about 14-18 characters at 9600 bps to a SoftwareSerial instance. Rx only; the Tx is not used.

This is something that has been working worse and worse while doing more with the logging and other things. I have more and more Serial.print() statements in the code. I often get illegal characters in the transmission, and have already written my code so that this is detected and ignored. Some transmissions fail, but enough that are received successfully to not cause issues.

wvmarle, 2019-05-27:
This is the function where the crash happens (I added a comment at the line that's pointed at in the stack trace). The char array postData contains a complete URL: http://host/path?key=value&...

    uint16_t HydroMonitorLogging::sendPostData(char* postData) {

    //  std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
    //  client->setInsecure();                                    // Do not check for https fingerprints (insecure: allows MiM attacks).

      WiFiClientSecure client;
      client.setInsecure();                                     // Do not check for https fingerprints (insecure: allows MiM attacks).
      HTTPClient https;
      
      // Open a connection to the host.
      uint16_t responseCode;
      uint32_t startTransmission = millis();
      Serial.print(F("Starting transmission of "));
      Serial.print(strlen(postData));
      Serial.print(F(" bytes: "));
      Serial.println(postData);
      if (https.begin(client, postData)) {                      // HTTPS connection.
        Serial.println(F("Connected."));
        responseCode = https.GET();                             // This is the point the crash data points at.
        Serial.println(F("Got the GET request result."));
        if (responseCode > 0) {                                 // httpCode will be negative on error
          if (responseCode == HTTP_CODE_OK || responseCode == HTTP_CODE_MOVED_PERMANENTLY) { // File found at server
            String payload = https.getString();
          }
        } 
      }
      https.end();
      client.stop();
      Serial.print(F("Transmission complete. Response code: "));
      Serial.print(responseCode);
      Serial.print(F(" Time taken: "));
      Serial.print(millis() - startTransmission);
      Serial.println(F(" ms."));
      return responseCode;    
    }

earlephilhower, 2019-05-27:
Wow, lots of updates since I glanced at it yesterday.

"Abort called" is the critical bit here, I think.  std::shared_ptr, when it can't allocate, will throw/abort.  Hence the OOM.  You need ~17KB free **contiguous** heap to make a SSL connection, and if there are no other BearSSL objects present you also need a 5.5KB contiguous chunk for the stack.

WiFiClientSecure is also very large itself, and putting it on the stack as shown here may also end up blowing past the end of stack and corrupting heap if your stack is within a KB or so of the end.

The "last failed alloc" is also a dead giveaway for memory alloc errors.  It being set means your app ran out of heap.  Sometimes you can recover (if you code defensively), but most times you can't.

See the below example to see how shared_ptr fails...

````
#include <memory>

void setup() {
  Serial.begin(115200);
  Serial.printf("\n\nTrying to std_shared_ptr...\n");
  auto test_abort = std::shared_ptr<unsigned char>(new unsigned char[60000], std::default_delete<unsigned char[]>());
}

void loop() { }
````

````

Trying to std_shared_ptr...

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 01b0
3fffff80:  0001c200 0000001c 3ffee244 4020ab90  
3fffff90:  3fffdad0 00000000 3ffee244 402010c5  
3fffffa0:  feefeffe 00000000 3ffee26c 402018a0  
3fffffb0:  feefeffe feefeffe 3ffe84f4 40100459  
<<<stack<<<

last failed alloc call: 4020ABBE(60000)
````

wvmarle, 2019-05-28:
Right... That seems to have been the problem indeed. I knew already that ssl takes a lot of memory, so I've been doing my best to keep the memory use low (upon compiling almost 48 kB free - when running the free heap is given at some 32 kB). I totally didn't see that coming: still not enough space for that 17 kB.

I tried to create the WiFiClientSecure and HTTPClient globally (at the top of my sketch), then pass as reference to the library object that needs them. That reduced the free heap significantly, but far from the expected 22 kB the two would need between them. And I still got the same crashes, so that didn't work.

In the end I've basically thrown in the towel. I've allowed my server to accept http connections on that specific URL, switched to WiFiClient, declare both the WiFiClient and HTTPClient locally again in the library, and so far so good: haven't been able to trigger the above crashes in the way I did before. Apparently running both a server (w/o encryption) and ssl client at the same time is simply too much.

Not the solution I hoped for but at least it works and that's the most important now. Thanks for the help, I'm closing this issue. Any tips of running the ssl client after all are welcome. I know it's not a support forum but my posts about the same on stackexchange and the esp8266 community forum unfortunately haven't gotten any replies yet...

closed 2019-05-28 by wvmarle

Ismail955, 2019-08-31:
Arduino: 1.8.9 (Windows 10), Papan:"WeMos D1 R1, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Opsi bangun berubah, membangun kembali semuanya
In file included from C:\Users\I S M A I L\OneDrive\Dokumen\Arduino\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:29:0,

                 from C:\Users\I S M A I L\OneDrive\Dokumen\Arduino\libraries\ESP8266WiFi\src/WiFiClientSecure.h:41,

                 from C:\Users\I S M A I L\OneDrive\Dokumen\Arduino\libraries\ESP8266WiFi\src/WiFiServerSecure.h:20,

                 from C:\Users\I S M A I L\OneDrive\Dokumen\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,

                 from C:\Users\ISMAIL~1\AppData\Local\Temp\arduino_modified_sketch_360260\FlashLED.ino:8:

C:\Users\I S M A I L\OneDrive\Dokumen\Arduino\libraries\ESP8266WiFi\src/BearSSLHelpers.h:148:34: error: 'virtual const unsigned char* BearSSL::HashSHA256::oid()' marked override, but does not override

     virtual const unsigned char *oid() override;

                                  ^

C:\Users\ISMAIL~1\AppData\Local\Temp\arduino_modified_sketch_360260\FlashLED.ino: In function 'void loop()':

C:\Users\ISMAIL~1\AppData\Local\Temp\arduino_modified_sketch_360260\FlashLED.ino:102:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

   if (millis() > Bot_lasttime + Bot_mtbs)  {

                                 ^

exit status 1
Error compiling for board WeMos D1 R1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.