esp8266/Arduino #6148: error: 'virtual int WiFiClient::connect(IPAddress, uint16_t)' marked override, but does not override

apanasara, 2019-05-27:
#### Platform

- Hardware:					ESP-8266 NodeMCU-1
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu|]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

after upgrading board, following error arised

### Debug Messages

```
C:\Users\x\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/WiFiClient.h:56:15: error: 'virtual int WiFiClient::connect(IPAddress, uint16_t)' marked override, but does not override

   virtual int connect(IPAddress ip, uint16_t port) override;
```

earlephilhower, 2019-05-27:
Looks like an installation issue.  Do you possibly have an Arduino WiFi library in your libraries directory?

apanasara, 2019-05-27:
```cpp
#include <ESPWiFi.h>
```

apanasara, 2019-05-27:
I tested given example **BearSSL_Vaidation.ino**, same error is raised by compiler.

earlephilhower, 2019-07-25:
Closing as this seems like an installation issue and is unreproducible by others.

closed 2019-07-25 by earlephilhower