esp8266/Arduino #6149: http.GET fails every other call

pgooch, 2019-05-27:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dior]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When I try and call against the DD-WRT router firmware with an http authorization username/password it will give me a -2 "send header failed" every other call. This does not happen when it is called from other devices. It also is not related to the time between calls, as it is always every other be it once per second or once per half second.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

v/* All the imports and setup for the wifi */
#include <Arduino.h>
#include <ESP8266WiFi.h>
// Imports
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
ESP8266WiFiMulti WiFiMulti;
WiFiClient client;
HTTPClient http;

// Settings 
const int time_between_updates = 500; // time between bandwidth checks in ms

// Variables we will be using
char last_router_return = 'Wait for it...';
int last_update = -1001;

void setup() {
  // Start the serial at 115200 baurd
  Serial.begin(115200);

  // Connect to the wifi
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");
}

void loop() {
  int now = millis();
  if(now>=last_update+time_between_updates){
    last_update = now;
    get_bandwidth_usage();
  }else{
    Serial.print(millis());
    Serial.print(" ");
    Serial.println(last_update+time_between_updates);
    delay(100);
  }
}

// Timer will call this every second (because internets are sold in mb/s)
bool get_bandwidth_usage(){
  if ((WiFiMulti.run() == WL_CONNECTED)) {
   http.begin(client, "http://ROUTER_USERNAME:ROUTER_PASSWORD@192.168.1.1/fetchif.cgi?br0");

    // We're connected so make that call!
    int httpCode = http.GET();
    Serial.println(httpCode);
    
    // If it's a positive number (because it didn't error arduino side) and it's 200 (because that means the server succeeded).
    if (httpCode > 0) {
      if (httpCode == HTTP_CODE_OK){
        // The stuff I want to do will go here.
      }
    }else{
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }

   }else{
    Serial.println("Looks like we're not connected to the WiFi yet. Lets try that next pass.");
  }
}```

### Debug Messages

```
-2
[HTTP] GET... failed, error: send header failed
```

dmdt, 2019-06-04:
I have described same problem, check this https://github.com/esp8266/Arduino/issues/5818#issuecomment-498567643 .
But not only problem with -2 code, it leave when I set reuse to true.
I can't recieve variable data, responses are the same as first.

earlephilhower, 2020-05-26:
This bug is quite old and there have been two major fixes to the `reuse` behavior.  I believe this was fixed with either #6457 or #6476.  If not, please re-test against master and open a new bug with MCVE/etc.

closed 2020-05-26 by earlephilhower