esp8266/Arduino #6151: Seg Fault when trying to initialize stringstream

blasco-burguillos, 2019-05-27:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Esp8266
- Core Version:			latest
- Development Env:		Make
- Operating System:		Arch

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v1.4
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Trying to instantiate a stringstream yields a segmentation fault. I've been trying to debug it with the GDBStub, but I'm stuck. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <GDBStub.h>

#include <string>
#include <sstream>

void setup() {
    Serial.begin(115200);

    gdbstub_init();
}

bool debug_stop = true;

void loop() {

    Serial.println("Running");
    delay(100);

    // set debug_stop = false from GDB in order to continue
    if (debug_stop == true) return; 

     std::stringstream ss("hello");
}
```

### Debug Messages
The seg fault:
```
Program received signal SIGSEGV, Segmentation fault.
0x40235268 in std::type_info::operator== (__arg=..., 
    this=0x402695c4 <typeinfo for std::ctype<char>>)
    at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/typeinfo:123
123 ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/typeinfo: No such file or directory.
[... 186 lines cut ...]
   *  @brief  Test for the presence of a facet.
   *  @ingroup locales
   *
   *  has_facet tests the locale argument for the presence of the facet type
   *  provided as the template parameter.  Facets derived from the facet
   *  parameter will also return true.
   *
   *  @tparam  _Facet  The facet type to test the presence of.
   *  @param  __loc  The locale to test.
   *  @return  true if @p __loc contains a facet of type _Facet, else false.
  */
  template<typename _Facet>
    bool
    has_facet(const locale& __loc) throw()
    {
      const size_t __i = _Facet::id._M_id();
      const locale::facet** __facets = __loc._M_impl->_M_facets;
      return (__i < __loc._M_impl->_M_facets_size
#ifdef __GXX_RTTI
          && dynamic_cast<const _Facet*>(__facets[__i]));
#else
              && static_cast<const _Facet*>(__facets[__i]));
#endif
    }
```

earlephilhower, 2019-05-27:
Looking at the trace you may be out of luck.  Seems like it's trying to do RTTI checks and dynamic casts, which are not supported on the 8266 (RTTI is disabled).  @devyte, can you take a look too?

blasco-burguillos, 2019-05-28:
Why is RTTI disabled? I tried to compile with RTTI enabled but that didn't fix the problem.

earlephilhower, 2019-05-28:
Memory issues and API compatibility w/the binary SDK blobs.  RTTI isn't enabled for the precompiled libs, either, so your change didn't have any effect.

devyte, 2019-05-28:
That is what the trace hints at. I've been trying to find some source that confirms/denies the dependence of std::stringstream on rtti, but no luck.
If it really uses dynamic_cast, we'll have to disable it from the stdc++lib.

j-blasco, 2019-05-28:
I guess I can't use stringstreams then. Anyhow, I wanted to compile the SDK my self, to test if the gdb TUI would work, for some reason is disabled in the provided build that one gest with get.py.

I tried compiling https://github.com/pfalcon/esp-open-sdk
But the xtensa gdb tool that I built fails to attach to the board. It says 
```
Remote 'g' packet reply is too long: 680f0040xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx1a00000000000000xxxxxxxxxxxxxxxxfada8cc2xxxxxxxx30000000xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx580f004090ffff3f000000002000000000000000840cff3fe057244001000000800cff3f000000009492fe3f18e7fe3fb0daff3f00000000b0dcff3f50e7fe3f
```

This also happens with the current master branch when I tried Arduino ESP866 master branch. Only works when I use the 2.5.2 branch.

I wonder how the SDK tools where built for that branch? I would like to rebuild them and test this issue too, what are your thoughts?

earlephilhower, 2019-05-28:
For GDB you need to use the toolchain we supply (basically plain GNU sources).  We don't use pfalcon's repo any more and are building from mostly unchanged GNU sources for everything.  See esp-quick-toolchain.

earlephilhower, 2019-07-21:
Actually, this was due to exception handlers accessing progmem, something that started failing when we moved SDKs back due to bugs in pre 3.0.0.

I just tried the example with PR #6273 , and it runs fine.  Try that PR, or wait for it to merge.  You need to run tools/get.py to download the new toolchain either way.

commit ac25f327 (2019-07-21): Upgrade to 2.5.0-4 toolchain w/improved pgm_read_x, C++ exceptions (#6273)

* Upgrade to 2.5.0-4 toolchain w/improved pgm_read_x

Rebuild the entire toolchain (including standard libraries) with the
latest pgm_read_xxx headers included (which fix unaligned dword reads
from progmem and run faster/smaller, and a pgm_read_byte change which
removes an instruction on each read saving flash).

Pull in latest bearssl while we're at it, too, which speeds up EC
handshakes and reduced ROM usage, too.

* Fix C++ exceptions

Exception code now only does 32b aligned reads from progmem to access
the eh_tabl [... cut]
closed 2019-07-21 by earlephilhower