esp8266/Arduino #6166: Increasing CONT_STACKSIZE causes infinity boot loop with rst cause:4,boot mode(3,6)

GiovanniCmpaner, 2019-05-30:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP07]
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[1MB/64K]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[none]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Increasing CONT_STACKSIZE using:
`build.extra_flags=-DCONT_STACKSIZE=8192`
Causes infinity boot loop with rst cause:4,boot mode(3,6).

Is there a way to increase CONT_STACKSIZE safely?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Messages

```
ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
```

devyte, 2019-05-30:
The CONT stacksize is by default overlayed on the SYS stack in order to save mem. There is virtually no room left for increasing it. If you need more, you would have to disable the overlay optimization, which I think is done by including a call to some empty function in your sketch (I don't remember the fn name) . That costs you 5K heap, plus whatever your increment is. I strongly advise against it. 
If you need more than 5K, I'd say you have a design issue. Consider new/malloc and/or use of a global.
Closing.

closed 2019-05-30 by devyte

d-a-v, 2019-05-30:
it is
```
#include <coredecls.h>
void setup()
{
  disable_extra4k_at_link_time();
...
}
...
```

GiovanniCmpaner, 2019-05-31:
Thanks for the answers, I will follow the advice:
> Consider new/malloc and/or use of a global