esp8266/Arduino #6167: about ADC max value

isharrrry, 2019-05-31:
Hello,
I only have an ESP12F which I am looking to use adc. The wiring is essentially:
+3.3v ------ 10k resistor ------ A0

code:
int values= analogRead(A0);

When reading the ADC value I get values  1024.
I check file core_esp8266_wiring_analog.c  Line 35:
    return digitalRead(pin) * 1023;

I want to get the adc values from 0 to 1023 (just  like other Arduino board adc value  0 ~ 2^10-1 ). 

I try use:
int values= analogRead(A0);
values=(values==0)?0:values-1;
Fix may problem, but this is a bug?

thanks
Huang

5chufti, 2019-06-01:
1024 is indicating overflow which isn't unexpected with 3.3V on input with Vmax.=1V.

jandrassy, 2019-06-01:
ADC pin of esp8266 can read max 1 V

closed 2019-06-02 by devyte