esp8266/Arduino #6168: Disabling boot messages using gate

GiovanniCmpaner, 2019-05-31:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-07]
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[1MB/64K]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[none]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[ERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I'm trying to disable the boot messages using a gate on TX, using this **simplified** schematic:
[image]
I cannot use any other GPIO besides GPIO0, as they are already in use.
But the gate has no effect, because when analysing GPIO0, it goes to LOW on boot, or more specifically, produces a high frequency clock.

Is there a way to disable boot messages or a way to stop GPIO0 from going LOW on boot?

devyte, 2019-05-31:
No and no, the boot messages come from code not under our control.
I suggest investigating an RC+comparator circuit at the input of the gate.
Not the right place to ask these questions, this is an issue tracker. Please seek further help at a community forum.

closed 2019-05-31 by devyte

GiovanniCmpaner, 2019-05-31:
Thanks.

> Not the right place to ask these questions, this is an issue tracker. Please seek further help at a community forum.
I'll remember that.