esp8266/Arduino #6169: http.GET() return connection refused error (-1)

comertial, 2019-05-31:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[NodeMCU 1.0 (ESP-12E Module)]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I try to get my public ip with the [ipify](http://api.ipify.org/), but running the code i get connection refused.. i use the **BasicHttpClient** exactly as it is and i just change my ssid and password and the `http.begin("http://api.ipify.org/");`! thats all the changes but i get a connection refused.. until yesterday everything was working fine!
I also tried the same example without changing anything else other than my ssid and passoword but again the same problem occurs!

### Debug Messages
on serial monitor this is what is being printed:
```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
```

5chufti, 2019-06-01:
so, and instead of waiting and seeing if this isn't just an error of ipify.org, you open an issue???
please check again - see that it works now - close this issue and think twice next time.
This is no board for general problems with esp and arduino. To handle such "issues" please use boards lie http://www.esp8266.com

comertial, 2019-06-01:
i gave ipify as an example.. i also tried [what is my ip api](bot.whatismyipaddress.com) but again the same problem occurs... i also tried to `http.begin("http://google.com/");` but again connection was refused.. finaly i just run the example without changing anything but again the problem is the same!! so.. i am sure that its not ipify's problem nor what is my ip's problem nor google's problem and probably its something wrong with the library, or the example, or i dont even know because if i knew it would be solved by me.. so this is the reason i opened the issue!
thanks in advance!

d-a-v, 2019-06-17:
@comertial enable debug mode (tools menu), and give all output from the BasicHttpClient sketch.

comertial, 2019-06-21:
For some reason that I can't explain resetting my modem did the job!

closed 2019-06-21 by comertial

teo1978, 2019-10-06:
Actually I was (or am) observing the same issue: random but persistent failures to connect reporting bogus "connection refused" errors.

By "random but persistent" I mean that I would randomly start having the issue when connecting to a new AP (same sketch and board that had been connecting to another AP for months without issues), and then it would persist, and then some day it would disappear. Of course I was tinkering with several parameters and every time I thought I had worked around it by changing something.

Today the issue went away after upgrading to the latest version of this project, but now I realise, I have also reset the router/AP, which in my case is my Android phone with Portable Hotspot turned on (by reset it I mean I turned Portable Hotspot off and on).

So, I wonder whether the issue has been fixed in a recent release, or if this is yet another instance of it going away randomly after resetting the router.

I had given some detailed debug output in
https://github.com/esp8266/Arduino/issues/5137

In that particular report I was  assuming that nothing was wrong in the http client itself (or wifi client or any layer below it) and I  was originally reporting the lack of detail in the error reporting, which actually turned out to be plain wrong error reporting (bogus "connection refused" when no connection was being refused), but then it became clear that there IS something wrong in the client.

So, is the issue (randomly failing http GETs and bogus "connection refused" errors) known to have been fixed?

teo1978, 2019-10-06:
Ok, no, it's not fixed.

I downgraded back to the version I was using (somewhere after https://github.com/esp8266/Arduino/commit/a1e59e9c014c907c008e96c9a3f1b7fc731adedb, not much later) and I still can't reproduce the issue, with the same sketch and same board and same router (my Android phone ad portable hotspot) where I was observing the issue in the first place, and where I stopped observing it today after upgrading. But now downgrading doesn't bring back the issue, so it's clear it simply randomly disappeared as it did many times before, just as randomly as it often reappears.

This issue is real and should be reopened, and it seems to be related to what I reported in https://github.com/esp8266/Arduino/issues/5137

I also wonder whether this is related to https://github.com/esp8266/Arduino/issues/4176
(see my comments starting from https://github.com/esp8266/Arduino/issues/4176#issuecomment-447581486)