esp8266/Arduino #6170: Empty reply from server using mDNS_Web_Server

pmalhaire, 2019-05-31:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-32]
- Core Version:			[l]
- Development Env:		[Arduino IDE]
- Operating System:		[Archlinux]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When using [mDNS_Web_Server](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L106)

I got an empty reply from server
```
[pierrot@pierrot-pc avahi-0.7]$ curl -v "http://esp32.local/"
*   Trying 192.168.1.23...
* TCP_NODELAY set
* Connected to esp32.local (192.168.1.23) port 80 (#0)
> GET / HTTP/1.1
> Host: esp32.local
> User-Agent: curl/7.64.1
> Accept: */*
> 
* Empty reply from server
* Connection #0 to host esp32.local left intact
curl: (52) Empty reply from server
* Closing connection 0
```

If I remove the `client.flush()` call here https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino#L106
and add a `client.stop()` at the end of the loop.
all is working fine.

```
curl -v "http://esp32.local/"
*   Trying 192.168.1.23...
* TCP_NODELAY set
* Connected to esp32.local (192.168.1.23) port 80 (#0)
> GET / HTTP/1.1
> Host: esp32.local
> User-Agent: curl/7.64.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: text/html
* no chunk, no close, no size. Assume close to signal end
< 
<!DOCTYPE HTML>
<html>Hello from ESP32 at 192.168.1.23</html>

* Closing connection 0
```

I am not sure if it's a bug or not but it seems legit to me not to flush the client and just stop the connection.

d-a-v, 2019-06-17:
@pmalhaire 
```
Hello from ESP32
```
You may have directed your issue to the wrong board repository

Have a try with an esp8266 :)

pmalhaire, 2019-06-18:
> @pmalhaire
> 
> ```
> Hello from ESP32
> ```
> 
> You may have directed your issue to the wrong board repository
> 
> Have a try with an esp8266 :)

Unfortunately I don't have one but it must work the same way.

d-a-v, 2019-06-18:
> I don't have one

Then I kindly redirect you to the right [issue tracker](https://github.com/espressif/arduino-esp32/issues/new/choose)

closed 2019-06-18 by d-a-v