esp8266/Arduino #6172: Random WDT after disconnecting Wifi. Occurs only with BearSSL, axTLS is fine

JiriBilek, 2019-06-01:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [n/a ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					NodeMCU 1.0 or ESP-12E
- Core Version:			2.5.1, git 2.6.0-dev from 5-30-2019
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU or Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		4MB (1MB SPIFFS)
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz or 160MHz
- Upload Using:		SERIAL
- Upload Speed:   921600
- SSL Support:   either all or basic SSL ciphers

### Problem Description

I am observing unstable behavior of the ESP8266 when I am turning on and off the Wifi frequently.
The example sketch is taken from my code and is greatly simplified. The point is to run ESP8266 with wifi turned off (low power consumption), accumulate data and send them to a server periodically. Here, the period is 30 seconds, in reality it is much longer, of course.
With axTLS the sketch runs easily overnight (e. g. 12 hours without reset, didn't test longer time), with BearSSL it does not survive 15 minutes, generally.
If you uncomment the prints in `loop()`, you can see that the wdt fires outside the sketch, the last character printed is always '-'. In this case use putty as serial monitor, the arduino serial monitor does not interpret BS character and you will get full screen of dots and hyphens :)
I tried to add delays in `wifiDisconnect` function, but no change.

I know your time is precious, so I tried to be as specific as I could. I am lost in debugging wdt outside the `loop `function, though.

Do you have any idea what to test? I hope the sketch is ok.
The same mechanism with disconnecting wifi is running in my devices based on v 2.5.0 dated 8-2018 (commit 641c5cd) with axTLS for months without an issue. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

//#define USING_AXTLS

#include <ESP8266WiFi.h>
#ifdef USING_AXTLS
    #include "WiFiClientSecureAxTLS.h"
    using namespace axTLS;
#endif

#define SSID "***"
#define PASSWORD "***"

static uint32_t MILLIS_TO_WAKE_UP = 30*1000;  // wake up after 30 seconds
const char* SERVERNAME = "www.example.com";
uint16_t PORT = 443;

#ifdef USING_AXTLS
    axTLS::WiFiClientSecure wifiClient;
#else
    BearSSL::WiFiClientSecure wifiClient;
#endif

uint32_t lastSleep = 0;  // millis of the last time when forced to sleep

void setup() {
    WiFi.setAutoConnect(false);
    Serial.begin(115200);

    pinMode(2, OUTPUT);
    digitalWrite(2, LOW);  // Start indicator

    // Wait for GPIO0 down as a start condition (we want to stop here after a wdt reset)
/*    Serial.println(F("\nConnect GPIO0 to GND to start"));
    pinMode(0, INPUT_PULLUP);
    while (digitalRead(0) == HIGH)
        delay(100);
*/
    pinMode(2, INPUT);  // Back to default

    Serial.print(ESP.getSdkVersion());
#ifdef USING_AXTLS
    Serial.println(F(", axTLS"));
#else
    Serial.println(F(", BearSSL"));
#endif

    wifiConnect();
    wifiSend(3);
    wifiDisconnect();
}

void loop() {
    //Serial.print("\x08.");
    
    if (millis() - lastSleep > MILLIS_TO_WAKE_UP) {
        wifiConnect();
        wifiSend(2);
        wifiDisconnect();
    }
    
    //Serial.print("\x08-");
}

void wifiConnect(void) {
    WiFi.mode(WIFI_STA);
    delay(100);

    WiFi.begin(SSID, PASSWORD);
    Serial.printf_P(PSTR("Connecting to %s "), SSID);

    while (WiFi.status() == WL_DISCONNECTED) {
        Serial.write('.');
        delay(500);
    }
    Serial.println();

    if (WiFi.status() == WL_CONNECTED) {
        Serial.printf_P(PSTR("WiFi connected (RSSI %d), IP address: %s, "), WiFi.RSSI(), WiFi.localIP().toString().c_str());
        Serial.printf_P(PSTR("mem: %d\r\n"), ESP.getFreeHeap());

        if (time(nullptr) < 100000000)
            readTime();
    }
}

void wifiDisconnect(void) {
    // Disconnecting wifi
    Serial.print(F("Disconnecting client"));
    wifiClient.stop();

    Serial.print(F(", wifi"));
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    delay(100);  // FIXME

    Serial.println(F(", sleeping"));
    WiFi.forceSleepBegin();  // turn off ESP8266 RF
    delay(100);  // FIXME

    lastSleep = millis();
}

boolean wifiSend(int8_t status) {
    // Check the wifi
    if (WiFi.status() != WL_CONNECTED) {
        Serial.println(F("[WiFi] Not connected to AP"));
        return false;
    }

#ifndef USING_AXTLS
    wifiClient.setInsecure();  // for testing ok
#endif    

   if (wifiClient.connect(SERVERNAME, PORT)) {
        Serial.println(F("[WiFi] Connected to server"));
    }
    else {
        Serial.println(F("[WiFi] Connection to server failed"));
        return false;
    }
    
    if (wifiClient.connected()) {
        // GET /test HTTP/1.1
        wifiClient.printf_P(PSTR("GET /test HTTP/1.1\nHost: %s\n\n"), SERVERNAME);
        
        Serial.print(F("[WiFi] Data sent, waiting for response ... "));

        // Wait max 5 seconds for server response
        long m = millis();
        while (millis() - m < 5000 && !wifiClient.available()) {
            delay(100);
        }

        // Read the response header
        Serial.println();
        while (wifiClient.connected()) {
            String line = wifiClient.readStringUntil('\n');
            Serial.println(line);
            if (line == "\r") {
//                Serial.println("headers received");
                break;
            }
            yield();
        }
        
        // Read and discard the data
        while (wifiClient.available() && wifiClient.connected()) {
            String line = wifiClient.readStringUntil('\n');
            yield();
        }
    }
    return true;
}

void readTime(void) {
    if (WiFi.status() != WL_CONNECTED) {
        return;
    }

    Serial.print(F("Setting time using SNTP "));

    configTime(1 * 3600, 0, "tik.cesnet.cz", "pool.ntp.org");

    // Read time, wait 5 seconds
    uint32_t m = millis();
    time_t now = time(nullptr);
    while (now < 100000000 && millis() - m < 5000) {
        delay(100);
        Serial.write('.');
        now = time(nullptr);
    }
    Serial.println();

    if (now < 100000000) {
        Serial.println(F("Time was not set."));
    }
    else {
        Serial.print(F("Current time: "));
        Serial.println(ctime(&now));
    }
}

```

### Debug Messages

Note that the 404 response is ok, we are using www.example.com server to test.

```
Connecting to BILNet ..........
WiFi connected (RSSI -52), IP address: 192.168.1.102, mem: 44784
[WiFi] Connected to server
[WiFi] Data sent, waiting for response ... 
HTTP/1.1 404 Not Found
Accept-Ranges: bytes
Cache-Control: max-age=604800
Content-Type: text/html; charset=UTF-8
Date: Sat, 01 Jun 2019 09:14:29 GMT
Expires: Sat, 08 Jun 2019 09:14:29 GMT
Last-Modified: Tue, 28 May 2019 06:46:04 GMT
Server: ECS (dcb/7EA6)
Vary: Accept-Encoding
X-Cache: 404-HIT
Content-Length: 1270

Disconnecting client, wifi, sleeping
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld
2.2.1(cfd48f3), BearSSL

```

JiriBilek, 2019-06-01:
First wdt with a stack trace appeared just now, perhaps it would be useful - crashes in malloc:

```
Exception 2: InstructionFetchError: Processor internal physical address or data error during instruction fetch
Decoding 85 results
0x401008ac: malloc at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 1677
0x4010458c: lmacProcessAckTimeout at ?? line ?
0x4020f763: new_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 235
0x4020fb54: ethernet_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 312
0x40105159: ets_timer_disarm at ?? line ?
0x402054c3: loop_task(ETSEventTag*) at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266/core_esp8266_main.cpp line 140
0x40216d7f: etharp_raw at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 1161
0x401008ac: malloc at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 1677
0x40211264: lwip_cyclic_timer at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 233
0x40216f7a: etharp_request at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 1202
0x4020fba8: do_memp_malloc_pool at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 254
0x40216fe4: etharp_tmr at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 203
0x40211264: lwip_cyclic_timer at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 233
0x40211264: lwip_cyclic_timer at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 233
0x40211274: lwip_cyclic_timer at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 243
0x4020fc0e: memp_free at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 447
0x4021140c: sys_check_timeouts at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 390
0x4023b434: ets_timer_handler_isr at ?? line ?
0x4023b441: ets_timer_handler_isr at ?? line ?
0x4023b486: ets_timer_handler_isr at ?? line ?
0x402054c3: loop_task(ETSEventTag*) at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266/core_esp8266_main.cpp line 140
0x40104980: call_user_start_local at ?? line ?
0x40104986: call_user_start_local at ?? line ?
0x4010000d: call_user_start at ?? line ?
0x4024a9e8: node_remove_from_list at ?? line ?
0x401026ee: wDev_ProcessFiq at ?? line ?
0x4021b5c7: sha2small_out at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 249
0x4024a8b0: node_remove_from_list at ?? line ?
0x4021b608: br_sha256_out at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/hash/sha2small.c line 305
0x4024a8b0: node_remove_from_list at ?? line ?
0x40221f55: br_hmac_out at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/ssl/bearssl/src/mac/hmac.c line 120
0x4024a8b0: node_remove_from_list at ?? line ?
0x4023545b: pp_attach at ?? line ?
0x402354aa: pp_attach at ?? line ?
0x402355b6: pp_attach at ?? line ?
0x4023545b: pp_attach at ?? line ?
0x402354aa: pp_attach at ?? line ?
0x402355b6: pp_attach at ?? line ?
0x40101482: pp_post at ?? line ?
0x40234567: ppTxPkt at ?? line ?
0x40227947: ieee80211_output_pbuf at ?? line ?
0x40104eff: wdt_feed at ?? line ?
0x40101482: pp_post at ?? line ?
0x40104877: lmacRxDone at ?? line ?
0x40102199: trc_NeedRTS at ?? line ?
0x40101482: pp_post at ?? line ?
0x4010236a: trc_NeedRTS at ?? line ?
0x4020f763: new_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 235
0x401027aa: wDev_ProcessFiq at ?? line ?
0x40102544: wDev_ProcessFiq at ?? line ?
0x401008ac: malloc at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 1677
0x40101482: pp_post at ?? line ?
0x40100daf: pp_soft_wdt_feed_local at ?? line ?
0x4010065c: _umm_free at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 1304
0x40102522: wDev_MacTim1Arm at ?? line ?
0x40102586: wDev_ProcessFiq at ?? line ?
0x4020f4e1: glue2esp_linkoutput at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 299
0x402133f6: pbuf_free_LWIP2 at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 786 (discriminator 1)
0x40102544: wDev_ProcessFiq at ?? line ?
0x402187d4: mem_malloc at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 210
0x40249f70: node_remove_from_list at ?? line ?
0x40105159: ets_timer_disarm at ?? line ?
0x40105159: ets_timer_disarm at ?? line ?
0x4010065c: _umm_free at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 1304
0x40100aa4: free at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 1764
0x401001c0: millis at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266/core_esp8266_wiring.cpp line 186
0x40205540: esp_yield at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266/core_esp8266_main.cpp line 97
0x40205561: esp_schedule at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266/core_esp8266_main.cpp line 102
0x402055f9: loop_wrapper() at D:\Programy\arduino\hardware\esp8266com-git_version\esp8266\cores\esp8266/core_esp8266_main.cpp line 134
```

Rob58329, 2019-06-04:
The below info might be of interest:

With what might be a similar issue, for my sketches I get Hardware Watchdog Reboots, which if I add a "delay(2000)" immediately after the "WiFi.mode(WIFI_OFF)", then the Hardware Watchdog always triggers during this delay.  However, without this "delay(2000)", the Hardware Watchdog triggers some dozen or more program steps after the "WiFi.mode(WIFI_OFF)" and subsequent "WiFi.forceSleepBegin()".

I reconnect over SSL every 8 minutes or so (and my SSL data exchanges always succeed and look the same as far as I can tell), and the subsequent crashes are intermittent (sometimes 3 in an hour, sometimes none for 24 hours).

My free-Heap does not appear to drop below approx 21000 bytes, and my free-Stack does not appear to drop below 1360 bytes.

[I am using an ESP8266 D1mini with the "ESP8266 Arduino Github software" (as at 31May19) with "BearSSL" and "IwIP variant v2 lower memory"].

I have not yet found a solution!..

JiriBilek, 2019-06-04:
@Rob58329 , thanks for the information.
Have you tried using axTLS? My devices with old core and axTLS work quite reliably (they connect to wifi once in 10 minutes). But merely switching from axTLS to BearSSL brings the issue.

Rob58329, 2019-06-05:
@JiriBilek:  My original ESP8266-sketch used the “ESP8266 Arduino Github Software core” from about 18 months ago (ie. which used axTLS and IwIP v1.4 (or perhaps earlier)), and my ESP8266 units were running for 3++ months without issue.  Annoyingly I have not yet been able to work out exactly which version-date of the Github Software I was using from 18 month ago...

But if the same sketch is compiled on the current (v31May19) “ESP8266 Arduino Github Software” (or in-fact using any of the Github versions from the last couple of months), it generates the above detailed intermittent Hardware Watchdog crashes. I still get the same crashes if I use the current Github software (v31May19) with axTLS, or BearSSL(which the sketch needed slight modification for), and with IwIP v1.4 or with IwIP v2.   I note that the current Github core compiles my sketches to use a bit more RAM when running (vs. 18months ago), but I currently don't think this is the issue.

earlephilhower, 2019-06-05:
What's interesting here is that you're turning off WiFi without ever closing the SSL connection.

Would you be able to check if the same thing happens if you add a `wifiClient.close()` at the end of your send routine?

It's possible that when you come back after the next time, or when the wifi power off event happens that some part of the LWIP closes everything, but the client still has a pointer to something that's no longer valid.  Then when you reconnect the first thing the client will do is to try to ::close() itself and uses this pointer and boom, memory corruption (==crash, WDT, whatever).

JiriBilek, 2019-06-05:
Thanks for a tip, but I am getting compilation error: class BearSSL::WiFiClientSecure' has no member named 'close'
I can't find `close()` function either in WiFiClientSecure, WiFiClient or Client.

earlephilhower, 2019-06-05:
Oops, it's `stop` not close I meant to type there.

JiriBilek, 2019-06-05:
I see. I am stopping the client in `wifiDisconnect()`. It is called immediately after `wifiSend()`.

earlephilhower, 2019-06-05:
There goes the easy bit. :(  Can you make the WiFiClientSecure a local variable in your send routine?  If that works reliably then it means there is a data lifetime issue in the client that needs looking at.  If it fails, too, then there's something very strange going on.

earlephilhower, 2019-06-05:
I replaced `wifiClient` with an temporary object who was `new`ed at the end of `wifiConnect()` and which was `delete`d immediately after its `stop` call.  So there was 0 possibility of the object using something invalidated after WIFI was turned off.

It still crashed.  I don't think this is related in any way to the WiFiClientSecure.  I think it's something in the SDK blob at this point going weird.

@d-a-v, @devyte, anything suspicious in the code here?

earlephilhower, 2019-06-05:
I replaced WiFiClientSecure with plain WiFiClient.

Even plain WiFiClient has the same crash in the waiting portion, so it's not HTTPS related, even.  Either WiFi powerdown/up causes the crashes or LWIP does (maybe trying something that's not valid to close a TCP connection buffer or something?)

````
Disconnecting client, wifi, sleeping
Connecting to NOBABIES ...........
WiFi connected (RSSI -47), IP address: 192.168.1.154, mem: 50848
[WiFi] Connected to server
[WiFi] Data sent, waiting for response ... 
HTTP/1.1 404 Not Found

Server: nginx

Date: Wed, 05 Jun 2019 23:31:31 GMT

Content-Type: text/html

Content-Length: 162

Connection: keep-alive

Disconnecting client, wifi, sleeping

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v7d5343a6
````

earlephilhower, 2019-06-06:
Using the base WiFiClient and only doing the WiFiClient.connect() and .WiFiClient.stop() in the loop (all data transmission is removed) also results in occasional WDTs.
````
...
WiFi connected (RSSI -43), IP address: 192.168.1.154, mem: 50024
[WiFi] Connected to server
Disconnecting client, wifi, sleeping
Connecting to NOBABIES .......
WiFi connected (RSSI -44), IP address: 192.168.1.154, mem: 50352
[WiFi] Connected to server
Disconnecting client, wifi, sleeping
Connecting to NOBABIES .......
WiFi connected (RSSI -45), IP address: 192.168.1.154, mem: 50024
[WiFi] Connected to server
Disconnecting client, wifi, sleeping

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v7d5343a6
~ld
````

JiriBilek, 2019-06-06:
Thanks, not being related to the SSL it is even more annoying.
Is there a chance to trace the wdt? I mean stack dump or any more information?

earlephilhower, 2019-06-06:
It's not in LWIP, either.  It's in the core blob.  I commented everything related to the WiFiClient:
````
void wifiDisconnect(void) {
    // Disconnecting wifi
    Serial.print(F("Disconnecting client"));
    //wifiClient->stop();
    delete wifiClient;
    
    Serial.print(F(", wifi"));
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    delay(100);  // FIXME

    Serial.println(F(", sleeping"));
    WiFi.forceSleepBegin();  // turn off ESP8266 RF
    delay(100);  // FIXME

    lastSleep = millis();
}

boolean wifiSend(int8_t status) {
    // Check the wifi
    if (WiFi.status() != WL_CONNECTED) {
        Serial.println(F("[WiFi] Not connected to AP"));
        return false;
    }

#if 0
#ifndef USING_AXTLS
    //wifiClient->setInsecure();  // for testing ok
#endif    
   if (wifiClient->connect(SERVERNAME, PORT)) {
        Serial.println(F("[WiFi] Connected to server"));
    }
...
    }
    #endif
    return true;
}
````

I still got a WDT overnight:
````
g to NOBABIES .......
WiFi connected (RSSI -49), IP address: 192.168.1.154, mem: 51464
Disconnecting client, wifi, sleeping
Connecting to NOBABIES .......
WiFi connected (RSSI -49), IP address: 192.168.1.154, mem: 51464
Disconnecting client, wifi, sleeping

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v7d5343a6
~ld
2.2.1(cfd48f3), BearSSL
Connecting to NOBABIES ......
````

There was another issue, same problem, but I can't seem to find it now.

WDT's handled by the RTC block, so I don't think you can get any info on where it happened.  To the CPU it looks like a simple reset.

earlephilhower, 2019-06-06:
Just to be clear, the actual guts of WiFi power off/power on stuff is in the Espressif binary-only blob.  So there is no way to debug or anything we can do here about it.

JiriBilek, 2019-06-06:
@earlephilhower , thanks a lot for your effort. I changed the title of this issue, it is misleading now.

tbdltee, 2019-06-09:
can you try change
   Serial.println(F(", sleeping"));
    WiFi.forceSleepBegin();  // turn off ESP8266 RF
    delay(100);  // FIXME
to
   Serial.println(F(", sleeping"));
    delay(100);  // FIXME
    WiFi.forceSleepBegin();  // turn off ESP8266 RF

see if it fixed.
I suspect that the Serial.print is still executing in background while you turn-off RF. I add delay to ensure the Serial.print has succesfully executed, before turning the RF off.

JiriBilek, 2019-06-10:
Yesterday, I compiled the original test sketch with an old core (2.5.0 dev, commit https://github.com/esp8266/Arduino/commit/641c5cd) and BearSSL.
It run for more than 18 hours without a problem. This confirms that the problem is not in the ssl library.

Ogauy, 2019-06-10:
Dear:
       Please send questions to this email address:support@aithinker.com.This doesn't deal with technical issues

JiriBilek, 2019-06-10:
@tbdltee : I tried what you suggested, unfortunately the wdts and even exceptions are there, although much less frequent (once per 2 hours approximately).
The testing setup was: the git version of library, BearSSL used to open a connection and send a GET request.

tbdltee, 2019-06-10:
@JiriBilek, general speaking, I used to have the wdt-reset situation and I've found out that I need to call the ESP function in the strict order, so wdt_reset never happended to me again.
I also turn-on/off wifi very often. When ESP wake-up frpm deepsleep, it execute setup(), acquire data, turn-on wifi, sending data, and goto sleep.

Here are my general code:
1. After Serial.print, Serial.println, need to wait at least 80ms before calling any command of WiFi.xxxx 
2. Sequence to turn-on WiFi, in this case your wifiConnect() function
void wifiConnect() {
  WiFi.forceSleepWake();
  delay (1);
  WiFi.mode (WIFI_STA);
  WiFi.begin (ssid, pass);
...}

3.Sequence to turn-off WiFi, in this case your wifiDisconnect() function
void wifiDisconnect(void) {
    Serial.print(F("Disconnecting client"));
    wifiClient.stop();
    Serial.print(F(", wifi"));
    delay (100);
    WiFi.disconnect();
    Serial.println(F(", sleeping"));
    delay (100);
    lastSleep = millis();
   WiFi.mode (WIFI_OFF);
   WiFi.forceSleepBegin();
   delay (1);
}

the delay (1); must be added. I don't know why but ESP seems not working without it.
Actually, I'm using ESP.deepsleep(). so my code is a little bit different. But the sequence of turn-on/off wifi are the same.

See if it helps?

JiriBilek, 2019-06-10:
@tbdltee: Thanks for the information but unfortunately, the proposed changes didn't work either. The device fires wdt in ca 1.5 hour frequency.
The only solution I see now is to revert to the old core.

The relevant part of the code:
```
void wifiConnect(void) {
    WiFi.forceSleepWake();
    delay(100);

    WiFi.mode(WIFI_STA);
    //delay(100);

    WiFi.begin(SSID, PASSWORD);
    Serial.printf_P(PSTR("Connecting to %s "), SSID);

    while (WiFi.status() == WL_DISCONNECTED) {
        Serial.write('.');
        delay(500);
    }
    Serial.println();

    if (WiFi.status() == WL_CONNECTED) {
        Serial.printf_P(PSTR("WiFi connected (RSSI %d), IP address: %s, "), WiFi.RSSI(), WiFi.localIP().toString().c_str());
        Serial.printf_P(PSTR("mem: %d\r\n"), ESP.getFreeHeap());

        if (time(nullptr) < 100000000)
            readTime();
    }
}

void wifiDisconnect(void) {
    // Disconnecting wifi
    Serial.print(F("Disconnecting client"));
    wifiClient.stop();

    Serial.print(F(", wifi"));
    delay(100);  // FIXME

    WiFi.disconnect();

    Serial.println(F(", sleeping"));
    delay(100);  // FIXME
    
    lastSleep = millis();

    WiFi.mode(WIFI_OFF);
    WiFi.forceSleepBegin();  // turn off ESP8266 RF
    delay(1);
}
```

tbdltee, 2019-06-11:
@JiriBilek. Thanks for update.
Currently, I'm using core 2.4.2, SDK2.2.1.  Core 2.5.x also have a lot of problem for me. I'll for a while until it's more stable.

JiriBilek, 2019-06-11:
But, after all, it is the best setup for the new core (2.6.0-dev). After two wdts in the evening it run overnight without problems (7 hours).

Just for information, the star in the graph is a reset and the gray circle is a successful transmittion. The first two stars in the picture are the resets caused by firmware upload.
[image]

12TA, 2019-06-11:
I can own this problem too. I spent ages trying everything. The issue is intermittent and I found my 'solution' was just to comment out my WiFi.disconnect(); line, and then it runs perfectly stably. I have not tested any effect on power consumption. Note the crash site is variable intermittent and some lines after the disconnect call. As I tinkered the problem got worse, not better and removing that line made all the difference. I can't explain it, but it is a simple thing to try and report back on. PaulS. 12TA.

Rob58329, 2019-06-14:
@JiriBilek: I note that [earlephilhower] above said that the guts of the "WiFi.mode(WIFI_OFF)" and similar commands are actually in the Espressif SDK.

Also that the latest esp8266 github (eg v31May19) for the Arduino IDE has a "Generic ESP8266 Module" with the options to select 3 different versions of the SDK, namely: "nonos-sdk 2.2.1 (legacy)=v2.1.0-10-g509eae8
",  "2.2.2-190313 (testing)=v2.2.1-61-gc7b580c" and "sdk pre-3 (known issues)=v2.2.0-28-g89920dc".

For comparison (from "esp8266/tools/sdk/version"):
* "https://github.com/esp8266/Arduino/tree/2.5.0 - igrr released on 6 Feb 2019": used "v2.2.0-28-g89920dc" (the "sdk pre-3")
..through to..
* "https://github.com/esp8266/Arduino/tree/2.5.0-beta1 - igrr released on 12 Dec 2018": all used "v2.2.0-28-g89920dc" (the "sdk pre-3")
* "https://github.com/esp8266/Arduino/tree/2.4.2 - igrr released on 3 Aug 2018": used "v2.1.0-10-g509eae8" (the so called "sdk 2.2.1(legacy)")

BUT, the default SDK version used in the current github(v31May19) for "Lolin D1 R2 & mini" (for example) is the "nonos-sdk 2.2.1 (legacy)" (which you only get the option-to-change if using the "Generic ESP8266 Module").  

However, it appears that you can add this SDK-option-choice to the current github (eg.v31May19) software by editing the Arduino IDE "esp8266/boards.txt" file.  For my "LOLIN(WEMOS) D1 R2 & mini" board, I found the section in this file which detailed my board, and added:

d1_mini.menu.sdk.nonosdk221=nonos-sdk 2.2.1 (legacy)
d1_mini.menu.sdk.nonosdk221.build.sdk=NONOSDK221
d1_mini.menu.sdk.nonosdk222=nonos-sdk 2.2.2-190313 (testing)
d1_mini.menu.sdk.nonosdk222.build.sdk=NONOSDK22x
d1_mini.menu.sdk.nonosdk3v0=nonos-sdk pre-3 (known issues)
d1_mini.menu.sdk.nonosdk3v0.build.sdk=NONOSDK3V0

[right above the "d1_mini.menu.ip.lm2f=v2 Lower Memory" line]

After an IDE restart, this then enables the above choice of SDK versions. For my current sketch which has the "Hardware Watchdog being intermittently triggered by (shortly after) WiFi.mode(WIFI_OFF)" issue:
* "nonos-sdk 2.2.1 (legacy)=v2.1.0-10-g509eae8" - causes intermitted Hardware Watchdog crashes.
* "sdk pre-3 (known issues)=v2.2.0-28-g89920dc" - still causes intermittent Hardware Watchdog crashes, BUT seems to be much less frequent.
* "2.2.2-190313 (testing)=v2.2.1-61-gc7b580c"   - not tested yet.

What I find interesting is that if I use "https://github.com/esp8266/Arduino/tree/2.5.0-beta2" (released 16 Dec 2018) with my current sketch, I have now been running for a week without any crashes.  And the "2.5.0-beta2" uses "SDK v2.2.0-28-g89920dc" which is the same as the above "sdk pre-3 (known issues)" which does crash intermittently, so I suspect there may be something else going on as well as the SDK version issue!??

JiriBilek, 2019-06-18:
@Rob58329: I think I am using the legacy core in all my tests. Not sure because now, I don't have my computer with me. Will be back in July.

d-a-v, 2019-06-18:
`WiFi.forceSleepBegin()` must be accompanied by `WiFi.forceSleepWake()`.

I could reproduce the bug, and it vanished with a call to `WiFi.forceSleepWake()` just before `WiFi.begin()`.

12TA, 2019-06-18:
In my opinion, which comes from experimentation too, the single command
WiFi.disconnect();
is the cause of the intermittent hardware resets since commenting it out cured the problem for me. Can someone else verify this? Paul

Rob58329, 2019-06-23:
I believe that my intermittent Hardware WDT crashes after "WiFi.mode(WIFI_OFF)" are caused by a commit dated 1Feb19, being:
* “stop lwIP dhcp client when WiFi goes off #5703”
(https://github.com/esp8266/Arduino/pull/5703/files/60560e634284a5beb194588bdb5696e396824044)
Which added the line “wifi_station_dhcpc_stop();”

(The above line is part of “bool ESP8266WiFiGenericClass::mode(WiFiMode_t m)”, so I assume this is what the arduino command "WiFi.mode(WIFI_OFF)" calls.)

Specifically, if I edit “libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp” and comment out the above line so it says:

`// wifi_station_dhcpc_stop(); // commented out as causing intermittent Hardware WDT`

Then my sketches no longer have these intermittent Hardware WDTs!

* My sketches now work fine using https://github.com/esp8266/Arduino/tree/2.5.0
* They also work fine using the recent github (31May19) with either “nonos-sdk 2.2.1 (legacy)=v2.1.0-10-g509eae8" or “sdk pre-3 (known issues)=v2.2.0-28-g89920dc", and either axTLS or BearSSL, and either IwIP v1.4 or IwIP v2  - all work fine for me now!

FYI:  I have 8 ESP8266-D1-Mini sensors, which connect to a remote server using SSL every 8 minutes or so.  Interestingly, each ESP8266 seemed to have these “intermittent Hardware WDT” crashes at different rates (NB. they all auto-recover from these crashes and log the crash-details to eeprom):  I have one sensor, which would run OK for a few hours, and then crash as above after every single connection (every 8 minutes) for 2+ hours until I removed it’s power.  However, most of my ESP8266 only crashed once every 12hours to 24hours or so, and one of my sensors only crashed after 7 days.  (NB.  But they all crashed with a Hardware WDT in the “delay(2000)” which I had added after my "WiFi.mode(WIFI_OFF)").  Now with the above “wifi_station_dhcpc_stop();” removal, all 8 sensors have all been running without issues for 3 days (ie 8*3=24 sensor-days) using the recent github (of 31May19).

(PS. My sketches use both WiFi.forceSleepBegin() and WiFi.forceSleepWake() in the correct places.)

JiriBilek, 2019-07-08:
@d-a-v:  In last 3 days, I was testing your suggestion in various setups and it works! The key is to change wifi mode with `WiFi.mode(WIFI_STA);` first and then after a small delay, wake up the radio `WiFi.forceSleepWake();`. The reverse case that I unfortunately had in my projects does not work in newer releases of esp8266/Arduino and fires intermittent wdts outside the `loop()` function.
Thank you, I would never discover it.
I will try to find out where delays are needed and where not but it doesn't change the basic idea.

@Rob58329: your investigation is great, please check if your app is working when you change the mode first and only then wake up the radio.

The relevant code once more:
```
void wifiConnect(void) {
  WiFi.mode(WIFI_STA);
  delay(100);
  WiFi.forceSleepWake();
  WiFi.begin(SSID, PASSWORD);
...
```

For me the problem is solved, thank you all for helping.

JiriBilek, 2019-07-08:
Sad to say but I was too optimistic. The intermittent wdts disappeared in my simple test but with a network traffic they are back here. Wdts fire not so frequently as before, though.
I am trying @Rob58329 tip with removing `wifi_station_dhcpc_stop()`.
Will keep posting here.

TD-er, 2019-07-08:
Try to do a WiFi disconnect from the AP and see if the node crashes.
See: https://github.com/esp8266/Arduino/issues/6266

Ideas to help force a disconnect:
- Change channel on AP
- Reboot AP
- Kick node from AP (only a few can do that)

JiriBilek, 2019-07-09:
@TD-er: does it make sense to force disconnect when the node is almost all the time disconnected (low power) and only once per 5 minutes connects, sends data and then disconnects again?

TD-er, 2019-07-09:
@JiriBilek the thing is, a forced disconnect can happen almost any time.
For example when the AP does change channel or when it is set to do a force disconnect at N received packet errors. (depends on brand/model/setting of AP)

And as far as I know, the problem is not really with the disconnect itself, but somehow the ESP does crash with a WD reboot at this (re)connect stage.
This does not happen always, but as far as I have seen, it can be triggered by just sending a disconnect message to the ESP.
With normal use, a disconnect from the WiFi doesn't happen that often and it also depends on a lot of factors.
Every now and then, the ESP doesn't even know it is being disconnected and just continues doing its work.
I don't know if in the background it may do something, but at least none of the WiFi events are fired to let me know something has changed.
Maybe it does do something in the background, because it sometimes also crashes with a WD reboot right after a WiFi disconnect message was sent to the ESP node, but no event was reported on the ESP itself (at least not an event I have a callback function for)

12TA, 2019-07-09:
Before I said this.
" In my opinion, which comes from experimentation too, the single command
WiFi.disconnect();
is the cause of the intermittent hardware resets since commenting it out cured the problem for me. Can someone else verify this? Paul "
The reason, I think, that people want to disconnect from WiFi is often to reduce power consumption by the ESP8266. Anyway, that is what I wanted. I was copying code snippets with that intention. Since then, I can confirm by experiment that power consumption falls to around 10ma in my NodeMCU board, so I assume WiFi is off and that my project really has stopped resetting. If others use that command in their projects, can they report on whether removing it fixes it. 
FYI the reset occurred some time after the command and did seem to be intermittent and random. As the problem has gone away and my program is massive, I can't easily post it all. Paul

TD-er, 2019-07-09:
> FYI the reset occurred some time after the command and did seem to be intermittent and random. As the problem has gone away and my program is massive

Well I am no stranger to "massive" regarding sketch size :)
Just removing the disconnect calls does not fix it here.
Still WD reboots every 2 or 3 times I force a WiFi disconnect from the AP.

Also removing references to the somewhat related function to set the WiFi mode to `WIFI_OFF` does not fix this issue.

JiriBilek, 2019-07-10:
@TD-er I am not sure if I understood your post. Does it mean that I can get intermittent WD reboots when connected with just disconnecting the client from the AP and the problem is not in this libraries but in the core?

Anyway, the day before yesterday, I changed my sketch as @Rob58329 suggested. The application works since then without a problem. I am continuing testing but starting to think the problem lies in commit https://github.com/esp8266/Arduino/pull/5703 that solves one issue and creates another one, unfortunately.

Rob58329, 2019-07-10:
@JiriBilek: Many thanks for above info: So I understand the suggested code changes the order of commands to:

```
void wifiConnect(void) {
  WiFi.mode(WIFI_STA);
  delay(100);
  WiFi.forceSleepWake(); delay(1);
  WiFi.begin(SSID, PASSWORD);
}
```
but still using:
```
void wifiOff(void) {
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  delay(2000); // increased to 2 seconds for testing to ensure crashes happen here & not later
  WiFi.forceSleepBegin(); delay(1);
}
```
[ I hadn't spotted that the @d-a-v code in his above post had "WiFi.forceSleepWake()" AFTER "WiFi.mode(WIFI_STA)"!! ]

I have tried changing my code to match the above, but sadly it makes no difference for me (ie. same as for you I understand): my code still has intermittent Hardware WDT crashes in the "delay(2000)" I have added after my "WiFi.mode(WIFI_OFF)". (In my short tests the frequency of crashes seems approximately the same as before.)

HOWEVER, as I detailed previously, editing "ESP8266WiFiGeneric.cpp" to comment out the "wifi_station_dhcpc_stop()" line solves my HWT crash issue, with both the above suggested order, and also my original "WiFi.forceSleepWake(); delay(1); WiFi.mode(WIFI_STA); delay(5);" order.

I would be interested to know what the "official(??)" suggested order for "WiFi.forceSleepWake()" and "WiFi.mode(WIFI_STA)" is though, as the @d-a-v post above is the only one I can currently find which has it in this "WiFi.mode(WIFI_STA) first and THEN WiFi.forceSleepWake()" order, and all the other posts I can find use the opposite order?

d-a-v, 2019-07-10:
About `wifi_station_dhcpc_stop()` a test is probably missing prior this call. I'll work on that asap.
It was added because a bug was suspected (dhcp timer still around and waking up when STA wasn't anymore). But the other checking whether STA is up before that call is missing.

About `WiFi.forceSleepWake()` order I am no authority.
I think it would be nice to put inside the core API the way to go to radio-sleep and to wake up from radio-sleep. So we can have a common basis to debug with and to improve.

d-a-v, 2019-07-10:
@Rob58329 @JiriBilek @TD-er 

> Then my sketches no longer have these intermittent Hardware WDTs!

Can you please try this in `ESP8266WiFiGeneric.cpp`:

declare just before `bool ESP8266WiFiGenericClass::mode(WiFiMode_t m)`:
```
extern "C" struct netif* eagle_lwip_getif(uint8 index);
```
then replace 
```
    if (m != WIFI_STA && m != WIFI_AP_STA)
```
by
```
    if (m != WIFI_STA && m != WIFI_AP_STA && eagle_lwip_getif(STATION_IF) != nullptr)
```
(and leave  `wifi_station_dhcpc_stop();` in place)

TD-er, 2019-07-11:
@d-a-v 
I will try the suggested check for the existence of the network adapter object (by lack of a better term).

And about moving stuff into the core; I think if steps are needed to have something functional, then yes!
Just because the core libraries should be as simple to operate as possible and thus remove the burden from the user to make explicit calls to some functions in a specific order.
But on the other hand, if it is a truly optional step with some specific use cases where it can be useful to not perform the steps then it should not be a fixed step in the core libraries.
If for some reason it is only needed to exclude a step in very specific use cases, then wrapping it in an `#ifdef` or `if (<some defined boolean>)` is also an option.

JiriBilek, 2019-07-11:
I am checking the fix and it's running smoothly for 90 minutes now. I plan to test it for at least 24 hours. Fingers crossed.

TD-er, 2019-07-11:
I just tested the proposed change, but it is at least not a fix to the WD reboots I am seeing when the unit does do a disconnect.
The code change suggests it should crash (null pointer dereference) when it happens what that extra check should fix, but I'm seeing that the ESP does simply halt or at least does not continue with running the loop() function.

d-a-v, 2019-07-11:
@TD-er when #6283 is solved maybe we can understand better your WDT issue.

d-a-v, 2019-07-11:
@TD-er I was thinking of a new `WiFi.mode(WIFI_SHUTDOWN)` whch would not overlap current API, but which can be taken into account by other modes in the core API when setting wifi up to another mode.

JiriBilek, 2019-07-11:
After 3 hours, I've got first reset. I'm keeping running the node to collect more data.
The previous configuration (dhcpc stopping commented out) ran 3 days without a problem.

12TA, 2019-07-11:
Does your sketch include the "WiFi.disconnect" command? If no, just say so,
if it does, have you tried commenting that command out. That was all I
needed to do. After, I had called
"WiFi.mode ( WIFI_OFF );"
but that alone seems enough to disconnect WiFi and reduce the power
consumption and no more random resets, which were driving me bonkers.

Dr Paul Stross.

d-a-v, 2019-07-11:
So it doesn't work like if you remove `wifi_station_dhcpc_stop()` @JiriBilek
Now can you please try and replace the test by:
```
    if (m != WIFI_STA && m != WIFI_AP_STA)
        // stop dhcp client on STA (safe to call if not started)
        for (netif* i = netif_list; i; i = i->next)
            if (i->num == STATION_IF)
                dhcp_stop(i);
```

and
```
extern "C" struct netif* eagle_lwip_getif(uint8 index);
```
by 
```
#include <lwip/dhcp.h>
```

d-a-v, 2019-07-11:
@12TA 
It's interesting that you state that `WiFi.mode(WIFI_OFF)` works where `WiFi.disconnect()` doesn't (wdt...).
The underlying API is different for the same desired effect and we might consider transparently replacing `WiFi.disconnect()` code (which disconnects STA) by a call to `WiFi.mode(AP or OFF)` (according to current mode).

JiriBilek, 2019-07-11:
@12TA: Yes, I have WiFi.disconnect(), see the fragment of the code:
```
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    Serial.println(F(", sleeping"));
    delay(100);
    WiFi.forceSleepBegin();  // turn off ESP8266 RF

```

@d-a-v: This time I stated to get wdt reset on every disconnect. Then it somehow settled (really strange, I did nothing and it started to work after a couple of minutes) and now it is running. I will keep watching it.
Note, after flashing the code I did a power cycle to be sure there is nothing left in RAM. 
The relevant code from ESP8266WiFiGeneric.cpp:
```
bool ESP8266WiFiGenericClass::mode(WiFiMode_t m) {
    if(_persistent){
        if(wifi_get_opmode() == (uint8) m && wifi_get_opmode_default() == (uint8) m){
            return true;
        }
    } else if(wifi_get_opmode() == (uint8) m){
        return true;
    }

    bool ret = false;

    if (m != WIFI_STA && m != WIFI_AP_STA)
        // calls lwIP's dhcp_stop(),
        // safe to call even if not started
        for (netif* i = netif_list; i; i = i->next)
            if (i->num == STATION_IF)
                dhcp_stop(i);

    ETS_UART_INTR_DISABLE();
    if(_persistent) {
        ret = wifi_set_opmode(m);
    } else {
        ret = wifi_set_opmode_current(m);
    }
    ETS_UART_INTR_ENABLE();

    return ret;
}

```

12TA, 2019-07-11:
For clarity I also used a
"WiFi.mode ( WIFI_OFF ); delay ( 10 ); WiFi.forceSleepBegin (); "

and removing

a prior
"WiFi.disconnect ( ); delay ( 10 );"
Fixed the reboots issue while still reducing power input and allowing a

Subsequent

"WiFi.forceSleepWake (); delay (1); WiFi.mode ( WIFI_STA ); delay (1); "

to resume WiFi functionality just prior to a quick post of data then back
to sleep.

I was copying code snippets I had found on line. Dr Paul Stross.

JiriBilek, 2019-07-11:
@12TA: I removed `WiFI.disconnect()` like you suggested. The MCU is entering low power mode (< 20 mA) and wakes up correctly. So far so good.
Testing with @d-a-v's latest fix.

12TA, 2019-07-11:
Thanks. I feel validated. The 'fix' for me has stood the test of time but
on 1 sketch and NodeMCU only (as that is all I have tried).

Dr Paul Stross.

TD-er, 2019-07-11:
> @TD-er when #6283 is solved maybe we can understand better your WDT issue.

I agree.
Not sure how long that issue may be present in the code, since the WD reboots did appear somewhere in mid 2018.
But the WD reboots may have several causes, and the symptoms described in that issue look very similar to what I'm seeing.

> @TD-er I was thinking of a new `WiFi.mode(WIFI_SHUTDOWN)` which would not overlap current API, but which can be taken into account by the core API when setting wifi up to another mode.

That sounds very good, since there is quite a lot of energy saving possible when turning off the WiFi, but right now I am not using that potential because of the reboots.
Should I add already an issue for it in the form of a feature request to keep this thread a bit more clean?

> 
> _edit_
> Also there is [this FW api](https://github.com/nodemcu/nodemcu-firmware/blob/master/sdk-overrides/include/user_interface.h) we don't so far know/use (not present in official `user_interface.h` but described in latest api reference datasheet).

I will look at what these do.
Is it part of the SDK3.x only?

JiriBilek, 2019-07-11:
The testing node is now running for 8 hours, no problem.
I really don't understand why removing `WiFi.disconnect()` could fix the issue :(
I am open to test whatever ideas.

Rob58329, 2019-07-11:
@d-a-v: Your first `"extern "C" struct...eagle_lwip_getif..."` suggestion does not work for me either.  If anything it seems to produce more frequent WDT crashes for me.  I will next try your `"#include <lwip/dhcp.h>"` suggestion...

@12TA & @JiriBilek: I would postulate that perhaps the "WiFi.disconnect()" command occasionally does the equivalent of the "wifi_station_dhcpc_stop()", and the crashes occur when "WiFi.disconnect()" does actually stop the dhcpc, and so the subsequent "wifi_station_dhcpc_stop()" line in "WiFi.mode(WIFI_OFF)" is not needed and so causes a delayed WDT crash.  Hence not doing the "WiFi.disconnect()" means the dhcpc always needs to be stopped by the "wifi_station_dhcpc_stop()" in "WiFi.mode(WIFI_OFF)" (hence no crash) - BUT this is just my supposition!??

Interestingly, in addition to the above "WiFi.disconnect()" perhaps not being needed,  I have just noticed that "WiFi.mode(WIFI_STA)" and "WiFi.forceSleepWake()" no longer appear to be necessary either.  It appears that the "WiFi.begin(ssid, password)" automatically does this on its own!??  (Although exactly which WiFi.mode()" it uses I'm not sure.) An Example sketch is:

```
#include <ESP8266WiFi.h>
const char* ssid     = "SSID";     // your network SSID (name of wifi network)
const char* password = "password"; // your network password
void setup() {
  Serial.begin(74880); delay(100);
  Serial.println("\nBooted ok - typical USB current now=75mA"); // typically for ESP8266-D1-mini
  WiFi.mode(WIFI_OFF); delay(100);
  Serial.println("-> WiFi.mode(WIFI_OFF) - typical USB current now=75mA");
  WiFi.forceSleepBegin(); delay(1);
  Serial.println("-> WiFi.forceSleepBegin() - typical USB current now=20mA");
  delay(1000);
  Serial.print("Attempting to connect to SSID: "); Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {Serial.print("."); delay(500);}
  Serial.print("\nConnected to "); Serial.print(ssid); Serial.println(" - typical USB current now=75mA"); 
  Serial.println("\nDemonstration that 'WiFi.mode(WIFI_STA)' and 'WiFi.forceSleepWake()' are no longer required!?"); 
}
void loop() {}

```
Which connects to my Wifi SSID fine, even without "WiFi.mode(WIFI_STA)" and "WiFi.forceSleepWake()"!?

d-a-v, 2019-07-12:
> since the WD reboots did appear somewhere in mid 2018.

#4482 was in March 2018 and it seems from that time assert/abort/panic are seen as WDT.

> > `WiFi.mode(`)`

> Should I add already an issue 

Why not, it would open the discussion

> Is it part of the SDK3.x only?

These symbols are present in our current firmware.
They will be useful when waking up from `WIFI_SHUTDOWN`.

TD-er, 2019-07-12:
> > since the WD reboots did appear somewhere in mid 2018.
> 
> #4482 was in March 2018 and it seems from that time assert/abort/panic are seen as WDT.
> 

OK, that does explain a lot.
I wasn't sure when the WD reboots did start to occur, since around that time we also had a lot of other issues at hand regarding build issues.  (Builds would break at random so it seems. Change a single line (even comments) and it would not connect to WiFi for example)
So if it was a commit from March 2018, I would say please make it the highest priority bug fix and it will probably eliminate lots and lots of other reported issues.

12TA, 2019-07-12:
I quite agree with you and don't understand the logic, as so intermittent
and random BUT I grappled with it for many hours trying work arounds, most
of which did not work and the problem seemed to get worse. Several of othrs
'Solved' solutions did not stand the tests of time. At one stage it got to
be every loop, till I removed the
WiFi.disconnect ( );
and suddenly it was stable! :-) My project is a GPS logger,  I feared
interference between WiFi and the GPS signals. Also it's battery powered,
so current saving was important. I also found the ESP8266 could not handle
the GPS serial stream while doing WiFi connections and postings. So, using
the WiFi and GPS alternately was desired,  with a 30 second loop between
ThingSpeak postings. It recovered from the reboots BUT I could not measure
cumulative distance. Now, I can.
I bought a crude USB current monitor and proved it had achieved a lower
power state. As I was confident my 'fix' worked, I wanted to share it with
people who might be able to explain it to me, and turn magic into logic,
and help others who had the same problem.

Paul Stross.

[39 more replies]

closed 2019-10-01 by d-a-v