esp8266/Arduino #6174: Cannot receive ESP-Now Broadcasts

littleyoda, 2019-06-02:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:				tested with D1 Mini and NodeCMU
- Core Version:			2.5.2 
- Development Env:		PlatformIO (espressif8266@2.2.1)
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			D1 Mini and NodeCMU
- Flash Mode:			DIO
- Flash Size: 		4MB
- lwip Variant:   -DLWIP_OPEN_SRC -DNONOSDK221=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200
- /usr/bin/python /home/XXX/.platformio/packages/tool-esptoolpy/esptool.py --chip esp8266 --port /dev/ttyUSB0" --baud 115200 write_flash 0x0 firmware.bin

### Problem Description

Currently the ESP8266 is able to send ESP-Now Broadcast, but is not able to receive ESP-Now Broadcast. 

The Problem was been fixed in the ESP8266_NONOS_SDK around Nov 2018.
https://github.com/espressif/ESP8266_NONOS_SDK/issues/8
https://github.com/espressif/ESP8266_NONOS_SDK/issues/134

Apparently this bugfix has not been applied yet. 
I tested the programm below against the ESP32. The ESP32 is able to receive the Broadcasts send by the ESP8266.

All in all, it is very difficult for an outstanding user to find out which changes from ESP8266_NONOS_SDK have been applied. Even after reading [SDK issues (SDK reverted from pre3 to 2.2.1)](https://github.com/esp8266/Arduino/issues/5784) and [Migrate core from NONOS SDK to FreeRTOS SDK](https://github.com/esp8266/Arduino/issues/5790)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>

#include <ESP8266WiFi.h>
extern "C" {
#include <espnow.h>
#include "user_interface.h"

}

#define WIFI_CHANNEL 1 

byte message[10] = {1,2,3,4,5,6,7,8,9,10};
unsigned long last = 0;

uint8_t broadcastMac[6] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
 

void setup() {
	Serial.begin(115200);
	delay(3000);

	WiFi.softAP("sender", "sendersender", WIFI_CHANNEL, false);
	WiFi.mode(WIFI_AP_STA);

  Serial.println("INIT: " + String(esp_now_init()));
	delay(10);
  Serial.println("Set Self Role: " + String(esp_now_set_self_role(ESP_NOW_ROLE_CONTROLLER)));

	// Callback for received messages
	esp_now_register_recv_cb([](uint8_t *mac, uint8_t *data, uint8_t len) {
		Serial.print("[" + String(millis()) + "] Received Message. Len=" + String(len));
	});
	WiFi.printDiag(Serial);
}

void loop() {
  // Send a new message every 5 sec
	if ((last + 5000) < millis()) {
		int ret = esp_now_send(broadcastMac, message, 10);
		last = millis();
		Serial.println("[" + String(millis()) + "] Sending new Message -- Result Code: " + String(ret));
	}
	delay(10);
}
```

devyte, 2019-06-02:
Unfortunately we're stuck on sdk 2.2.x for the moment, and this fix, among others, were applied post sdk 3.
Migration attempts to sdk3 have so far failed, and even the pre3 sdk was found to have issues. At this time, it looks like migrating to sdk 3 will require a rewrite from scratch of the core-sdk interface and/or of our build system.
Closing.

closed 2019-06-02 by devyte

d-a-v, 2019-07-05:
@littleyoda can you tripple check per [this comment](https://github.com/esp8266/Arduino/pull/6257#issuecomment-508686742)

littleyoda, 2019-07-06:
I tested it, and it works. Great. Thanks. 

If someone wants to reproduce it:
- Deinstall the esp8266 from the Boardmanager of Arduino
- [Install the development Version via git](https://github.com/esp8266/Arduino#using-git-version-basic-instructions)
- Select the Generic 8266 Board
- Select as SDK Version 2.2.1+100

If someone knows how to enable this version in platforumio, I would be happy about an message.

d-a-v, 2019-07-07:
For PlatformIO, with current master of this core, you can add `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22y` to your defines
(and remove `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x` or `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK3`).

littleyoda, 2019-07-08:
thanks .... Works ... great

For Platforumio:
```cpp
[env:d1_mini]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = d1_mini
framework = arduino

build_flags = -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22y
```

I added also a ESP.eraseConfig(), because sometimes the old wifi credentials causes problems. 

```cpp
#include <Arduino.h>

#include <ESP8266WiFi.h>
extern "C" {
#include <espnow.h>
#include "user_interface.h"

}

#define WIFI_CHANNEL 1 

byte message[10] = {1,2,3,4,5,6,7,8,9,10};
unsigned long last = 0;

uint8_t broadcastMac[6] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
 

void setup() {
	Serial.begin(115200);
	WiFi.disconnect();
	ESP.eraseConfig();
	delay(3000);

	  Serial.println(ESP.getSdkVersion());
 
	  WiFi.softAP("sender", "sendersender", WIFI_CHANNEL, false);
	WiFi.mode(WIFI_AP_STA);

  Serial.println("INIT: " + String(esp_now_init()));
	delay(10);
  Serial.println("Set Self Role: " + String(esp_now_set_self_role(ESP_NOW_ROLE_CONTROLLER)));

	// Callback for received messages
	esp_now_register_recv_cb([](uint8_t *mac, uint8_t *data, uint8_t len) {
		Serial.println("[" + String(millis()) + "] Received Message. Len=" + String(len));
	});
	WiFi.printDiag(Serial);
}

void loop() {
  // Send a new message every 5 sec
	if ((last + 5000) < millis()) {
		int ret = esp_now_send(broadcastMac, message, 10);
		last = millis();
		Serial.println("[" + String(millis()) + "] Sending new Message -- Result Code: " + String(ret));
	}
	delay(10);
}
```

theicfire, 2019-08-19:
Amazing, thank you!!! Here's a full project where it's working with PlatformIO: https://github.com/theicfire/led_sync/tree/master/led_hat

bwjohns4, 2022-07-13:
Just FYI (a few years later) this works with the default PlatformIO setup, sdk, etc. I was able to send/receive messages via broadcast (FF:FF:FF:FF:FF:FF) without any under the hood modifications. Also, there was no need to add_peer the broadcast address of (FF:FF:FF:FF:FF:FF).