esp8266/Arduino #6178: analogWrite() disable timer1 interrupt

Rofl92, 2019-06-04:
### Problem
Hi.
The timer1 works fine until i use the function analogWrite(pin, state).
What could be the problem?
If analogWrite() uses a timer1, how do I work with the analogWrite() and the timer1 at the same time? The Ticker library is not suitable for solving my problems, I need a hardware timer.

### Sketch

```cpp

#define PIN_LED 2
bool b = false;
void ICACHE_RAM_ATTR timer_callback(void){
  b=!b;
  Serial.println(b);
  wdt_reset();
}
void setup() {
  pinMode(PIN_LED, OUTPUT);
  analogWrite(PIN_LED, 512);
  Serial.begin(115200);
  timer1_disable();
  timer1_attachInterrupt(timer_callback);
  timer1_isr_init();
  timer1_enable(TIM_DIV16, TIM_EDGE, TIM_LOOP);
  timer1_write(100000);

//Set up ESP watchdog
  ESP.wdtDisable();
  ESP.wdtEnable(WDTO_8S);
}
void loop() {

}

```
```
10101010101010101010101010101010101010101010101010analogWrite enabled
```

#### Platform

- Hardware:				[ESP-12E]
- Core Version:			[latest]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB/3MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

JiriBilek, 2019-06-04:
AnalogWrite uses timer1, see https://github.com/esp8266/Arduino/blob/8859b818d8557ebada8d60e76891ac7e23f5aa96/cores/esp8266/core_esp8266_waveform.cpp#L133

You can't use both AnalogWrite and timer1 at the same time.

devyte, 2019-06-04:
Previous comment is correct. If you must have both, I suggest writing an extension to the waveform mechanism.
Closing due to not an issue.

closed 2019-06-04 by devyte