esp8266/Arduino #6179: No authentication module for https

slackwaree, 2019-06-04:
Hello,

I am using the latest git version of this library with the **WifiServerSecure** function.

This unlike the standard HTTP webserver does NOT provide **server.authenticate(www_username, www_password).** 

Are there any plans to implement basic http authentication on https or should I look for other library?

slackwaree, 2019-06-05:
Although this is a quite old thread:

https://arduino.stackexchange.com/questions/47111/https-with-esp8266webserver

The SSL support is still shit. I also having "TLS Connection was non properly terminated" with some curl clients.

Max key length is 1024 bits. 

If you don't answer I will just switch back to http and fuck you :(

Pablo2048, 2019-06-05:
Well, you definitely know, how to get the right answer... I personally think that the answer from developers would be something like "Switch back then and f**k yourself..."

closed 2019-06-05 by earlephilhower