esp8266/Arduino #6181: ESP8266WebServer example executes event twice

didier9, 2019-06-04:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it (there was none as the process did not crash).
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				[D1 mini]
- Core Version:			[not available]
- Development Env:		[Arduino IDE 1.8.9]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[Wemos D1 mini]
- Flash Mode:			[do not know, there is no "Flash Mode" field in the Tools menu]
- Flash Size: 		        [4MB/1MB SPIFFS]
- lwip Variant:                  [v2 pre-build MSS=536]
- Reset Method:		[ck]
- Flash Frequency:	        [do not know, default I guess]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL over USB]
- Upload Speed:              [115200]

### Problem Description

Example runs apparently fine but the event handler is executed twice. I did not check if the web page is actually sent twice.

I ran the example below by calling "http://192.168.1.52/" 
The Serial Monitor prints the line "Home" twice

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#ifndef STASSID
#define STASSID "XXXXXXX"
#define STAPSK  "xxxxxxxxxxxxxxxxxxxxxxxxx"
#endif

const char *ssid = STASSID;
const char *password = STAPSK;

// web server library: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/README.rst
const String & arg();     // get request argument value
const String & argName(); // get request argument name
int args();               // get arguments count
bool hasArg();            // check if argument exist

ESP8266WebServer server(80);

void setup( void ){
  pinMode( BUILTIN_LED, OUTPUT );
  digitalWrite( BUILTIN_LED, 0 );
  Serial.begin( 115200 );
  WiFi.mode( WIFI_STA );
  WiFi.begin( ssid, password );
  Serial.println("");

  // Wait for connection
  while( WiFi.status() != WL_CONNECTED ){
    delay( 500 );
    Serial.print(".");
  }

  Serial.println("");
  Serial.print( "Connected to " );
  Serial.println( ssid );
  Serial.print( "IP address: " );
  Serial.println( WiFi.localIP() );

   if( MDNS.begin( "esp8266" )){
    Serial.println( "MDNS responder started" );
  }

  server.on( "/", [](){
    Serial.println( "Home" );
    server.send( 200, "text/plain", "Home page" ); 
  });
  server.onNotFound( handleNotFound );
  server.begin();
  Serial.println( "HTTP server started" );
}

void loop( void ){
  server.handleClient();
  MDNS.update();
}

void handleNotFound() {
  digitalWrite( BUILTIN_LED, 1 );
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += ( server.method() == HTTP_GET ) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";

  for( uint8_t i = 0; i < server.args(); i++ ){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }

  server.send( 404, "text/plain", message );
  digitalWrite( BUILTIN_LED, 0 );
}

### Upload Messages
Sketch uses 282179 bytes (27%) of program storage space. Maximum is 1044464 bytes.
Global variables use 35212 bytes (42%) of dynamic memory, leaving 46708 bytes for local variables. Maximum is 81920 bytes.
Uploading 286320 bytes from C:\Users\xxxxxxx\AppData\Local\Temp\arduino_build_748390/sketch_jun04a.ino.bin to flash at 0x00000000
................................................................................ [ 28% ]
................................................................................ [ 57% ]
................................................................................ [ 85% ]
........................................                                         [ 100% ]

### Serial Monitor

..
Connected to XXXXXXX
IP address: 192.168.1.52
MDNS responder started
HTTP server started
Home
Home

```
Debug messages go here
```

didier9, 2019-06-04:
I did check with Wireshark and it looks as if the desktop (Chrome 192.168.1.38) is requesting the page twice:
[image]
Checked on Firefox and it looks normal (one request) under Firefox.
Not sure what is going on but it does not look like the issue is with the Arduino/ESP8266

didier9, 2019-06-04:
Digging further came across this:
https://bugs.chromium.org/p/chromium/issues/detail?id=64810
It is an old issue but it may be that Chrome is not happy about the way the page is loaded and resends the request, even though Wireshark does not seem to indicate anything improper. 
Any help fixing that would be appreciated, it is important to me that any request be served just once.

earlephilhower, 2019-06-04:
In a perfect world, HTTP GET is supposed to be idempotent so it should not change object state even if it's executed dozens of times.  Figuring out why Chromium is unhappy is probably a good thing, but I fear it's a ton of work.

HTTP POST, however, isn't and as such Chrome cannot re-execute them willy-nilly.  Would it be possible to change from GET to POST in your app and sidestep the issue?

devyte, 2019-06-06:
Closing due to not a core issue.

closed 2019-06-06 by devyte