esp8266/Arduino #6184: Crash on WiFiClientSecure::_connectSSLServerRSA

esseredigitale, 2019-06-06:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.2.1]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266]
- Flash Mode:			[esptool]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Crash "Abort called" using ESP8266WebServerSecure.h for make a secure webserver.
Going deeply to code I found line of the crash on method WiFiClientSecure::_connectSSLServerRSA on WiFiClientSecureBearSSL.cpp, line specified below.

### [MCVE] Sketch

```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServerSecure.h>
#include <WiFiClient.h>

BearSSL::ESP8266WebServerSecure webserver(443);

void setup() {
    BearSSL::X509List *serverCertList = new BearSSL::X509List(serverCert);
    BearSSL::PrivateKey *serverPrivKey = new BearSSL::PrivateKey(serverKey);
    webserver.setRSACert(serverCertList, serverPrivKey);
    webserver.on("/", [](){
      webserver.send(200, "Hello World!");
      }
    );
    webserver.begin();
}

void loop() {
    webserver.handleclient(); /// CRASH
}

```

`bool WiFiClientSecure::_connectSSLServerRSA(const X509List *chain,
    const PrivateKey *sk,
    const X509List *client_CA_ta) {
  _freeSSL();
  _oom_err = false;
  _sc_svr = std::make_shared<br_ssl_server_context>();
  _eng = &_sc_svr->eng; // Allocation/deallocation taken care of by the _sc shared_ptr
  _iobuf_in = std::shared_ptr<unsigned char>(new unsigned char[_iobuf_in_size], std::default_delete<unsigned char[]>()); //////// CRASH
  _iobuf_out = std::shared_ptr<unsigned char>(new unsigned char[_iobuf_out_size], std::default_delete<unsigned char[]>());
  if (!_sc_svr || !_iobuf_in || !_iobuf_out) {
    _freeSSL();
    _oom_err = true;
    DEBUG_BSSL("_connectSSLServerRSA: OOM error\n");
    return false;
  }

  br_ssl_server_init_full_rsa(_sc_svr.get(), chain ? chain->getX509Certs() : nullptr, chain ? chain->getCount() : 0, sk ? sk->getRSA() : nullptr);
  br_ssl_engine_set_buffers_bidi(_eng, _iobuf_in.get(), _iobuf_in_size, _iobuf_out.get(), _iobuf_out_size);
  if (client_CA_ta && !_installServerX509Validator(client_CA_ta)) {
    DEBUG_BSSL("_connectSSLServerRSA: Can't install serverX509check\n");
    return false;
  }
  if (!br_ssl_server_reset(_sc_svr.get())) {
    _freeSSL();
    DEBUG_BSSL("_connectSSLServerRSA: Can't reset server ctx\n");
    return false;
  }

  return _wait_for_handshake();
}

`
Can someone help me?
Thanks a lot

earlephilhower, 2019-06-06:
We need a full MCVE that compiles and shows the failure, and the full stack trace decoded (including any lines before or after like "last failed alloc..." and the PC/registers).

You are 99% likely running out of memory and std::shared_pointer is trying to `throw` an out of memory exception, but since exceptions are disabled you get an abort().

esseredigitale, 2019-06-06:
> We need a full MCVE that compiles and shows the failure, and the full stack trace decoded (including any lines before or after like "last failed alloc..." and the PC/registers).
>
full MCVE is complicated with a lot of files!
If I use my code on a empty sketch, crash don't happens.
 
> You are 99% likely running out of memory and std::shared_pointer is trying to `throw` an out of memory exception, 

how can I prevent it?

> but since exceptions are disabled you get an abort().

how can I enable exceptions?
Thanks

earlephilhower, 2019-06-06:
Post the entire raw and decoded crash dumps, from start to finish at least.  There will most likely be a line after the stack with "last failed alloc" or something similar.

Exceptions are a menu option, just select it and rebuild (but your code size will bloat a lot).  It won't stop you from crashing, of course, but you will get a message about the kind of exception thrown.

Preventing running out of memory?  Not sure what to say other than "use less."

esseredigitale, 2019-06-06:
Resolved!
As you anticipated, it is an out of memory problem: a lot of resources in memory.
Thanks a lot

closed 2019-06-06 by esseredigitale