esp8266/Arduino #6185: WString.h: SSOSIZE is 12, shouldn't it be bigger?

everslick, 2019-06-06:
SSOSIZE is currently only as big as the struct needed to store the pointer to heap, string capacity and string length. I wonder, if we should not also take the heap allocation overhead into account, which is, I believe, 16 bytes at least. so I propose to increase the SSOSIZE to sizeof (struct _ptr) + 4 + heap allocation overhead = 28 bytes?

earlephilhower, 2019-06-06:
I think maybe you're thinking of 64b pointers.  sizeof(struct_ptr) == 4+2+2 == 8 bytes.  So adding 4 more bytes gives you 12, and that leaves 4 bytes for the heap overhead and gives a round 16.  

````
String *s;

void setup() {
  Serial.begin(115200);
  Serial.printf("\n\npre: %d\n", ESP.getFreeHeap());
  s = new String();
  Serial.printf("\n\npost: %d\n", ESP.getFreeHeap());
}

void loop() {}
````

gives

````
pre: 54424

post: 54408
````

closed 2019-06-06 by earlephilhower

everslick, 2019-06-06:
so why not making SSOSIZE 16 then? we still use the same amount of bytes and avoid going through the heap, no?

earlephilhower, 2019-06-06:
The heap overhead uses the last 4 bytes of the 16-byte allocation unit.  Changing SSOSIZE to 16-1 makes the allocation use 24 bytes (because the memory allocation unit is 8 bytes, not 16...16 is String's allocation block):
````
pre: 54416
post: 54392
````

everslick, 2019-06-06:
Thank you very much!

earlephilhower, 2019-06-06:
Sure.  It was a good observation, and worth double checking.  Thx!