esp8266/Arduino #6189: HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

Shaileshverma123, 2019-06-07:
Hi all I am trying to do OTA update but facing these issue. Anyone can help me out.
Thank you 

I am using Arduino 1.8.8
ESP8266         2.4.2 
  here is the path and details for update.

void Device::firmwareUpdate(String ver)
{
	// delay(2000);
 //  Serial.println(F("Connected, updating"));
    if ((WiFi.status() == WL_CONNECTED)) {

      t_httpUpdate_return ret = ESPhttpUpdate.update("http://xyz.com/febin/COG-SON-BS-1", ver, fingerprint);
        
        
        
        //t_httpUpdate_return ret = ESPhttpUpdate.update("https://xyz.com/febin/COG-SON-BS-9.bin");

      
      switch (ret) {
        case HTTP_UPDATE_FAILED:
          Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
          break;

        case HTTP_UPDATE_NO_UPDATES:
          Serial.println(F("HTTP_UPDATE_NO_UPDATES"));
          break;

        case HTTP_UPDATE_OK:
          Serial.println(F("HTTP_UPDATE_OK"));
          break;
      }
    }
  Serial.println(F("Update failed"));
  ESP.reset();

}

Shaileshverma123, 2019-06-07:
@igrr  Can you help me out

Shaileshverma123, 2019-06-07:
@scargill Can you help me
anyone knows anything about the same please let me

dmdt, 2019-06-07:
You trying update through HTTP, but call overloaded method for HTTPS update. If server does not redirect http requests - you can try to remove "fingerprint" parameter. Else you should pass to method link with HTTPS protocol and make sure that you enter correct fingerprint.

devyte, 2019-06-08:
Closing due to user error. Also, this is an issue tracker, not a support forum for general help.

closed 2019-06-08 by devyte

Shaileshverma123, 2019-06-10:
@devyte I tried without fingerprint then also same error.

Shaileshverma123, 2019-06-10:
@devyte  Thank you for your reply ... can you help me out