esp8266/Arduino #6191: Activating static IP makes WiFi.status() return WL_CONNECTED

aerlon, 2019-06-10:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			Latest Git as of 2019-06-10
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

The changes made in PR #6026 by @d-a-v makes `WiFi.status() == WL_CONNECTED` become true whenever one activates static IP via `WiFi.config`. This was not the case before the change, and it also seems strange that the status would be connected even when no connection has been made, so it is likely a bug. Calling `WiFi.disconnect()` after static IP activation makes `WiFi.status()` return 0 (`WL_IDLE_STATUS`), as was the case before the PR change. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>

void activateStaticIP()
{
  // WiFi mode does not matter 
  WiFi.mode(WIFI_STA);
  //WiFi.mode(WIFI_AP_STA); 
  
  WiFi.config(IPAddress(192,168,4,22), IPAddress(192,168,4,1), IPAddress(255,255,255,0)); 
}

void setup() {
  // put your setup code here, to run once:

  WiFi.persistent(false);
  
  Serial.begin(115200);
  delay(50); // Wait for Serial.

  Serial.println();
  Serial.println();

  WiFi.disconnect();
    
  activateStaticIP();
}

void loop() {
  // put your main code here, to run repeatedly:
  
  Serial.print("status pre " + String(WiFi.status()));
  if(WiFi.status() == WL_CONNECTED)
    Serial.println(", this means WiFi.status() == WL_CONNECTED");
  else
    Serial.println();
    
  WiFi.disconnect(); // Adding this disconnect seems to fix WiFi status.
  Serial.println("status post " + String(WiFi.status()));

  delay(3000);
}
```

d-a-v, 2019-06-10:
@aerlon can you give #6194 a try ?

aerlon, 2019-06-11:
@d-a-v I can confirm #6194 fixes the problem.

PR #6194 (2019-06-17): lwip2: (re)fix setting static ip address
#6026 was not tested in every cases.
This PR hopefully fixes a missing one
(trigger address change by calling lwIP-v1.4 api *only when ip is already set by other mean (generally dhcp)*)
fixes #6191
closed 2019-06-11 by aerlon