esp8266/Arduino #6192: Incorrect String length after applying WString::replace

Hieromon, 2019-06-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU1.0
- Core Version:			2.5.2 release
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			NodeMCU1.0
- Flash Mode:			dio
- Flash Size: 		4MB/1MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

Depending on the length of WString::replace, replaced string actual length may be 1 byte redundant.
The result of the MCVE sketch will output to the serial monitor as below:
```
abcde12345⸮
11
```
The last character will be garbage.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>

void setup() {
  delay(1000);
  Serial.begin(115200);
  String s = "#12345";
  String l = s;
  l.replace("#", "abcde");
  Serial.println(l.c_str());
  Serial.println(l.length());
}

void loop() {
}
```

earlephilhower, 2019-06-10:
@Hieromon, the String short string optimization was changed recently.  Could you try your app with GIT head and see if it fixes things?  The test case you've give passes, but I'll see about adding it and some more to the CI suite.

Hieromon, 2019-06-11:
@earlephilhower, I confirmed that WString::replace works correctly with the recent GIT with more shorter strings to make sure. They are replaced correctly.
Thank you for your quick response.

closed 2019-06-11 by Hieromon